arendst/Tasmota #3841: Sonoff 4CH and external switches for every relay

Kepete, 2018-09-18:
So first I would like to point out that there are 4 types of the Sonoff 4CH: 
- Sonoff 4CH
- Sonoff 4CH R2
- Sonoff 4CH PRO
- Sonoff 4CH PRO R2
So I understand the difference between 4CH and 4CH PRO, but what does the R2 give me? 

But now to the main point...
I'm in a need for a 4 gang Sonoff and I want to be able to switch all the relays with mqtt/web/http and also external switches. I'm doing the same with Sonoff Dual R2 and Sonoff Basic. 

Solution with 4CH and/or 4CH R2: 
I had a look a the module and I solder wires to the 4 buttons on the module and use them as external switches. But one button is already connected to GPIO0, which will bring the module into Programming Mode. And that I don't want to happen. I can't see any other free GPIO's. 
What possibility would there be to add a external switch for the 4th relay? 

Solution with 4CH PRO and/or 4CH PRO R2: 
The PRO versions have already more GPIO's free to use. But it seems here that the module won't go into programming mode using the buttons provided. So I could again solder wires to the 4 buttons on the module and use them as external switches. But the PRO is a little more complex and I need only to switch the mains, so PRO wouldn't be needed. It also costs a little more.

ascillato, 2018-09-18:
Hi

Sorry, What is the Tasmota software issue? 

As you already said: just solder wires in your gpios.

closed 2018-09-18 by ascillato2

meingraham, 2018-09-18:
@Kepete R2 is just a newer revision of the device. Same functionality. Usually the PDB changes. Sometimes different (or newer versions of) ICs. This means that the location of GPIO access points on the PCB have moved.

One example of a change from one revision to the next is that on the 4Ch Pro R2, one of the device buttons is connected to GPIO0 so you can easily put the device in programming mode (like you do with the Sonoff Basic). On the 4Ch Pro (revision 1), you have to find the GPIO0 pin on the ESP8285 itself in order to pull it to ground during power up.

Subtle changes, but usually to address an issue or make an improvement of some sort.

Regards.

Mike