arendst/Tasmota #3842: The id (POWER1,2,3,4...) is not added if we have more switches (09Switch1 10Swich2...)

gabrielklein, 2018-09-18:
Hi,

I use a Generic Module with 4 switches:

- 09 Switch1
- 10 Switch2
- 11 Switch3
- 12 Switch4

My goal is to send a MQTT message when a switch is switched.

I want to have when I switch button 1 then button 2.

> /sonoff/physwitchup/RESULT {"POWER1":"ON"}
> /sonoff/physwitchup/POWER1 ON
> /sonoff/physwitchup/RESULT {"POWER1":"OFF"}
> /sonoff/physwitchup/POWER1 OFF
> /sonoff/physwitchup/RESULT {"POWER2":"ON"}
> /sonoff/physwitchup/POWER2 ON
> /sonoff/physwitchup/RESULT {"POWER2":"OFF"}
> /sonoff/physwitchup/POWER2 OFF

I have

> /sonoff/physwitchup/RESULT {"POWER":"ON"}
> /sonoff/physwitchup/POWER ON
> /sonoff/physwitchup/RESULT {"POWER":"OFF"}
> /sonoff/physwitchup/POWER OFF
> /sonoff/physwitchup/RESULT {"POWER":"ON"}
> /sonoff/physwitchup/POWER ON
> /sonoff/physwitchup/RESULT {"POWER":"OFF"}
> /sonoff/physwitchup/POWER OFF

In sonoff.ino, we count the number of devices as
```
      if (pin[GPIO_REL1 +i] < 99) {
        pinMode(pin[GPIO_REL1 +i], OUTPUT);
        devices_present++;
      }
```
and we use that to chose if we add the 1,2,3,4 after POWER.
Of course we have a small issue here.

If I add 2 "virtual" relays, the number is correctly added... but I don't want to add relays :) 
It seams that we have mixed relays and switches.

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Plat
- [x] Hardware used : NodeMCU V3
- [x] Provide the output of command ``status 0`` :
```
22:38:17 SRC: WebConsole from 10.0.0.151
22:38:17 RSL: Received Topic /status, Data Size 1, Data 0
22:38:17 RSL: Group 0, Index 1, Command STATUS, Data 0
22:38:17 MQT: /sonoff/physwitchup/STATUS = {"Status":{"Module":18,"FriendlyName":["physwitchup"],"Topic":"physwitchup","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:38:17 MQT: /sonoff/physwitchup/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:46:57","StartupUTC":"2018-09-18T20:51:20","Sleep":0,"BootCount":8,"SaveCount":86,"SaveAddress":"F6000"}}
22:38:17 MQT: /sonoff/physwitchup/STATUS2 = {"StatusFWR":{"Version":"6.2.1.4","BuildDateTime":"2018-09-18T22:32:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:38:17 MQT: /sonoff/physwitchup/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["KleinUp","Klein"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
22:38:17 MQT: /sonoff/physwitchup/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":536,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F8AE794","240003A0","23B617CE","000002C0"]}}
22:38:17 MQT: /sonoff/physwitchup/STATUS5 = {"StatusNET":{"Hostname":"physwitchup-7846","IPAddress":"10.0.0.205","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"A0:20:A6:25:1E:A6","Webserver":2,"WifiConfig":5}}
22:38:17 MQT: /sonoff/physwitchup/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.60","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_251EA6","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:38:17 MQT: /sonoff/physwitchup/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 18 21:38:17 2018","Local":"Tue Sep 18 22:38:17 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:14","Sunset":"18:39"}}
22:38:17 MQT: /sonoff/physwitchup/STATUS10 = {"StatusSNS":{"Time":"2018-09-18T22:38:17","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON"}}
22:38:17 MQT: /sonoff/physwitchup/STATUS11 = {"StatusSTS":{"Time":"2018-09-18T22:38:17","Uptime":"0T00:46:57","Vcc":2.674,"Wifi":{"AP":1,"SSId":"KleinUp","BSSId":"C4:E9:84:EF:37:52","Channel":1,"RSSI":100}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-18:
Hi,

> My goal is to send a MQTT message when a switch is switched.

If you want to send MQTT messages when using SWITCHES or BUTTONS, you can use Rules for that. You don't need to use virtual Relays.

Please check the examples in the wiki. (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold)

> ... `devices_present` ... Of course we have a small issue here.

What you found is not a Bug, Sorry. That is the way is working. `devices_present` is equal to amount of relays + lights (ws2812, etc). Rules were designed for changing Tasmota Behaviours when you want a different behaviour than the standard Sonoff Hardware.
Switches are inputs, Relays are outputs, `devices_present` are outputs to be used for example with Alexa.
Hope this clarifies it.

Do not hesitate on asking anything else you need. For fast answers, please find us in Discord Chat.

Thanks.

closed 2018-09-18 by ascillato2

gabrielklein, 2018-09-19:
Sorry, but I think we really have a small issue.
It's not logical... or you have another more global logic.

But it was reported a few times in the past (or there is a miss-understanding somewhere).
https://github.com/arendst/Sonoff-Tasmota/issues/922
https://github.com/arendst/Sonoff-Tasmota/issues/1653

When I define 2,3,4,5... switches or buttons I expect to be able to differentiate them on the message that is sent using MQTT!

In this code
```
char* GetPowerDevice(char* dest, uint8_t idx, size_t size, uint8_t option)
{
  char sidx[8];

  strncpy_P(dest, S_RSLT_POWER, size);                // POWER
  if ((devices_present + option) > 1) {
    snprintf_P(sidx, sizeof(sidx), PSTR("%d"), idx);  // x
    strncat(dest, sidx, size);                        // POWERx
  }
  return dest;
}
```
That is called by 
`    GetTopic_P(stopic, CMND, key_topic, GetPowerDevice(scommand, device, sizeof(scommand), key));   // cmnd/switchtopic/POWERx

ascillato, 2018-09-19:
Hi,

Have you tried the example of the link to the wiki I have post for you?

gabrielklein, 2018-09-19:
Hi, I'll give it a try in a few hours (at work just now).

If it doesn't work, I'll flash a modified version of the code :) That's what I love with opensource!

Thank you anyway :)

ascillato, 2018-09-19:
Hi,

No problem at all.

Do as you are pleased. But remember that it is not a bug in the code. Tasmota is meant for sonoff devices, if you want to use a sonoff in a non-standard way, Theo has developed the rules features to modify the standard behaviour to make what you need.