arendst/Tasmota #3843: MQTT connection via mDNS discovery fails after upgrade to v6.2.1

dweston, 2018-09-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._
MQTT connection via mDNS discovery has failed after upgrade to v6.2.1.

A Sonoff POW module running v5.13.0 firmware had been operating flawlessly for some time. It was configured to reference the MQTT Host by mDNS discovery `xxxx.local`.

After a successful upgrade (via v5.14.0) to v6.2.1 the module will not connect to the MQTT broker:
`MQT: Connect failed to xxxx.local:xxxx, rc -2. Retry in 10 sec`

If the module's MQTT Host configuration is subsequently reconfigured to replace the mDNS reference with the specific MQTT broker IP address, MQTT connectivity is immediately re-established.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Sonoff POW
- [x] _Development/Compiler/Upload tools used_ : None
v6.2.1 firmware was loaded via OTA option: "Upgrade by web server"
Source: [http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin](url)
- [x] _Provide the output of command_``status 0`` :
``status 0`` output with MQTT Host referenced via mDNS
```
11:09:47 RSL: stat/test/STATUS = {"Status":{"Module":6,"FriendlyName":["Test"],"Topic":"test","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:09:47 RSL: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:21:26","StartupUTC":"2018-09-19T00:48:21","Sleep":0,"BootCount":26,"SaveCount":151,"SaveAddress":"FA000"}}
11:09:47 RSL: stat/test/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:09:47 RSL: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx","xxxx"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
11:09:47 RSL: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
11:09:47 RSL: stat/test/STATUS5 = {"StatusNET":{"Hostname":"test","IPAddress":"xx.xx.xx.xx","Gateway":"xx.xx.xx.xx","Subnetmask":"255.255.255.0","DNSServer":"xx.xx.xx.xx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
11:09:47 RSL: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx.local","MqttPort":xxxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3752FD","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:09:47 RSL: stat/test/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 19 01:09:47 2018","Local":"Wed Sep 19 11:09:47 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"05:48","Sunset":"17:48"}}
11:09:47 RSL: stat/test/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:09:47 RSL: stat/test/STATUS10 = {"StatusSNS":{"Time":"2018-09-19T11:09:47","ENERGY":{"Total":0.048,"Yesterday":0.103,"Today":0.048,"Power":4,"Factor":0.24,"Voltage":242,"Current":0.076}}}
11:09:47 RSL: stat/test/STATUS11 = {"StatusSTS":{"Time":"2018-09-19T11:09:47","Uptime":"0T00:21:26","Vcc":3.218,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","RSSI":100,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```

``status 0`` output with MQTT Host referenced by specific IP address
```
11:17:42 MQT: stat/test/STATUS = {"Status":{"Module":6,"FriendlyName":["Test"],"Topic":"test","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:17:42 MQT: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:37","StartupUTC":"2018-09-19T01:17:05","Sleep":0,"BootCount":27,"SaveCount":154,"SaveAddress":"F7000"}}
11:17:43 MQT: stat/test/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:17:43 MQT: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx","xxxx"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
11:17:43 MQT: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
11:17:44 MQT: stat/test/STATUS5 = {"StatusNET":{"Hostname":"test","IPAddress":"xx.xx.xx.xx","Gateway":"xx.xx.xx.xx","Subnetmask":"255.255.255.0","DNSServer":"xx.xx.xx.xx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
11:17:44 MQT: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"xx.xx.xx.xx","MqttPort":xxxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3752FD","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:17:45 MQT: stat/test/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 19 01:17:44 2018","Local":"Wed Sep 19 11:17:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"05:48","Sunset":"17:48"}}
11:17:45 MQT: stat/test/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:17:45 MQT: stat/test/STATUS10 = {"StatusSNS":{"Time":"2018-09-19T11:17:45","ENERGY":{"Total":0.049,"Yesterday":0.103,"Today":0.049,"Power":4,"Factor":0.00,"Voltage":244,"Current":8.547}}}
11:17:46 MQT: stat/test/STATUS11 = {"StatusSTS":{"Time":"2018-09-19T11:17:45","Uptime":"0T00:00:40","Vcc":3.217,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","RSSI":98,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
MQTT Host configured to mDNS discovery
MQTT connectivity fails - `rc -2`

MQTT Host configured as static IP address
MQTT connectivity successful

**Expected behavior**
_A clear and concise description of what you expected to happen._
MQTT connectivity should still be possible via mDNS discovery

**Screenshots**
_If applicable, add screenshots to help explain your problem._
n/a

**Additional context**
_Add any other context about the problem here._
n/a

**(Please, remember to close the issue when the problem has been addressed)**
Noted

ascillato, 2018-09-19:
Hi,

I could reproduce this issue. mDNS is not working for MQTT Discovery.

arendst, 2018-09-20:
Can't see any code change between 5.13.0 and 6.2.1 with regards to mqtt mDNS.

The way it should works is:
- Enable defines USE_DISCOVERY and MQTT_HOST_DISCOVERY
- Reset current mqtthost to empty string (``mqtthost 0`` or ``mqtthost "``)
- After restart mDNS is enabled and the mqtthost name is requested from the mDNS sever using mqtt and tcp keywords.
- If mDNS server replies with IP addresses the first IP address is selected and replaces the empty mqtthost.
- Mqtt then uses this IP address to connect the mqtt server.

The next restart will keep using the stored IP address and no mDNS request is made as requested by users.

Can you verify if this scenario works as I do not own a mDNS server (I use dnsmasq, a local full DNS server).

lalo-uy, 2018-09-20:
That way mDNS supply the mqtt host ip.
What Dion do is provide a name like "myhost.local" as mqtt host and what
that mDNS resolve that name.

arendst, 2018-09-20:
In that case, if it worked in 5.13.0 it should work in 6.2.1 too as nothing has changed there.

The local mDns server service on the MQTT server seems to fail to recognize request from the device...

ascillato, 2018-09-20:
Hi, 

using the solution from @arendst it works for me. (`mqtthost "` and checking my network for other mDNS devices)

In my case the issue was that when Tasmota was asking for MQTT server I had 2 Raspberry answering that, and the one that Tasmota was connecting had a misconfigured MQTT server. Turning off that forgotten and misconfigured RPI, Tasmota was able to find the MQTT server I was using before (in manual configuration).

So, Tasmota mDNS is working fine. If it is not working for you, please check your Network. Remember that mDNS sometimes take some seconds to find it the first time.

Thanks.

closed 2018-09-20 by ascillato2

dweston, 2018-09-21:
> Hi,
> 
> Using the solution from @arendst it works for me. (`mqtthost "` and checking my network for other mDNS devices)
> 
> In my case the issue was that when Tasmota was asking for MQTT server I had 2 Raspberry answering that, and the one that Tasmota was connecting had a misconfigured MQTT server. Turning off that forgotten and misconfigured RPI, Tasmota was able to find the MQTT server I was using before (in manual configuration).
> 
> So, Tasmota mDNS is working fine. There is no Bug. If it is not working for you, please check your Network for other devices. Remember that mDNS sometimes take some seconds to find it the first time.
> 
> Thanks.

@ascillato When you advise me to use @arendst 's solution I'm not sure what you are referring to. How do I check multiple local mDNS servers?

dweston, 2018-09-21:
> The way it should works is:
>
>     * Enable defines USE_DISCOVERY and MQTT_HOST_DISCOVERY
>
>     * Reset current mqtthost to empty string (`mqtthost 0` or `mqtthost "`)
>
>     * After restart mDNS is [...]

Oh I see you mean - follow this scenario. I will try to fathom this and see how I go.

ascillato, 2018-09-21:
> When you advise me to use @arendst 's solution I'm not sure what you are referring to.

well, the steps Theo said:

> * Enable defines USE_DISCOVERY and MQTT_HOST_DISCOVERY
> * Reset current mqtthost to empty string (mqtthost 0 or mqtthost ")
> * After restart mDNS is enabled and the mqtthost name is requested from the [...]

> How do I check multiple local mDNS servers?

You can use an App for your phone
And also check if you have another devices MQTT capable like Raspberry Pi, Vacuum Cleaners, etc.
Also enable the `weblog 4` in order to see all the Tasmota Log at startup.

dweston, 2018-09-21:
To follow the first dot point looks like it requires locally compiling a .bin file with those defines set.

Or can USE_DISCOVERY and MQTT_HOST_DISCOVERY be set from the device Console?

ascillato, 2018-09-21:
If you use the precompiled bins from release, those are already included.