arendst/Tasmota #3845: topic template not used for stat response when toggled via MQTT

mrambossek, 2018-09-19:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
since version 6(?), the default/preferred mqtt topic template is appearently
> tasmota/%topic%/%prefix%/

this is properly used when, for example, toggling the state via webif, to build a stat like this:
> tasmota/3ddrucker/stat/POWER

other things like the LWT are fine too. however, if i trigger the state via MQTT like this:
> tasmota/3ddrucker/cmnd/POWER:  ON

the responses will be:
> stat/3ddrucker/RESULT: {"POWER":"ON"}
> stat/3ddrucker/POWER: ON

.. which does not follow the template.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : S20 Socket and others
- [X] _Development/Compiler/Upload tools used_ : VSCode, PlatformIO
- [X] _Provide the output of command_``status 0`` :
{"Status":{"Module":8,"FriendlyName":["3ddrucker"],"Topic":"3ddrucker","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

**To Reproduce**
1. set MQTT_FULLTOPIC to "tasmota/%topic%/%prefix%/"
2. send a POWER cmnd via mqtt
3. look at stat answer

**Expected behavior**
response should follow the topic template?

ascillato, 2018-09-19:
Sorry,

Tasmota does not work that way.

Please, read the wiki. Thanks

mrambossek, 2018-09-19:
> Sorry,
> Tasmota does not work that way.

what way, exactly?

> Please, read the wiki. Thanks

i did. from what i see here https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features#mqtt-topic-definition i would suggest that what i am expecting is correct. can you point out where it says that it isnt?

ascillato, 2018-09-19:
MQTT:

Status sent from Tasmota to MQTT Broker:

stat/[sonofftopic]/[status of relays or sensors or other]

Commands sent from MQTT Broker to Tasmota

cmnd/[sonofftopic]/[command]

mrambossek, 2018-09-19:
are you saying that the 5.0.5+ feature of MQTT_FULLTOPIC is simply not fully implemented?
again, if i toggle the relais via webif, i get the "properly formatted" message via mqtt.
its only the "reponse to mqtt toggle" that is "wrong".

ascillato2, 2018-09-19:
> are you saying that the 5.0.5+ feature of MQTT_FULLTOPIC is simply not fully implemented?

never say that.

> its only the "reponse to mqtt toggle" that is "wrong".

It is not wrong. It is the way it is implemented.

closed 2018-09-19 by ascillato2

mrambossek, 2018-09-19:
sorry but you are not making any sense to me. i am not asking a question.
let me re-iterate. if i press the TOGGLE webinterface button on my S20, i get the following messages via mqtt:

> tasmota/3ddrucker/stat/RESULT: {"POWER":"OFF"}
> tasmota/3ddrucker/stat/POWER: OFF

however, if i toggle via mqtt (node-red), which sends the command:
> tasmota/3ddrucker/cmnd/POWER: OFF

the result is

> stat/3ddrucker/RESULT: {"POWER":"OFF"}
> stat/3ddrucker/POWER: OFF

which means that it is ignoring my MQTT_FULLTOPIC "tasmota/%topic%/%prefix%/" template.

mrambossek, 2018-09-19:
can you please read what i wrote and not simply close my issue? not very helpful.

ascillato, 2018-09-19:
Hi,

I understand you. But that is not the way Tasmota works. Sorry. You expect something that it is not there. It is just as simple as:

MQTT:

Status sent from Tasmota to MQTT Broker:

`stat/[sonofftopic]/[status of relays or sensors or other]`

Commands sent from MQTT Broker to Tasmota

`cmnd/[sonofftopic]/[command]`

Sorry if this don't fulfill your expectations. You are free to modify the code and adapt it to your needs.

mrambossek, 2018-09-19:
hi,
i - or rather, a friend of mine - found the solution to my problem, best described here https://github.com/arendst/Sonoff-Tasmota/issues/2165#issuecomment-373298180 . appearently i just kept searching for the wrong things.

i would still consider this a bug, #1893 unfortunately does not have a resolution or anything. but i have a workaround now.

im still confused with your answer though - as i stated in the very first post, the mqtt messages from tasmota to the broker *if and only if initiated by the webif* do not look like you wrote here, but instead, are formatted as expected. so that *works* and *is implemented*.

its just that when you trigger via mqtt, do they fall back to this "old, un-formatted" schema.