arendst/Tasmota #3847: Enhance Timer to support "x hours/minutes from now"

aalcock, 2018-09-19:
I have a dumb sensor attached to the Sonoff - it brings GPIO14 high on detect, then drops to low after a short period. I would like to use this to trigger the Sonoff relay to switch on for a fixed duration period (say 5 minutes) then switch off. I can easily trigger the relay to go on, but I can't easily switch it off again after a *specific duration*, only at a specific time.

I tried using a Timer for this, but I can't set the timer's time relative to *now*, specifically the moment at which the timer command is processed by the firmware. In the current implementation, time is expressed only as `HH:MM` relative to midnight, sunrise or sunset.

I would like to be able to do something like this:

     Timer1 {"Arm":1,"Mode":0,"Time":"+00:05","Days":"1111111","Repeat":0,"Output":1,"Action":0}

meaning: Arm Timer1 to turn off Switch1 5 minutes from now. 

An alternative could use a different value for Mode: 

     Timer1 {"Arm":1,"Mode":3,"Time":"00:05","Days":"1111111","Repeat":0,"Output":1,"Action":0}

It would then be trivial to use Rules to control the Sonoff device. When GPIO14 goes high, one rule would:

* Switch on Switch1
* Create/update Timer1 to wait 5 minutes and switch off Switch1

Whenever GPIO goes high again, the timer is updated.

Right now I have set up Node Red to perform this logic, but it is obviously better that this logic be on the device.

Jason2866, 2018-09-19:
Should be possible with RuleTimer. 
- Define GPIO 14 as a switch2.
- fire Rule with Switch2 event and Power Relais1 on and start RuleTimer
- Rule with RuleTimer event to Power Relais1 Off

closed 2018-09-19 by ascillato2

aalcock, 2018-09-19:
Agreed - thanks for your help