arendst/Tasmota #3848: Restart (boot) loop

poldim, 2018-09-19:
Sonoff S31 with v6.0.0a flashed with Arduino IDE

Unit was working just fine but is now in a constant boot loop.  Below is the only output of the console that I can get before it reboots:

```
00:00:00 APP: Set Baudrate to 4800
00:00:00 Project sonoff S31 (Topic sonoff, Fallback sonoff_7D2340, GroupTopic sonoff) Version 6.0.0a-2_4_1
00:00:00 WIF: Connecting to AP1 HAN in mode 11N as Sonoff 3D Printer...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on Sonoff 3D Printer.local with IP address 192.168.3.73
```

This is the output from pinging the device:

```
64 bytes from 192.168.3.73: icmp_seq=1717 ttl=127 time=211.414 ms
64 bytes from 192.168.3.73: icmp_seq=1718 ttl=127 time=9.181 ms
64 bytes from 192.168.3.73: icmp_seq=1719 ttl=127 time=101.358 ms
Request timeout for icmp_seq 1720
Request timeout for icmp_seq 1721
Request timeout for icmp_seq 1722
Request timeout for icmp_seq 1723
Request timeout for icmp_seq 1724
Request timeout for icmp_seq 1725
Request timeout for icmp_seq 1726
Request timeout for icmp_seq 1727
Request timeout for icmp_seq 1728
Request timeout for icmp_seq 1729
Request timeout for icmp_seq 1730
Request timeout for icmp_seq 1731
Request timeout for icmp_seq 1732
64 bytes from 192.168.3.73: icmp_seq=1733 ttl=127 time=13.583 ms
64 bytes from 192.168.3.73: icmp_seq=1734 ttl=127 time=55.125 ms
Request timeout for icmp_seq 1735
Request timeout for icmp_seq 1736
Request timeout for icmp_seq 1737
Request timeout for icmp_seq 1738
Request timeout for icmp_seq 1739
Request timeout for icmp_seq 1740
Request timeout for icmp_seq 1741
Request timeout for icmp_seq 1742
Request timeout for icmp_seq 1743
Request timeout for icmp_seq 1744
Request timeout for icmp_seq 1745
Request timeout for icmp_seq 1746
64 bytes from 192.168.3.73: icmp_seq=1746 ttl=127 time=1100.480 ms
64 bytes from 192.168.3.73: icmp_seq=1747 ttl=127 time=99.297 ms
64 bytes from 192.168.3.73: icmp_seq=1748 ttl=127 time=7.608 ms
Request timeout for icmp_seq 1750
Request timeout for icmp_seq 1751
Request timeout for icmp_seq 1752
Request timeout for icmp_seq 1753
Request timeout for icmp_seq 1754
Request timeout for icmp_seq 1755
Request timeout for icmp_seq 1756
Request timeout for icmp_seq 1757
Request timeout for icmp_seq 1758
Request timeout for icmp_seq 1759
Request timeout for icmp_seq 1760
Request timeout for icmp_seq 1761
64 bytes from 192.168.3.73: icmp_seq=1762 ttl=127 time=5.626 ms
64 bytes from 192.168.3.73: icmp_seq=1763 ttl=127 time=10.764 ms
Request timeout for icmp_seq 1764
Request timeout for icmp_seq 1765
Request timeout for icmp_seq 1766
Request timeout for icmp_seq 1767
Request timeout for icmp_seq 1768
Request timeout for icmp_seq 1769
Request timeout for icmp_seq 1770
Request timeout for icmp_seq 1771
Request timeout for icmp_seq 1772
Request timeout for icmp_seq 1773
Request timeout for icmp_seq 1774
Request timeout for icmp_seq 1775
64 bytes from 192.168.3.73: icmp_seq=1776 ttl=127 time=860.301 ms
64 bytes from 192.168.3.73: icmp_seq=1777 ttl=127 time=14.463 ms
64 bytes from 192.168.3.73: icmp_seq=1778 ttl=127 time=6.061 ms
Request timeout for icmp_seq 1779
Request timeout for icmp_seq 1780
```

ascillato, 2018-09-19:
> Unit was working just fine but is now in a constant boot loop.

So, you change something (hardware, config, network) ?

Which hardware are you using?

Do you have any hardware modification or it is just an Standard Sonoff?

Have you tried to reflash it ?

Jason2866, 2018-09-19:
You use arduino core 2.4.1. It is know to have issues.
Use core 2.3.0 it is know to work well or if you dont use sleep you can use 2.4.2.

ascillato, 2018-09-19:
@Jason2866 what a sharp eye you have!

Yeap that is the problem.

Please use the precompiled bins or recompile using 2.3.0 or 2.4.2

Thanks

closed 2018-09-19 by ascillato2

poldim, 2018-10-04:
Reflashing with current 6.2.1 worked.