arendst/Tasmota #3851: Protected switch

OTiby, 2018-09-20:
Is it possible to make a code/password protected switch?

Example:
I want to have a shortcut on my android phone desktop for the gate. If I press this button/widget, a code/password is needed to open the gate.

A way can be to use an app like Tasker. But is it also possible to do it with tasmota to protect a switch or with yeti?
Or another way to prevent people without the code to open the gate?

Jason2866, 2018-09-20:
With Tasmota no. Tasmota is only a actor without securing functions
You have to secure the possibility to "switch" something in a higher level system.
Dont forget to secure the Tasmota device "mechanic" and secure your network!

ascillato, 2018-09-20:
You can add the admin password from the webmenu at configuration/configuration extra. Anyone that does not have that password can not use Tasmota.

closed 2018-09-20 by ascillato2