arendst/Tasmota #3853: Network connectivity stutters after upgrade

enesbcs, 2018-09-20:
**Describe the bug**
Upgrading Sonoff Touch from Tasmota 6.1.1 to 6.2.1 sometimes causes network connectivity to stutter. 
Pinging the device with firmware 6.2.1 gives big ping times and web interface go to veeeery slow. After downgrading to 6.1.1, everything goes smoothly, web interface is responsive. Tested with 6 Sonoff Touch device, 4 of them produces this error with 6.2.1, and 2 of them works well with 6.2.1.

Upgrading method: uploading binary through web interface

- [X] Searched the problem in issues and in the wiki
- [X] Hardware used : Sonoff Touch x 6
- [X] Development/Compiler/Upload tools used : pre-compiled binaries from tasmota webpage
- [X] Provide the output of command ``status 0`` :

STATUS 0 OUTPUT HERE (DEVICE THAT DO NOT BEHAVE WELL AFTER UPGRADE SO DOWNGRADED)
```
20:43:27 CMD: status 0
20:43:27 MQT: stat/sonoff150/STATUS = {"Status":{"Module":10,"FriendlyName":["Dolgozó világítás"],"Topic":"sonoff150","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:43:27 MQT: stat/sonoff150/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"6T02:28:28","StartupUTC":"2018-09-14T17:14:59","Sleep":0,"BootCount":23,"SaveCount":46,"SaveAddress":"FA000"}}
20:43:27 MQT: stat/sonoff150/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-23T11:18:52","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:43:27 MQT: stat/sonoff150/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["MYWIFI","backup"],"TelePeriod":0,"SetOption":["00000009","55818000"]}}
20:43:27 MQT: stat/sonoff150/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0052A794","0C000008","00261796","00000000"]}}
20:43:27 MQT: stat/sonoff150/STATUS5 = {"StatusNET":{"Hostname":"sonoff150-1978","IPAddress":"192.168.2.150","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:9B:87:BA","Webserver":2,"WifiConfig":3}}
20:43:27 MQT: stat/sonoff150/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9B87BA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:43:27 MQT: stat/sonoff150/STATUS7 = {"StatusTIM":{"UTC":"Thu Sep 20 19:43:27 2018","Local":"Thu Sep 20 20:43:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
20:43:27 MQT: stat/sonoff150/STATUS10 = {"StatusSNS":{"Time":"2018-09-20T20:43:27"}}
20:43:27 MQT: stat/sonoff150/STATUS11 = {"StatusSTS":{"Time":"2018-09-20T20:43:27","Uptime":"6T02:28:28","Vcc":3.188,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MYWIFI","RSSI":92,"APMac":"A0:F3:C1:BF:AC:89"}}}
```

STATUS 0 OUTPUT HERE (DEVICE THAT WORKS OK AFTER UPGRADE)
```
20:49:58 CMD: status 0
20:49:58 MQT: stat/sonoff151/STATUS = {"Status":{"Module":10,"FriendlyName":["KONYHA Mennyezeti Vilagitas"],"Topic":"sonoff151","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":0,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
20:49:58 MQT: stat/sonoff151/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.100/sonoff.bin","RestartReason":"Power on","Uptime":"7T18:57:46","StartupUTC":"2018-09-13T00:52:12","Sleep":50,"BootCount":99,"SaveCount":1024,"SaveAddress":"FB000"}}
20:49:58 MQT: stat/sonoff151/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:48:36","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:49:58 MQT: stat/sonoff151/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["MYWIFI","backup"],"TelePeriod":0,"SetOption":["54000008","55818000","00000001"]}}
20:49:58 MQT: stat/sonoff151/STATUS4 = {"StatusMEM":{"ProgramSize":475,"Free":528,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","6050A790","000003A8","210217C6","00000000"]}}
20:49:58 MQT: stat/sonoff151/STATUS5 = {"StatusNET":{"Hostname":"sonoff151-4353","IPAddress":"192.168.2.151","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:85:B1:01","Webserver":2,"WifiConfig":4}}
20:49:58 MQT: stat/sonoff151/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.100","MqttPort":1883,"MqttClientMask":"KONYHA-VIL1","MqttClient":"KONYHA-VIL1","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:49:58 MQT: stat/sonoff151/STATUS7 = {"StatusTIM":{"UTC":"Thu Sep 20 19:49:58 2018","Local":"Thu Sep 20 20:49:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
20:49:58 MQT: stat/sonoff151/STATUS10 = {"StatusSNS":{"Time":"2018-09-20T20:49:58"}}
20:49:58 MQT: stat/sonoff151/STATUS11 = {"StatusSTS":{"Time":"2018-09-20T20:49:58","Uptime":"7T18:57:46","Vcc":3.196,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"MYWIFI","RSSI":74,"APMac":"A0:F3:C1:BF:AC:89"}}}
```

Frogmore42, 2018-09-20:
Follow the steps in the wiki.  You sometimes have erase the flash to get things stable.  It is also good to indicate what kind of a network environment your devices are in, ie how many networks are visible and how clear is the channel yours is on.

ascillato, 2018-09-20:
Also, please try the development builds at http://thehackbox.org/tasmota/

enesbcs, 2018-09-20:
The network contains a TP-Link TL-WDR3600 v1 router with OpenWRT, 2PC at 5Ghz, and the rest is on 2.4Ghz Wifi 4xRaspberry PI, 10xSonoff device, 3xNodemcu all in the same Lan. If channels not clear, than other devices should also be bad, and downgrade should not help, i think.
Tried with development builds from last week without success, i will try again a new one. All of my Sonoff Touch-es mounted in the wall, and the last thing that I want to do is to disassemble&upgrade them with serial. If noone else experienced this specific problem, then maybe there is a flash related problem in the background. I tried with a spare Sonoff T1 on my desk and it worked...

enesbcs, 2018-09-20:
> Also, please try the development builds at http://thehackbox.org/tasmota/
I tried the current 20180920 build on a problematic Touch switch, and it seems to work! Version displayed is 6.2.1.4. Ping seems OK, web gui is responsive. I will try the others tomorrow, and i'll write the results.

Frogmore42, 2018-09-21:
Just because something works with one or more devices does not mean it will work with all.  Wi-Fi and RF is highly dependent on the exact signal and subtle changes can make it go from working great to not at all.  While generally the Wi-Fi stack works well on the esp8266, there have been many iterations on it and some of them worked excellently in many places and poorly or not at all in others.  I suspect that there are still certain conditions that cause things to work poorly.

enesbcs, 2018-09-21:
Anyway I can confirm, that all problematic devices upgraded to 6.2.1.4 and all works OK. Ping time is normal, and web GUI is responsive.

closed 2018-09-21 by enesbcs

Sym-Link, 2018-09-21:
I was on 6.2.1 and was having WiFi issues as well with some of mine. 4 out 17 were not reconnecting after flash. On serial power they seemed a little better as far as connecting and staying connected. I ended going back down to 6.1.1 and things were much better. I then downgraded all of my others to 6.1.1 as well.

This morning I flashed the latest 09/21 dev builds, 6.2.1.5 and it seems good too. The only problem I had with the dev builds is that two of my Sonoffs are setup for KNX. I noticed the dev KNX build is a bit smaller than the release one but I went ahead and flashed. They came up fine but KNX wasn't working so I went back down to 6.1.1 on those two.