arendst/Tasmota #3856: Rule on power#state after software restart

KPK75, 2018-09-21:
After software restart first event from rule on power#state doesn't fire.

To Reproduce define rule
`on power1#state=0 do event test=0 endon on power1#state=1 do event test=1 endon`

Expected behavior:
It should work like after power on.

`0:00:06 RSL: INFO1 = {"Module":"Generic","Version":"6.2.1.4","FallbackTopic":"DVES_04C812","GroupTopic":"sonoffs"}
00:00:06 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2066","IPAddress":"172.17.10.117"}`
`00:00:06 RSL: INFO3 = {"RestartReason":"External System"}`
`00:00:06 RSL: RESULT = {"POWER":"OFF"}`
`00:00:06 RSL: POWER = OFF`
`10:30:38 RSL: STATE = {"Time":"2018-09-21T10:30:38","Uptime":"0T00:00:15","Vcc":3.442,"POWER":"OFF","Wifi":{"AP":2,"SSId":"DI","BSSId":"00:1A:73:30:62:52","Channel":1,"RSSI":88}}`
`10:30:51 CMD: power on`
`10:30:51 RSL: RESULT = {"POWER":"ON"}`
`10:30:51 RSL: POWER = ON`
`10:30:51 RUL: POWER1#STATE=1 performs "backlog event test=1"`
`10:30:52 RSL: RESULT = {"Event":"Done"}`

Now command "restart 1"

`00:00:06 RSL: INFO1 = {"Module":"Generic","Version":"6.2.1.4","FallbackTopic":"DVES_04C812","GroupTopic":"sonoffs"}
00:00:06 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2066","IPAddress":"172.17.10.117"}`
`00:00:06 RSL: INFO3 = {"RestartReason":"Software/System restart"}`
`00:00:06 RSL: RESULT = {"POWER":"OFF"}`
`00:00:06 RSL: POWER = OFF`
`10:33:17 RSL: STATE = {"Time":"2018-09-21T10:33:17","Uptime":"0T00:00:15","Vcc":3.442,"POWER":"OFF","Wifi":{"AP":2,"SSId":"DI","BSSId":"00:1A:73:30:62:52","Channel":1,"RSSI":88}}`
**`10:33:40 CMD: power on`
`10:33:40 RSL: RESULT = {"POWER":"ON"}`
`10:33:40 RSL: POWER = ON`**
`10:33:54 CMD: power off`
`10:33:54 RSL: RESULT = {"POWER":"OFF"}`
`10:33:54 RSL: POWER = OFF`
`10:33:54 RUL: POWER1#STATE=0 performs "backlog event test=0"`
`10:33:54 RSL: RESULT = {"Event":"Done"}`

So it missed first change.

ascillato, 2018-09-21:
Hi,

Rules are triggers, so they work with changes only.

At boot time your can use the trigger SYSTEM#BOOT or TIME#INITIALIZED

If you need more information, please find us in the Discord chat.

Thanks.

KPK75, 2018-09-21:
No. You missunderstood. There is a change, look in the result ( state is changed from off to on )
After power on rule is working fine but the same rule after restart doesn't work ( no event ) for the first change.
So it's not a question but a bug :)

ascillato, 2018-09-21:
Hi,

Tested and works as expected. Sorry. I could not reproduce it. Have you set rule set at once? can you share your status 0 and the output of rule?

```
00:00:00 CFG: Cargado desde flash en 3F5, Conteo 237
00:00:00 APP: Conteo Reinicios 125
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.4-2_4_2
00:00:00 CFG: Grabado a la flash en 3F4, Conteo 238, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:06 MQT: Intentando conectar...
00:00:07 APP: (UTC) Fri Sep 21 13:39:14 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
10:39:16 MQT: Conectado
10:39:16 MQT: tele/living/LWT = Online (Grabado)
10:39:16 MQT: cmnd/living/POWER = 
10:39:16 MQT: Suscribir a cmnd/living/#
10:39:16 MQT: Suscribir a cmnd/sonoffs/#
10:39:16 MQT: Suscribir a cmnd/DVES_591719/#
10:39:16 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.2.1.4","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
10:39:16 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
10:39:16 MQT: tele/living/INFO3 = {"RestartReason":"Software/System restart"}
10:39:16 MQT: stat/living/RESULT = {"POWER1":"ON"}
10:39:16 MQT: stat/living/POWER1 = ON
10:39:19 APP: Botón1 multi-press 1
10:39:20 MQT: cmnd/living/POWER = OFF
10:39:21 SRC: MQTT
10:39:21 RSL: Topic Recibido cmnd/living/POWER, Tamaño de Datos 3, Datos OFF
10:39:21 RSL: Grupo 0, Índice 1, Comando POWER, Datos OFF
10:39:21 MQT: stat/living/RESULT = {"POWER1":"OFF"}
10:39:21 MQT: stat/living/POWER1 = OFF
10:39:21 RUL: POWER1#STATE=0 performs "backlog event test=0"
10:39:21 SRC: Rule
10:39:21 RSL: Topic Recibido /backlog, Tamaño de Datos 12, Datos event test=0
10:39:21 RSL: Grupo 0, Índice 1, Comando BACKLOG, Datos event test=0
10:39:22 SRC: Backlog
10:39:22 RSL: Topic Recibido /event, Tamaño de Datos 6, Datos test=0
10:39:22 RSL: Grupo 0, Índice 1, Comando EVENT, Datos test=0
10:39:22 MQT: stat/living/RESULT = {"Event":"Done"}
10:39:22 CFG: Grabado a la flash en 3FB, Conteo 239, Bytes 3584
10:39:25 MQT: tele/living/STATE = {"Time":"2018-09-21T10:39:25","Uptime":"0T00:00:18","Vcc":2.993,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":60}}
```

ascillato, 2018-09-21:
Hi,

Yes, I can try that. Of course.

Turned Off MQTT. And it works as expected.

Output after `restart 1` command:

```
00:00:00 CFG: Cargado desde flash en 3F8, Conteo 255
00:00:00 APP: Conteo Reinicios 132
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.5-2_4_2
00:00:00 CFG: Grabado a la flash en 3F7, Conteo 256, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:05 RSL: INFO1 = {"Module":"Generic","Version":"6.2.1.5","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
00:00:05 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 RSL: RESULT = {"POWER1":"OFF"}
00:00:05 RSL: POWER1 = OFF
00:00:07 APP: (UTC) Fri Sep 21 16:58:42 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
13:58:46 APP: Botón1 multi-press 1
13:58:46 SRC: Button
13:58:46 RSL: RESULT = {"POWER1":"ON"}
13:58:46 RSL: POWER1 = ON
13:58:46 RUL: POWER1#STATE=1 performs "backlog event test=1"
13:58:46 SRC: Rule
13:58:46 RSL: Topic Recibido /backlog, Tamaño de Datos 12, Datos event test=1
13:58:46 RSL: Grupo 0, Índice 1, Comando BACKLOG, Datos event test=1
13:58:46 SRC: Backlog
13:58:46 RSL: Topic Recibido /event, Tamaño de Datos 6, Datos test=1
13:58:46 RSL: Grupo 0, Índice 1, Comando EVENT, Datos test=1
13:58:46 RSL: RESULT = {"Event":"Done"}
13:58:47 CFG: Grabado a la flash en 3F6, Conteo 257, Bytes 3584
13:58:48 RSL: STATE = {"Time":"2018-09-21T13:58:48","Uptime":"0T00:00:13","Vcc":2.993,"POWER1":"ON","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":46}}
13:58:48 APP: Botón1 multi-press 1
13:58:49 SRC: Button
13:58:49 RSL: RESULT = {"POWER1":"OFF"}
13:58:49 RSL: POWER1 = OFF
13:58:49 CFG: Grabado a la flash en 3F5, Conteo 258, Bytes 3584
13:58:49 RUL: POWER1#STATE=0 performs "backlog event test=0"
13:58:49 SRC: Rule
13:58:49 RSL: Topic Recibido /backlog, Tamaño de Datos 12, Datos event test=0
13:58:49 RSL: Grupo 0, Índice 1, Comando BACKLOG, Datos event test=0
13:58:49 SRC: Backlog
13:58:49 RSL: Topic Recibido /event, Tamaño de Datos 6, Datos test=0
13:58:49 RSL: Grupo 0, Índice 1, Comando EVENT, Datos test=0
13:58:49 RSL: RESULT = {"Event":"Done"}
```

Have you modified your version of Tasmota?

ascillato, 2018-09-21:
Great Job!!!! :+1: You find it!

I can reproduce that !

```
setoption0 0
rule on power1#state=0 do event test=0 endon on power1#state=1 do event test=1 endon
rule 1
restart 1
```

then the first trigger of the power toggle is not catched by rules.

ascillato, 2018-09-21:
If changing your rule to:

```
rule on system#boot do power 0 endon on power1#state=0 do event test=0 endon on power1#state=1 do event test=1 endon
```

it will be triggered also the first one if `setoption0 0`

KPK75, 2018-09-21:
Yes. But the relay state won't be preserved after restart as it hapens without on system#boot rule.

ascillato2, 2018-09-22:
@arendst 

Confirmed,

Bug Solved. :+1: 

Thanks Theo!

closed 2018-09-22 by ascillato2