arendst/Tasmota #3857: WiFi connection status light turns off when switch is on ?

L00000813, 2018-09-21:
T1 UK 2 gang switch connected to Wi-Fi and MQTT. 

Wi-Fi light is on constant showing connection as expected however If either gang is switched on the Wi-Fi light goes off until the gang is switched off again. 

Has anyone experienced this and no a solution?

11:01:16 CMD: status 0
11:01:16 MQT: stat/A2/STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"A2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:01:16 MQT: stat/A2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:13","StartupUTC":"2018-09-21T09:59:03","Sleep":0,"BootCount":1022,"SaveCount":1108,"SaveAddress":"F8000"}}
11:01:16 MQT: stat/A2/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-18T10:27:10","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:01:16 MQT: stat/A2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["eir46325537-2.4G","Wi-Fi-Raspberry"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
11:01:16 MQT: stat/A2/STATUS4 = {"StatusMEM":{"ProgramSize":509,"Free":492,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAEB94","000003A0","23B617CE","000002C0"]}}
11:01:16 MQT: stat/A2/STATUS5 = {"StatusNET":{"Hostname":"A2-5980","IPAddress":"192.168.1.4","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:81:F7:5C","Webserver":2,"WifiConfig":2}}
11:01:16 MQT: stat/A2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_81F75C","MqttUser":"","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:01:16 MQT: stat/A2/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 21 10:01:16 2018","Local":"Fri Sep 21 11:01:16 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:35","Sunset":"18:51"}}
11:01:16 MQT: stat/A2/STATUS10 = {"StatusSNS":{"Time":"2018-09-21T11:01:16"}}
11:01:16 MQT: stat/A2/STATUS11 = {"StatusSTS":{"Time":"2018-09-21T11:01:16","Uptime":"0T00:02:13","Vcc":3.457,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"eir46325537-2.4G","BSSId":"50:01:D9:11:E0:10","Channel":9,"RSSI":66}}}

Jason2866, 2018-09-21:
Checked command LedState and LedPower?
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

L00000813, 2018-09-21:
The messages that I am passing through the Web App  "Toggle" are:

11:35:26 MQT: stat/A2/POWER1 = OFF
11:35:56 MQT: stat/A2/RESULT = {"POWER1":"ON"}
11:35:56 MQT: stat/A2/POWER1 = ON

This is turning the Gang (switch on/off and that gange LED is going on/off as expected.

The Wi-Fi connection light also goes on/off when a gang (switch is toggled. This is an error either on my part or otherwise? Toggling the switch has nothing to do with the Wi-Fi connection and therefore it should not toggle the Wi-FI connection status light.

[image]
[image]

ascillato, 2018-09-21:
Hi,

Please, check the commands in the wiki to set the behaviour you want for the Led and the wifi (ledstate, ledpower and SetOption31)

If all those don't fit what you want, you can set the behaviour you want using rules.

Thanks.

closed 2018-09-21 by ascillato2

L00000813, 2018-09-21:
I do not understand what you are telling me to do. 

I am simply turning on/off a switch using Tasmota and the Wi-Fi light is going on and off when it SHOULD NOT.

What can I change in the source code to fix this?

ascillato, 2018-09-21:
> I do not understand what you are telling me to do.

Go to the Tasmota console and use the commands (`ledstate`, `ledpower `and `SetOption31`)

The following is a COPY and PASTE from the WIKI at COMMANDS (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) for you to know what each command do:

Command | Payload | Description
-- | -- | --
LedPower |   | Show current led power state as On or Off
LedPower | 0 / off | Turn led AND LedState Off
LedPower | 1 / on | Turn led On AND LedState Off
LedState |   | Show current led state as 0 to 7
LedState | 0 / off | Disable use of LED as much as possible
LedState | 1 / on | Show power state on led (inverted for Sonoff Touch)
LedState | 2 | Show MQTT subscriptions as a led blink
LedState | 3 | Show power state and MQTT subscriptions as a led blink
LedState | 4 | Show MQTT publications as a led blink
LedState | 5 | Show power state and MQTT publications as a led blink
LedState | 6 | Show all MQTT messages as a led blink
LedState | 7 | Show power state and MQTT messages as a led blink
SetOption31 | 1 / on | Disable wifi (once every second) and mqtt (once every 2 seconds) status information blink of wifi led

And **_Rules_** is a **Feature of Tasmota** to add/modify the actual behaviours in order to do what you want.

See all the related information and examples in the WIKI at RULES (https://github.com/arendst/Sonoff-Tasmota/wiki/Rules)

> What can I change in the source code to fix this?

There is no need to modify the code because a lot of different options are already implemented for making the behaviour you want.

If you need more help, please find us in Tasmota Discord Chat.

L00000813, 2018-09-21:
Turning a switch (gang) on or off should not affect the Wi-Fi LED connection status.
I know how to manually turn on/off the LED from the Wiki, but why should I have to manually change the Wi-Fi LED light every time I change the switch from on to off. 
The Switch is for a relay and the Wi-Fi LED light I am having the issue with indicates Wi-Fi and MQTT connection. 
They are two separate things! One should not interfere with the other. This is a genuine issue - why have you closed it? The person who opens an issue should be the one to close it when confirmation of a solution is found. 
Please reopen this issue should it can be looked at and resolved.

Jason2866, 2018-09-21:
Lets talk about here https://discord.gg/Ks2Kzd4