arendst/Tasmota #3858: Ping vs. indicator.connected

daniel-lo, 2018-09-21:
I have 2 Sonoff 4ch-Pro-R2 devices which I monitor via a ping. What I notice is the following: Quite frequently the devices are not reachable via ping. The outage time is always 60/61 seconds. And even though the pings fail "indicator.connected" stays true.

My assumption is that the devices will operate even though pings fail but nevertheless it might be good to look into this from the Tasmota perpective.

I use blockly scripts in ioBroker.
Beware .. I am no dev.

Jason2866, 2018-09-21:
Please fill out issue template. Without no help is possible. `Status 0` is mandantory

daniel-lo, 2018-09-21:
[image]

ascillato, 2018-09-21:
Hi,

Tasmota is a MQTT device, you can use the LWT to see if it is online or not.

daniel-lo, 2018-09-21:
Sorry .. I don't know what LWT is. Isn't that the indicator.connected state I see in the sonoff adapter on iobroker?

ascillato, 2018-09-21:
LWT is the online/offline indicator in the MQTT protocol. If you have configured all the parameters as explained in the wiki, you will have the online/offline in your MQTT Broker.

Jason2866, 2018-09-21:
Yes, but seeing is device online or not is the second step...
Why does the device goes offline?!  For that we need `status 0` output.
We can track down (hopefully) the reason for going offline

ascillato, 2018-09-21:
I don't know how to configure IOBroker. I use HomeAssistant and the config there is like:

```yaml
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false
```

so, there you see the `availability_topic `that is watching the LWT

Jason2866, 2018-09-21:
@ascillato 
If you have wifi issues you wont receive anything

daniel-lo, 2018-09-21:
I don't have wifi issues as such.

this is what i have in iobroker:

[image]

Jason2866, 2018-09-21:
Are you 100% sure your sonoff device has no wifi issues?
I ask because there were MANY problems not long ago....

daniel-lo, 2018-09-21:
Well .. not 100% sure .. but I'd expect "indicator.connected" to go down too. I think now I understand what you mean with Status 0 .. will provide asap.

daniel-lo, 2018-09-21:
`14:07:38 CMD: status 0
14:07:38 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":["sonoff-garage1","sonoff-garage2","sonoff-garage3","sonoff-garage4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
14:07:38 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T07:09:58","StartupUTC":"2018-09-21T04:57:40","Sleep":0,"BootCount":30,"SaveCount":63,"SaveAddress":"FA000"}}
14:07:38 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-05T10:41:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:07:38 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
14:07:38 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","6050A790","24000008","210217C6","00000000"]}}
14:07:38 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-garage","IPAddress":"192.168.178.62","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"60:01:94:A1:9F:89","Webserver":2,"WifiConfig":3}}
14:07:38 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.30","MqttPort":1883,"MqttClientMask":"sonoff-garage","MqttClient":"sonoff-garage","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:07:38 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 21 12:07:38 2018","Local":"Fri Sep 21 14:07:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
14:07:38 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-21T14:07:38"}}
14:07:38 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-21T14:07:38","Uptime":"0T07:09:58","Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box","RSSI":10,"APMac":"34:31:C4:7D:FD:2C"}}}`

ascillato2, 2018-09-21:
@daniel-lo 

If you are having disconnections, you are having wifi issues. My devices never lose a ping.

If what you want is just to know if a device is online or not, you can use the LWT of the MQTT broker.

The `Status 0` is a command to type in the console and that the output give us all your config and versions to do some troubleshooting. (it dont contain password.)

daniel-lo, 2018-09-21:
maybe more legible:

https://pastebin.com/qnth53eU

daniel-lo, 2018-09-21:
I have two devices connected to the same repeater (already tried another repeater) .. and both have ping-outages at different times. One device a lot more often than the other.

ascillato2, 2018-09-21:
Please, try the last version from http://thehackbox.org/tasmota/

If you have downloaded the v6.2.1 some days ago, that had wifi problems, the new one is fixed. But it is better to test directly the last version.

daniel-lo, 2018-09-21:
Will look at that. Hmm .. do I have 1M or 4M?
What I wanted to add is that the former version also had this.

ascillato2, 2018-09-21:
Oh repeaters.... ufff

So there isn't much to do. The wifi libraries that uses the esp8266 based devices have some issues with some brands of repeaters.

That is outside the control of Tasmota.

May be in the future Arduino and Espressif SDK could provide a wider compatible version.

Just to narrow the issue. Please, try to upgrade your device to the last version in http://thehackbox.org/tasmota/

1M

ascillato2, 2018-09-21:
if with the last version, you still have ping problems, a workaround, that had work for a lot of users, is to connect your Sonoff directly to your router instead to a repeater

daniel-lo, 2018-09-21:
I've updated .. will now have to wait a bit. Connecting directly to the router will be possible with 1 of the devices .. the other is too far away. But .. I'll do that next. Thanks.

ascillato2, 2018-09-21:
> I've updated .. will now have to wait a bit.

Ok, I think will be more dynamic to chat about this at the Tasmota Discord Chat (https://discord.gg/Ks2Kzd4)

See you there.

closed 2018-09-21 by ascillato2

Jason2866, 2018-09-21:
"RSSI":10  !!!!   That is horrible. You have a wifi problem. If you dont change the repeater it will never work reliable!

ascillato, 2018-09-21:
@Jason2866 again you have surprised me with your precise eye :+1:

daniel-lo, 2018-09-21:
AP1 SSId (RSSI) | FRITZ!Box (100%)

daniel-lo, 2018-10-15:
Final update: Switched from Raspberry to Intel NUC and all problems are gone. I have no real idea why.