arendst/Tasmota #3859: 2 events or 2 variables and one URL/HTTP command possible?

Kiritzai, 2018-09-21:
I'm wondering if this is even possible what i'm trying to accomplish, cause im unable to find anything on wiki or google. What i'm trying to do is make 1 URL that can change 2 variables at a time with 1 or 2 events.

### 1 Event
```
rule on event#doorbell do var1 %var1%; var2 %var2% endon
```
HTTP Call
```
http://<tasmotaIP>/cm?cmnd=event%20doorbell=1&34
```

### 2 Events
```
rule on event#doorbell do var1 %var1% endon on event#idx do var2 %var2% endon
```
HTTP Call (IDX number is just a random number):
```
http://<tasmotaIP>/cm?cmnd=event%20doorbell=1&event%20idx=34
```

When I type var2 in console it's still empty for both events and doesn't change it, but it does change for var1.

ascillato, 2018-09-21:
Hi,

Sorry, Tasmota only supports 1 parameter.

closed 2018-09-21 by ascillato2

Kiritzai, 2018-09-21:
Then I would like to turn this into a feature request. ;-)