arendst/Tasmota #3860: How to calibrate MH-Z19?

ThinkPadNL, 2018-09-21:
Hi,

I am using MH-Z19B CO2-sensor. I bought a second one for my bedroom and i see that it is a bit off from the one i was already using. I would like to calibrate it to 400 ppm. In the sensor code i see the command is available: https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_15_mhz19.ino#L81

But how do i access that command from the webgui console for example? Or via some sort of JSON URL, so i can schedule it (crontab) to run at night when the room is empty.

Jason2866, 2018-09-21:
No need for:
`/*********************************************************************************************\
 * Source: http://www.winsen-sensor.com/d/files/infrared-gas-sensor/mh-z19b-co2-ver1_0.pdf
 *
 * Automatic Baseline Correction (ABC logic function)
 *
 * ABC logic function refers to that sensor itself do zero point judgment and automatic calibration procedure
 * intelligently after a continuous operation period. The automatic calibration cycle is every 24 hours after powered on.
 *
 * The zero point of automatic calibration is 400ppm.
 *
 * This function is usually suitable for indoor air quality monitor such as offices, schools and homes,
 * not suitable for greenhouse, farm and refrigeratory where this function should be off.
 *
 * Please do zero calibration timely, such as manual or commend calibration.
\*********************************************************************************************/

#define MHZ19_ABC_ENABLE             1       // Automatic Baseline Correction (0 = off, 1 = on (default))

/*********************************************************************************************/`

Jason2866, 2018-09-21:
I can say after a few month of use this selfcalibration does work.
You have to open window or door open a long time to get 400ppm !

ThinkPadNL, 2018-09-21:
I read that, but what if i want to do it myself anyway?

Jason2866, 2018-09-21:
Code has to be changed for that. Not implemented

ThinkPadNL, 2018-09-21:
OK. I will put it near a open window for 24h and see what that brings.

ascillato, 2018-09-21:
I'm just asking because there is nothing in the wiki and I don't have this sensor to tes.

the command MHZ_CMND_ZEROPOINT does not work from the console?

arifintajul4, 2020-06-01:
> Hi,
>
> I am using MH-Z19B CO2-sensor. I bought a second one for my bedroom and i see that it is a bit off from the one i was already using. I would like to calibrate it to 400 ppm. In the sensor code [...]

The repository is 404 not found

ascillato, 2020-06-02:
Sorry, you are reading a very old and obsolete issue (2 years!!!). 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)