arendst/Tasmota #3861: RFBridge reports different codes after update from 5.7.1 to 6.2.1

sgofferj, 2018-09-21:
To have more functions, I don't use the 16 memories of the RF bridge but process the MQTT rfreceived message in Node RED. Today I ran the upgrade path from 5.7.1 to 6.2.1 via 5.14.1 and now the RFbridge reports completely different codes. I have a bunch of switches, door sensors, hand remotes, all based on the PT2262 chip. Under 5.7.1, the RFBridge reported 6-digit hexadecimal codes for all of them. The codes also made sense based on the wiring of the transmitters.
Now with 6.2.1, codes are reported differently, as 6 to 8 digit long numbers (not sure which format but at least without any A-F in them).

I report this as a bug as I'm not sure if there was an intended change in the firmware, an unintended change or if the codes are going to be reported differently after each update now, e.g. because of some timing issue.
In any case, the six digit hex codes made the most sense as they corresponded directly to the wiring / jumpers of the transmitter.

ascillato, 2018-09-21:
Hi,

Could be that you have set to see your codes as DEC instead of HEX by `SetOption28 `?

Have you look at the wiki at RF Codes and Commands? https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-rf-bridge-433

Please, try that command.

Thanks.

closed 2018-09-21 by ascillato2

sgofferj, 2018-09-21:
I haven't consciously set that but I will check.

sgofferj, 2018-09-21:
Jup, that was it. Was the option newly introduced between 5.7.1 and 6.2.1?

ascillato, 2018-09-21:
Yes, and a lot of new options and features since 5.7.1. (1 year old)

See [changelog](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino) for more information.

sgofferj, 2018-09-21:
Might I suggest that on upgrade the option defaults to HEX - which is the old behavior? So people don't run into unexpected troubles on upgrade,

ascillato, 2018-09-21:
When upgrading your firmware, all your config is retained in your device. There is no need to do anything else. All your config will be kept in your Sonoff.

Your version was very old. That command was not implemented in your version. So upgrading, Tasmota uses a the value that was stored in the new memory location. In your case is 0, so it will display setoption28 to 0.

So, your issue was not a bug. Just a configuration.