arendst/Tasmota #3862: Magic Home - decoding IR codes over Web

autofrank, 2018-09-21:
Hi,
I can see the magic home receiving the IR command

`19:44:37 RSL: RESULT = {"IrReceived":{"PROTOCOL":"NEC","BITS":32,"DATA":"FF9867"}}
19:44:52 RSL: RESULT = {"Command":"Unknown"}`

But am not sure how to retrieve this over web
The full Status json response doesn't seem to have the data.

`{
  "Status": {
    "Module": 18,
    "FriendlyName": "spare 182",
    "Topic": "MagicLed1",
    "ButtonTopic": "0",
    "Power": 1,
    "PowerOnState": 3,
    "LedState": 1,
    "SaveData": 1,
    "SaveState": 1,
    "ButtonRetain": 0,
    "PowerRetain": 0
  },
  "StatusPRM": {
    "Baudrate": 115200,
    "GroupTopic": "sonoffs",
    "OtaUrl": "http:\/\/domus1:80\/api\/arduino\/MagicLed1.ino.bin",
    "Uptime": "0T00:11:32",
    "Sleep": 0,
    "BootCount": 12,
    "SaveCount": 55,
    "SaveAddress": "FB000"
  },
  "StatusFWR": {
    "Version": "5.12.0",
    "BuildDateTime": "2018-09-21T18:03:06",
    "Boot": 7,
    "Core": "2_4_0",
    "SDK": "2.1.0(deb1901)"
  },
  "StatusLOG": {
    "SerialLog": 2,
    "WebLog": 2,
    "SysLog": 0,
    "LogHost": "domus1",
    "LogPort": 514,
    "SSId1": "SonoffStarWifi",
    "SSId2": "",
    "TelePeriod": 300,
    "SetOption": "00008001"
  },
  "StatusMEM": {
    "ProgramSize": 521,
    "Free": 480,
    "Heap": 21,
    "ProgramFlashSize": 1024,
    "FlashSize": 1024,
    "FlashMode": 3
  },
  "StatusNET": {
    "Hostname": "MagicLed1-7918",
    "IPAddress": "192.168.1.182",
    "Gateway": "192.168.1.1",
    "Subnetmask": "255.255.255.0",
    "DNSServer": "192.168.1.1",
    "Mac": "DC:4F:22:C0:FE:EE",
    "Webserver": 2,
    "WifiConfig": 2
  },
  "StatusTIM": {
    "UTC": "Fri Sep 21 18:39:36 2018",
    "Local": "Fri Sep 21 19:39:36 2018",
    "StartDST": "Sun Mar 25 02:00:00 2018",
    "EndDST": "Sun Oct 28 03:00:00 2018",
    "Timezone": 1
  },
  "StatusSNS": {
    "Time": "2018-09-21T19:39:36"
  },
  "StatusSTS": {
    "Time": "2018-09-21T19:39:36",
    "Uptime": "0T00:11:32",
    "Vcc": 3.44,
    "POWER": "ON",
    "Dimmer": 7,
    "Color": "000011",
    "Scheme": 0,
    "Fade": "OFF",
    "Speed": 1,
    "LedTable": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "SonoffStarWifi",
      "RSSI": 46,
      "APMac": "06:18:D6:E1:DD:3D"
    }
  }
}`

Is there a different web command I can use to capture the IR code. I am looking to match the IR code to an action in my HA system

Thanks
_af

ascillato, 2018-09-21:
Hi,

Tasmota is MQTT based. You need to set a MQTT broker in your HA system in order to receive all the status.

closed 2018-09-21 by ascillato2

autofrank, 2018-09-21:
Thanks @ascillato 
The HA system I use is the Fibaro HC2 and it doesn't have native MQTT support.
I interact with all my sonoff devices over web.

Do you know if this is possible ?

ascillato, 2018-09-21:
Yes, It is possible, as my previous comment: you will need to set a rule to re send that by web. (`websend `command) For more information about that, please see the wiki.

autofrank, 2018-09-21:
Thanks @ascillato 
I'll take a read off the websend command

ascillato, 2018-09-21:
And also the rules feature. You will need both.

ascillato, 2018-09-21:
Also search in the issues about magichome that there are examples of people using rules

autofrank, 2018-09-21:
@ascillato 
Will do - thanks again for the help....