arendst/Tasmota #3868: Incorrect wiring information

pmknowles, 2018-09-22:
The information on this page is incorrect
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs
**Connect jack**
Instead of connecting a switch, you could connect a 4-pin 2.5mm jack, with the pins wired:
tip pin5 (GPIO14)
r1 no connection
r2 pin1 (3.3v)
r3 pin4 (ground)
You can then plug a sensor into the jack like you would to a Sonoff TH10/TH16 and define what sensor you have connected to GPIO14
According to https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations
Ring 3 is 3.3V
And
Ring 2 is Ground

Frogmore42, 2018-09-22:
You know you can edit the page to fix it?
Just tap/click the edit button in the upper right.

pmknowles, 2018-09-22:
Yes, but always worry if the other person's data was correct at one point and that gets lost.

arendst, 2018-09-22:
Before making a change verify the actual situation. Copying without verification is useless.

In this case find out what is the correct connection and then update pls.

Frogmore42, 2018-09-22:
The TH10/16 does indeed have 3v3 on the sleeve, and gnd on the ring next to it.  I smoked (literally) a SHT15 sensor because I didn't double check this and thought who would NOT put gnd on the sleeve.

Frogmore42, 2018-09-22:
The jack is a TRRS one.  Tip, Ring, Ring, Sleeve.  2.5mm.
I prefer the 3.5mm size as it is much easier to get decent quality parts.  All of the 2.5mm ones are of very poor quality.

pmknowles, 2018-09-22:
Have amended the page

closed 2018-09-22 by pmknowles