arendst/Tasmota #3869: Sonoff Basic R2 current-protection triggers on usb-ttl

Dees7, 2018-09-22:
**Describe the bug**
I recived 2 Sonoff Basic. On PCB printed:
```
Sonboff Basic R2 
V1.0
2017-10-11 1816
```
There is firmware 1.6 installed onto Sonoff so i cant use [SonOTA](https://github.com/mirko/SonOTA).
I tied installing via usb-ttl. It flashed good. After this Sonoff boots one time normal. 
The second time there is no com-port output.
I tried reconnect Vcc to sonoff but now usb-ttl current-protection disables usb-ttl (as short circuit). 
But there is no short circuit. 

Sonoff turns on normal and connects o WiFi. But now I can't change Module type.
I can changing other settings  normally.

Both of mine Sonoff Basic behave the same way.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : 
Sonboff Basic R2, NodeMCU as USB-TTL
- [x] _Development/Compiler/Upload tools used_ :
Arduino IDE, ESP Core 2.3.0
- [x] _Provide the output of command_ :
```
stat/bedroom/STATUS = {"Status":{"Module":18,"FriendlyName":["bedroom"],"Topic":"bedroom","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/bedroom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:45","StartupUTC":"2018-09-22T07:52:59","Sleep":0,"BootCount":20,"SaveCount":37,"SaveAddress":"F7000"}}
stat/bedroom/STATUS2 = {"StatusFWR":{"Version":"6.1.1.12","BuildDateTime":"2018-09-21T21:55:29","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/bedroom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"172.16.95.3","LogPort":514,"SSId":["195","95"],"TelePeriod":60,"SetOption":["00008009","55818000","00000000"]}}
stat/bedroom/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8827B4","000003A0","25021796","00000000"]}}
stat/bedroom/STATUS5 = {"StatusNET":{"Hostname":"bedroom","IPAddress":"172.16.95.90","Gateway":"172.16.95.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"DC:4F:22:00:00:00","Webserver":2,"WifiConfig":5}}
stat/bedroom/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.95.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0000","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/bedroom/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 22 07:55:44 2018","Local":"Sat Sep 22 12:55:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":5,"Sunrise":"10:36","Sunset":"22:49"}}
stat/bedroom/STATUS10 = {"StatusSNS":{"Time":"2018-09-22T12:55:44","BME280":{"Temperature":20.8,"Humidity":46.7,"Pressure":982.5},"TempUnit":"C"}}
stat/bedroom/STATUS11 = {"StatusSTS":{"Time":"2018-09-22T12:55:44","Uptime":"0T00:02:45","Vcc":3.116,"POWER":1,"Wifi":{"AP":1,"SSId":"195","RSSI":60,"APMac":"C8:D3:A3:00:00:00"}}}
```

**Additional context**
There is some strange parameter "Programmer" in menu
[image]

Frogmore42, 2018-09-22:
You need a computer that is capable of providing enough current, or use a powered USB hub, or provide power to the Sonoff with a 3v3 supply, not from your computer.

What computer are you using?

Dees7, 2018-09-22:
I have trouble only if connect rx\tr. In case Vcc\GND connected sonoff boots well.
I use laptop Sony Vaio and it works well with NodeMCU proper and Sonoff Dual R2 (with NodeMCU as usb-ttl).
I tested FTDI232 3.3V (current-protection triggers too). I have burned out noname-USB-ttl which didn't had current-protection.

ascillato2, 2018-09-22:
If you have burned another USB-ttl adapter, means that your device (or those I/O) has short circuit. So it is defective. Sorry. There is nothing we can do from the software side.

Have you checked that you are powering with 3.3v? there are some adapters that gives 5v and burn your esp8266

Dees7, 2018-09-22:
I'm sure adapter 3.3V (NodeMCU and  FTDI232). 
I checked with Multimeter. I inspect soldering. where is no short circuit.
Both of mine Sonoff Basic behave the same way. It strange.

No one have troubles with Sonoff Basic R2? Is do my trouble in my hands? :)

ascillato2, 2018-09-22:
I have several installed at home and no problem at all. Sorry

Dees7, 2018-10-01:
Might be this will be useful for someone.

I connect rx\tx via 100Ohm resistors and Vcc via 10 Ohm. After this sonoff starts and flashed normal.

But it looks like ESP8266 burned out =(
After flash I see 
```
rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

```