arendst/Tasmota #3875: MQTT Topic Case Sensitivity

thehookup, 2018-09-23:
Since my Tasmota retain video I've been helping a bunch of people troubleshoot their command topic issues.  A theme that I am seeing is that tasmota is really forgiving about errors in command topic and still processing commands.  For instance:

Tasmota has the command topic "cmnd/sonoff/POWER", but because it is subscribed to "cmnd/sonoff/#" it will also respond to commands send on "cmnd/sonoff/power" (lowercase) and "cmnd/sonoff/POWER " (extra space at the end).  On the broker these are different topics and can therefore retain different messages.

I'm not sure if this is by design, or just a quirk of the way that command topic is processed, but I think some mystery switching issues (that are ultimately caused by errors in user configuration) could be avoided if tasmota only responded to the case sensitive correct topic.

Thanks for all your hard work!

ascillato, 2018-09-23:
Hi, 

Thanks for helping solving users issues. 

Tasmota uses trim to eliminate spaces and uppercase to process all parameters, so that behaviour is by design.

ascillato2, 2018-09-23:
I will add this to the wiki.

closed 2018-09-23 by ascillato2