arendst/Tasmota #3876: Native support for HomeKit

LorisBatacchi, 2018-09-23:
Hi, 
I saw some Sonoff projects that has native support for HomeKit (they don't need homebridge) like this one: 
https://github.com/Gruppio/Sonoff-Homekit

Is it possible to have something like that in Tasmota?

Frogmore42, 2018-09-23:
Since the author of that project didn't actually share the source code, it is not a simple answer to the question.  Clearly, it is possible, but without the source code it is highly unlikely to happen soon.

ascillato, 2018-09-23:
The author of that firmware is asking for money in order to give the source code. So, It is unlikely that he/she will share the code to Tasmota.

closed 2018-09-23 by ascillato2

ageorgios, 2018-10-03:
This is not true,
Great work has been done by this guy
https://github.com/maximkulkin/esp-homekit

ascillato, 2018-10-03:
Hi,

> This is not true

Well, it is. https://github.com/Gruppio/Sonoff-Homekit is asking for money.

> Great work has been done by this guy
> https://github.com/maximkulkin/esp-homekit

Indeed. Great work is there.
As that is not a library, a major adaptation need to be done in order to be added to Tasmota or any other project.

> I already have HomeKit compatibe sonoffs, I would love to see it integrated with Tasmota!

If you have already working HomeKit compatibe sonoffs, please, adapt it to Tasmota and make a PR. At this moment we are focused on other features to be added to Tasmota and it is unlikely to be taken into account soon, so, if you want you are free to adapt it.

Thanks for share the links and ideas.

Frogmore42, 2018-10-03:
The other thing to remember is that Tasmota is already running low on RAM.  HomeKit does not appear to be a simple protocol with minimal RAM requirements.  It is more likely like TLS which is so large you can't have much else running or it will run out of resources.

If you need HomeKit you can achieve that with the solution you already have if you need HomeKit and MQTT you probably should consider something like a HomeKit to MQTT adapter that can run on something that has more resources.

tahitibub, 2019-12-20:
Fyi, Apple published source code of HomeKit on Github
https://github.com/apple/HomeKitADK

jakobsudau, 2019-12-24:
> Fyi, Apple published source code of HomeKit on Github
> https://github.com/apple/HomeKitADK

Hello! 
Would be really awesome to see this integrated and have native HomeKit support!

sj26, 2020-02-04:
@gruppio has released their sonoff homekit code:

https://github.com/Gruppio/Sonoff-Homekit/tree/master/sonoff_advanced

It is based on esp-homekit. Native integration should be possible! 🤞

mriksman, 2020-02-24:
+1 Adding HomeKit to Tasmota... Wowee.

ricargr, 2020-03-10:
+1 it will be the perfect firmware

tonysprenk, 2020-05-09:
It is now possible to easily convert Tasmota devices to HAA. HAA, or Home Accessory Architect is native HomeKit firmware for esp devices.  https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#installing-from-tasmota

jeylites, 2020-05-09:
@tonysprenk 

Very cool, I’m assuming after one converts Tasmota to HAA you can’t go back to Tasmota if needed?

tonysprenk, 2020-05-10:
Not without flashing by wire.

S474N, 2020-09-05:
And will Tasmota sometimes support Homekit natively?

naoisegolden, 2020-10-17:
Seems they added detailed instructions on how to achieve this from available open source code here https://github.com/Gruppio/Sonoff-Homekit/pull/8

ialidigital, 2020-11-24:
The code to make this happen seems to be here: 
[https://github.com/thefonseca/Sonoff-Homekit/blob/master/sonoff_advanced/main.c](https://github.com/thefonseca/Sonoff-Homekit/blob/master/sonoff_advanced/)

start with [main.c](https://github.com/thefonseca/Sonoff-Homekit/blob/master/sonoff_advanced/main.c)

Im a Sonoff newb, but that looks like all that's needed?

Frogmore42, 2020-11-24:
It is great that everyone is interested in HomeKit.  It is even greater that there are options for converting your Sonoff devices to use Homekit.  But, ease of adding this capability to Tasmota hasn't changed in the 2+ years people have been asking for the capability.  This repo appears to have source code: https://github.com/maximkulkin/esp-homekit that probably works.  But, this code is not an Arduino library, so it is not compatible with Tasmota (which requires an Arduino compliant library).  It also likely uses a lot of code space, so even if it were compatible, it would probably not fit with Tasmota in the available code space. 

I am not sure I am clear on what people really expect to be able to do with HomeKit control of Sonoff devices that they also need Tasmota.  This is the mission statement for Tasmota:
_easy configuration using webUI, OTA updates, automation using timers or rules, expandability and entirely local control over MQTT, HTTP, Serial or KNX._

HomeKit appears to be a system that wants to be the controller of the system.  If you have bought into the Apple HomeKit view of the automation world, you don't need (and probably don't want) Tasmota.  Tasmota, on the other hand, uses much more open/simple protocols that allow you to use whatever system you want to be in control.  I have many Tasmota devices in my home.  They all speak MQTT.  I use Node-Red for some control and data gathering.  I also use OpenHAB for some other devices.  I use HA-Bridge (to speak MQTT to Tasmota devices) to provide Alexa something she can control for those few devices I want to directly control.

I don't think the answer (for HomeKit) is going to change until the next generation of Sonoff devices that have much greater capability, but still the same amount of openness.

Hustenbonbon, 2020-12-03:
Here is a arduino library: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266 

(Doesn't solve the space and if it philosophically fits to Tasmota though)

Currently testing the library

s-hadinger, 2023-01-02:
The plan in Tasmota is to eventually support Matter protocol over Wifi which will be more largely supported than Homekit. iOS Matter support is now part of Homekit.

Neifab, 2023-03-13:
> The plan in Tasmota is to eventually support Matter protocol over Wifi which will be more largely supported than Homekit. iOS Matter support is now part of Homekit.

Definitely a good thinking

kpyszkowski, 2024-11-28:
@s-hadinger 
> The plan in Tasmota is to eventually support Matter protocol over Wifi which will be more largely supported than Homekit. iOS Matter support is now part of Homekit.

_Eventually_? Isn't the Matter integration included in stable releases?

s-hadinger, 2024-11-28:
Yes, this message was 23 months ago. Matter is indeed in stable releases.