arendst/Tasmota #3877: tasmota retains power from home assistant after restarting

NischalBhandari, 2018-09-23:
I am using tasmota along with home assistant . I have  the following code in home assistant 
  - platform: mqtt
    name: "ROOF_TOP_LIGHTS"
    command_topic: "cmnd/ROOF_TOP_LIGHTS/POWER1"
    state_topic: "stat/ROOF_TOP_LIGHTS/POWER1"
    availability_topic: "tele/ROOF_TOP_LIGHTS/LWT"
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false

After i use the configuration of poweonstate 3 in tasmota  without keeping the mqtt topic i get the expected state (which was present before restart) after the tasmota has been restarted .
While if the mqtt topic is eastablished and it connects to home assistant . The device always goes to off state after the tasmota has been restarted .
I figured out that that the tasmota has been retaining the off state from the home -assistant but cannot figure out what to do to keep the state  when the device restarted .
Can anyone please help me with this issue ?

NischalBhandari, 2018-09-23:
The issue seems to be with home assistant 
 mosquitto_sub -h 192.168.15.10 -t '+/MASTER_BED_ROOM/+' -v
tele/MASTER_BED_ROOM/LWT Online
cmnd/MASTER_BED_ROOM/power1 OFF
cmnd/MASTER_BED_ROOM/power2 OFF
stat/MASTER_BED_ROOM/POWER1 OFF
stat/MASTER_BED_ROOM/POWER2 OFF

The home assistant is sending this topic ...
has anyone got any idea how to flush these

ascillato, 2018-09-23:
Hi,

Duplicated issue. This was addressed several times in issues. 

Please watch this video

https://youtube.com/watch?v=31IyfM1gygo

Thanks

closed 2018-09-23 by ascillato2

NischalBhandari, 2018-10-01:
> [image: Boxbe] <https://www.boxbe.com/overview> Adrian Scillato (
> notifications@github.com) is not on your Guest List
> <https://www.boxbe.com/approved-list?tc_serial=43455156165&tc_rand=1232711784&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=CaOIQPrQ%2BuqKfXwTUjhFuf%2B9WKwwT06MMVCQtinAhr0%3D&token=4JSm05AM3rQlBcuEG2yk9Z0GbwsBCkw6vSd5p2zQtrj%2Fmm3gJv%2FZpLBkyu5X9VKX>
> | Approve sender
> <https://www.boxbe.com/anno?tc_serial=43455156165&tc_rand=1232711784&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=CaOIQPrQ%2BuqKfXwTUjhFuf%2B9WKwwT06MMVCQtinAhr0%3D&token=4JSm05AM3rQlBcuEG2yk9Z0GbwsBCkw6vSd5p2zQtrj%2Fmm3gJv%2FZpLBkyu5X9VKX>
> | Approve domain
> <https://www.boxbe.com/anno?tc_serial=43455156165&tc_rand=1232711784&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&dom&key=CaOIQPrQ%2BuqKfXwTUjhFuf%2B9WKwwT06MMVCQtinAhr0%3D&token=4JSm05AM3rQlBcuEG2yk9Z0GbwsBCkw6vSd5p2zQtrj%2Fmm3gJv%2FZpLBkyu5X9VKX>
>
> Hi,
>
> Duplicated issue. This was addressed several times in issues.
>
> Please watch this video
>
> https://youtube.com/watch?v=31IyfM1gygo
>
> Thanks

Thanks sir,
I was actually looking for clearing the mqtt retain option which was
already set to flush it i used mosquitto_pub -h x.x.x.x -t tele/topic1/lwt
-r -n to flush from bbroker...
Thanks for the reply i really appreciate it..

With regards ,
Nischal bhandari