arendst/Tasmota #3878: Shelly2: No calibration possible (Version 6.2.1.5)

h-tro, 2018-09-23:
I've just flashed my Shelly2 with Tasmota 6.2.1.5 and connected a 60 W bulb with an external power meter for calibration.

I have run the VoltageSet, CurrentSet, and PowerSet commands with the values of the external power meter. But the values displayed by Tasmota do not change. No calibration possible. Can anybody confirm this? 
A bug?

arendst, 2018-09-23:
Enable more logging with weblog 4. Execute command ``VoltageSet xxx`` and report the output pls.

h-tro, 2018-09-23:
Output in the file
(MQTT is disabled)

[Shelly2Log1.txt](https://github.com/arendst/Sonoff-Tasmota/files/2408641/Shelly2Log1.txt)

arendst, 2018-09-23:
Looks fine. Response changes indeed from 08FA (=229.8V) to 090A (=231.4V)

h-tro, 2018-09-23:
Now it's working.
Is the problem serial log level?

arendst, 2018-09-23:
If seriallog is >0 it interferes with energy monitoring. I'll disable it for shelly2 in next update.

closed 2018-09-23 by arendst