arendst/Tasmota #3882: Shelly2: FrequencySet is not working properly (Tasmota Version 6.2.1.5)

h-tro, 2018-09-23:
After flash standard FrequencySet is 0 -> Tasmota shows 51 Hz.
My meter shows 50 Hz.
I set FrequencySet 50 -> weblog shows FrequencySet 500 -> Tasmota shows 66 Hz.
I set back to 0 or another value, Tasmota always shows 66 Hz.
It is no longer possible to calibrate the frequency.
Tasmota always shows 66 Hz. A restart does not help.
[Shelly2Log2.txt](https://github.com/arendst/Sonoff-Tasmota/files/2408759/Shelly2Log2.txt)

ascillato, 2018-09-23:
@h-tro

Hi,

Theo has updated the repository. Please, can you test again with last firmware if the issue is solved?

Thanks

arendst, 2018-09-23:
Pls install the latest released development version.

As your frequency gain setting is now wrong (0) you will have to execute command ``FrequencySet 50`` twice to get a valid indication.

From now on this fault should not occur anymore as it was caused by bad calculation trying to set frequency to 0.5Hz. Also input range checking has now been applied.

Regarding your obervations:

> After flash standard FrequencySet is 0 -> Tasmota shows 51 Hz.
> My meter shows 50 Hz.

As there is no initial calibration relation needed between Tasmota and the energy chip you will observer this value. Once a FrequencySet command is given Tasmota will show a value of Hz * 1000.

> I set FrequencySet 50 -> weblog shows FrequencySet 500 -> Tasmota shows 66 Hz.

Calculation was wrong and calibrated frequency was set to 0.5Hz resulting in overflow indication (65.535Hz)

> I set back to 0 or another value, Tasmota always shows 66 Hz.
> It is no longer possible to calibrate the frequency.
> Tasmota always shows 66 Hz. A restart does not help.

Due to mis-calibration the gain register was set to 0 resulting in a calibration formula that cannot calculate a valid value anymore.
The changes in the latest release will detect this situation when a FrequencySet command is executed and set both gain register and reference register back to a valid 50Hz value.

h-tro, 2018-09-23:
Version 6.2.1.6 starts with serial log level = 0
and the frequency calibration works well.

You should give a hint in the wiki,
that for the calibration of Shelly2 the Serial-Log level must be 0.

ascillato, 2018-09-23:
Hi,

As this is fixed, please close this issue. Thanks