arendst/Tasmota #3883: Sonoff T1 2CH EU dead after "Reset configuration" pressed

j1gg3, 2018-09-23:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ x] Hardware used : RaspberryPi to reflash via ESPTool 
- [ x] Provide the output of command ``status 0`` :
``` no output possible, device not responding

```

**(Please, remember to close the issue when the problem has been addressed)**

I have a Sonoff T1 2CH EU version which I've flashed successfully with sonota, also I've updated to latest available Tasmota release. Everything was working fine but due to lot of testing I decided to press the button for resetting whole configuration. The device never came up. Status LED is blinking for approx 10 seconds. I tried also enabling WIFI AP mode but unfortunately this did not the trick.  I tried also to reflash the firmware via my Raspberry but as of I'm connecting VCC my Pi stops working. Before I also had some one Sonoff devices which bricked and I was able to recover them via ESPTool and my Pi. 

So, did I missed to press something in a combination?

ascillato, 2018-09-23:
No, you did that ok. I don't know what may have happened in your device. 

If you have used v6.2.1 from 10 days ago, that one had a firmware compilation problem (misalignment). Now it is corrected. 

Please, try to erase all flash and flash again using esptool.py to the last development version from http://thehackbox.org/tasmota/

j1gg3, 2018-09-23:
I am not able to access the device. My Pi is crashing when I'm connecting the sonoff board

ascillato, 2018-09-23:
Then, you can try with a FTDI adapter and a PC.

If that also don't work, your device is burnt. Sorry. 

There isn't much we can do from the Tasmota software side.

closed 2018-09-24 by ascillato2