arendst/Tasmota #3884: @reboot switching relay when mqtt is enabled

tarbax, 2018-09-23:
i am having a weird issue
once mqtt is enabled i get ghost switching at reboot.
even if i set my topics to something random (to be sure that nothing is configured in HA)

connected/configured:

d1 = relay1
d2 = switch2
d3 = switch3
d4 = switch4
d5 = DHT11

my settings:
switchtopic garage_door
PulseTime1 5
switchmode2 1
switchmode3 1
switchmode4 1

ascillato, 2018-09-23:
Hi

That is due to retained messages in your MQTT broker. 

Please, what the following video in order to properly configure your Tasmota and Home Automation Software.

https://youtube.com/watch?v=31IyfM1gygo

Thanks

tarbax, 2018-09-23:
> Hi
> 
> That is due to retained messages in your MQTT broker.
> 
> Please, watch the following video in order to properly configure your Tasmota and Home Automation Software.
> 
> https://youtube.com/watch?v=31IyfM1gygo
> 
> Thanks

Thanks this solves 1 of the 2 switches at restart.
the second relay switch is from my reed switches.
how to solve this ?

ascillato, 2018-09-23:
If your wire is too long and close to AC mains there isn't much to do. Noise is too much for 3.3v.

Please search in the issues. Some people have put resistors, capacitors and shielded wires to reduce this long wire issue. Also in the videos of DrZzs this was addressed (link in Tutorials in the wiki)

tarbax, 2018-09-23:
> 👍

found the problem it is because the door is closed (switch 2) 
if the door is open i don't have this problem anymore how can i fix this ?

ascillato, 2018-09-23:
I don't understand. Please, elaborate more.

tarbax, 2018-09-23:
if the door is closed i get a relay switch at reboot
if the door is open i don't get the relay switch at reboot
this is from d2pin configured at switch2

ascillato, 2018-09-23:
Please, use the command weblog 4 in order to see more logging and copy and paste the console output at startup

tarbax, 2018-09-23:
```
00:00:00 CFG: Loaded from flash at 3F6, Count 94
00:00:00 APP: Boot Count 53
00:00:00 SRC: Restart
00:00:00 Project sonoff garagepoort (Topic garage, Fallback DVES_0AB97B, GroupTopic sonoffs) Version 6.2.0-2_4_2
00:00:00 SRC: Switch
00:00:00 RSL: stat/garage/RESULT = {"POWER":"OFF"}
00:00:00 RSL: stat/garage/POWER = OFF
00:00:00 CFG: Saved to flash at 3F5, Count 95, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Black_Ubiquiti_2.4Ghz_IOT_02 in mode 11N as garage-6523...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 SRC: Switch
00:00:04 RSL: stat/garage/RESULT = {"POWER":"ON"}
00:00:04 RSL: stat/garage/POWER = ON
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on garage-6523.local with IP address 192.168.1.152
00:00:04 SRC: PulseTimer
00:00:04 RSL: stat/garage/RESULT = {"POWER":"OFF"}
00:00:04 RSL: stat/garage/POWER = OFF
00:00:04 APP: (UTC) Sun Sep 23 15:41:27 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
16:41:28 HTP: Console
16:41:29 MQT: Attempting connection...
16:41:29 MQT: Connected
16:41:29 MQT: tele/garage/LWT = Online (retained)
16:41:29 MQT: cmnd/garage/POWER = 
16:41:29 MQT: Subscribe to cmnd/garage/#
16:41:29 MQT: Subscribe to cmnd/sonoffs/#
16:41:29 MQT: Subscribe to cmnd/DVES_0AB97B/#
16:41:29 MQT: tele/garage/INFO1 = {"Module":"Generic","Version":"6.2.0","FallbackTopic":"DVES_0AB97B","GroupTopic":"sonoffs"}
16:41:29 MQT: tele/garage/INFO2 = {"WebServerMode":"Admin","Hostname":"garage-6523","IPAddress":"192.168.1.152"}
16:41:29 MQT: tele/garage/INFO3 = {"RestartReason":"Software/System restart"}
16:41:29 MQT: stat/garage/RESULT = {"POWER":"OFF"}
16:41:29 MQT: stat/garage/POWER = OFF (retained)
16:41:37 MQT: tele/garage/STATE = {"Time":"2018-09-23T16:41:37","Uptime":"0T00:00:14","Vcc":3.069,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Black_Ubiquiti_2.4Ghz_IOT_02","RSSI":68,"APMac":"8A:8A:20:4C:95:63"}}
16:41:37 MQT: tele/garage/SENSOR = {"Time":"2018-09-23T16:41:37","Switch2":"OFF","Switch3":"ON","Switch4":"ON","DHT11":{"Temperature":21.7,"Humidity":56.0},"TempUnit":"C"}
```

tarbax, 2018-09-23:
keep in mind i have the pulsetimer on

ascillato, 2018-09-23:
Your switch is turning it off and on.

tarbax, 2018-09-23:
> Your switch is turning it off and on.

how do i fix that ?

ascillato, 2018-09-23:
I don't know what is happening with your switch. You will need to measure it.

Another thing you can try is to change to another gpio.

ascillato, 2018-09-24:
Have you read this part of the wiki?

https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

ascillato, 2018-09-24:
Hi,

Have you managed to solve your issue?

ascillato2, 2018-09-25:
Hi,

Please, when you have time, share your results. 

Thanks

closed 2018-09-25 by ascillato2