arendst/Tasmota #3885: Tasmota & Domoticz - one-way only :-(

mkrawcz1, 2018-09-23:
I'm testing connection to Domoticz with T1 EU 2Ch.
I've set IDX 1 and 2 for both channels.
If I press button 1 on T1, I can see in Mosquitto:
`domoticz/in {"idx":1,"nvalue":1,"svalue":"","Battery":79,"RSSI":7}
stat/sonoff/RESULT {"POWER1":"ON"}
stat/sonoff/POWER1 ON
domoticz/out {
   "Battery" : 79,
   "RSSI" : 7,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014051",
   "idx" : 1,
   "name" : "Kuchnia 1",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}`

But if I use virtual switch from Domoticz, I can see on MQTT:
`domoticz/out {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014051",
   "idx" : 1,
   "name" : "Kuchnia 1",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}`
but no response (and no light of course) from T1.

ascillato, 2018-09-23:
Please, check the Tasmota wiki at domoticz chapter in order to properly configure it

mkrawcz1, 2018-09-23:
I've chcked this, as you can see Tasmota is connected to MQTT broker and publishes messages in 'domoticz/in' topic. Proper IDX values are set. But seems that Tasmota doesn't react on messages with propoer IDX, published in 'domoticz/out' topic.

ascillato, 2018-09-23:
Please, use the command weblog 4 in the console and then try from domoticz and paste the console output here.

mkrawcz1, 2018-09-23:
>18:11:31 CMD: weblog 4
18:11:31 MQT: stat/sonoff/RESULT = {"WebLog":4}
18:11:31 CFG: Zapisane do pamięci flash w F7, Licz 29, Bajtow 3584
18:11:50 WIF: Sprawdzanie połączenia...
18:11:50 WIF: Połączony
18:12:10 WIF: Sprawdzanie połączenia...
18:12:10 WIF: Połączony

In the meantime on MQTT preview:
>domoticz/out {
   "Battery" : 79,
   "RSSI" : 7,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014051",
   "idx" : 1,
   "name" : "kuch_1",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

>domoticz/out {
   "Battery" : 79,
   "RSSI" : 7,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014052",
   "idx" : 2,
   "name" : "kuch_2",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

>stat/sonoff/RESULT {"WebLog":4}

>domoticz/out {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014051",
   "idx" : 1,
   "name" : "kuch_1",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}

>domoticz/out {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014051",
   "idx" : 1,
   "name" : "kuch_1",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}

ascillato, 2018-09-23:
Hi,

From your console output, nothing is reaching Tasmota. I don't use Domoticz, so I don't know what ia missing in your domoticz configuration. 

Can you try the automatic configuration in order to compare what is missing/wrong in your domoticz config?

arendst, 2018-09-24:
It works for me as expected.
```
11:48:42 SRC: MQTT
11:48:42 RSL: Received Topic domoticz/out, Data Size 259, Data {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}

11:48:42 DOM: idx 177, nvalue 0
11:48:42 DOM: Received Topic /Power, Data 0
11:48:42 RSL: Group 0, Index 1, Command POWER, Data 0
11:48:42 MQT: stat/wkaku5/RESULT = {"POWER":"OFF"}
11:48:42 MQT: stat/wkaku5/POWER = OFF
11:48:42 CFG: Saved to flash at FA, Count 102, Bytes 3584

11:48:47 SRC: MQTT
11:48:47 RSL: Received Topic domoticz/out, Data Size 259, Data {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}

11:48:47 DOM: idx 177, nvalue 1
11:48:47 DOM: Received Topic /Power, Data 1
11:48:47 RSL: Group 0, Index 1, Command POWER, Data 1
11:48:47 MQT: stat/wkaku5/RESULT = {"POWER":"ON"}
11:48:47 MQT: stat/wkaku5/POWER = ON
11:48:47 CFG: Saved to flash at F9, Count 103, Bytes 3584

11:49:27 APP: Button1 multi-press 1
11:49:27 SRC: Button
11:49:27 MQT: domoticz/in = {"idx":177,"nvalue":0,"svalue":"","Battery":58,"RSSI":9}
11:49:27 MQT: stat/wkaku5/RESULT = {"POWER":"OFF"}
11:49:27 MQT: stat/wkaku5/POWER = OFF
11:49:27 CFG: Saved to flash at F8, Count 104, Bytes 3584
11:49:27 SRC: MQTT
11:49:27 RSL: Received Topic domoticz/out, Data Size 237, Data {
   "Battery" : 58,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 0,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

11:49:27 DOM: idx 177, nvalue 0

11:49:32 APP: Button1 multi-press 1
11:49:32 SRC: Button
11:49:33 MQT: domoticz/in = {"idx":177,"nvalue":1,"svalue":"","Battery":58,"RSSI":9}
11:49:33 MQT: stat/wkaku5/RESULT = {"POWER":"ON"}
11:49:33 MQT: stat/wkaku5/POWER = ON
11:49:33 CFG: Saved to flash at F7, Count 105, Bytes 3584
11:49:33 SRC: MQTT
11:49:33 RSL: Received Topic domoticz/out, Data Size 237, Data {
   "Battery" : 58,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

11:49:33 DOM: idx 177, nvalue 1

11:49:39 APP: Button1 multi-press 1
11:49:39 SRC: Button
11:49:39 MQT: domoticz/in = {"idx":177,"nvalue":0,"svalue":"","Battery":58,"RSSI":9}
11:49:39 MQT: stat/wkaku5/RESULT = {"POWER":"OFF"}
11:49:39 MQT: stat/wkaku5/POWER = OFF
11:49:39 CFG: Saved to flash at F6, Count 106, Bytes 3584
11:49:40 SRC: MQTT
11:49:40 RSL: Received Topic domoticz/out, Data Size 237, Data {
   "Battery" : 58,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 0,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

11:49:40 DOM: idx 177, nvalue 0

11:49:44 APP: Button1 multi-press 1
11:49:45 SRC: Button
11:49:45 MQT: domoticz/in = {"idx":177,"nvalue":1,"svalue":"","Battery":58,"RSSI":9}
11:49:45 MQT: stat/wkaku5/RESULT = {"POWER":"ON"}
11:49:45 MQT: stat/wkaku5/POWER = ON
11:49:45 SRC: MQTT
11:49:45 RSL: Received Topic domoticz/out, Data Size 237, Data {
   "Battery" : 58,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 177,
   "name" : "mqtt wkaku5",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

11:49:45 DOM: idx 177, nvalue 1
11:49:45 CFG: Saved to flash at F5, Count 107, Bytes 3584
```
Pls provide output of command ``status 0``.

ascillato, 2018-09-24:
Hi,

Have you managed to solve your issue?

plampix, 2018-09-24:
Are you sure your MQTT_MAX_PACKET_SIZE is 1000?

I had the same problem, probably because the mqqt lib got updated.

ascillato2, 2018-09-26:
Closing as there is no feedback. Please, ask to reopen if you have further questions. Thanks.

closed 2018-09-26 by ascillato2

JoergZ2, 2018-10-01:
Maybe it's a matter of wifi power. The RSSI value between 7 and 12 is very poor. Please try to power up wifi energy or (for testing) use a smaller distance between wifi access point and T1. Or check if there are channel conflicts with other wifi networks.

mkrawcz1, 2018-10-03:
After upgrading:
- Arduino to 1.8.7,
- Tasmota to 6.2.1
automatic discovery seems to work OK.
I've left Domoticz as it was.

droneando, 2019-02-07:
Old topic but I have same issues just today. Finally figured out it has related to mosquitto user topic privileges config! Allowing mosquitto user to read/write in domoticz/# topic solves the issue.

TamojitSaha, 2019-02-10:
@droneando did you made some changes in domoticz configuration or inside tasmia firmware?
Please help me as I am still suffering with the same issue of one-sided(cannot use *domoticz* switch to control sonoff)

droneando, 2019-02-10:
All I did was follow the tasmota wiki to configurate the options in domoticz and sonoff device itself! My issue was related to wrong mosquitto user privilege configurattion. After allowing the mqtt user configurated in sonoff device (in /etc/mosquitto/acl file) to read and write in domoticz/# topic all was fine!
As pointed in tasmota wiki you can check correct mqtt user configuration sending and received manually   mqtt messages with mosquitto_sub & mosquitto_pub, if you are using mosquitto.
If all is fine with mqtt broker there is no reason for domoticz not working with your sonoff tasmota device.

TamojitSaha, 2019-02-10:
@droneando
Thanks for fast response!

In my setup domoticz with mqtt is running on RPI Zero W and sonoff device is connected to it. Till now everything is fine.

In Sonoff MQTT config page,
_Topic:_ sonoff
_Full Topic:_ %prefix%/%topic%/
  
And Domoticz as usual configuration(if you want i can post a screenshot)

The issue:
I can toggle from Sonoff server page which gets reflect on Domoticz dashboard but not the reverse one.
Is this issue only related to the topic names inside tasmota firmware??? 
As pointed by you, changing */etc/mosquito/acl file* will do the trick?

droneando, 2019-02-10:
I'm not an expert in domoticz, only I'm testing for about 4 or 5 days! As I said I followed instructions in tasmota wiki and, after solving the mqtt user privileges issue, all worked fine.
This is the tasomta wiki page [https://github.com/arendst/Sonoff-Tasmota/wiki/Domoticz](url)
May be you find more answers there

TamojitSaha, 2019-02-10:
Can you post your *acl* file if possible?
It would be of great help

sghazagh, 2019-10-21:
Guys, I am suffering the same issue. 
Have loaded the Tasmota into "Lolin D1 Mini Pro" which is a ESP8266 based board and it works just find in hence of web interface and controling the switch (GPIO pin) from web interface but cannot control the switch (GPIO pin) from Domoticz.
It works fine if I toggle the switch from web interface and it updates the Domoticz but not other way around!
Did you guys find out how can we fix that issue?
@TamojitSaha, did you resolve your issue at all?
@droneando, is there anything else you can think of to help me resolve this issue?!

TamojitSaha, 2019-10-22:
@sghazagh I switched to NodeRED and stopped working on it.

sghazagh, 2019-10-22:
@TamojitSaha I found a workaround to put the http url into switch action and it works just fine.
Interested to get it working by receiving the message through "domoticz/out" MQTT topic. 
I think it does not subscribe to that topic and that's the issue!!!

How to you use Node-red to manage this issue?
I already have Node-red with all my sensors register in MQTT topic, interested to know if you are using the MQTT to send the response back from Domoticz in node-red?

redguyz, 2021-09-26:
> 
> 
> Are you sure your MQTT_MAX_PACKET_SIZE is 1000?
> 
> I had the same problem, probably because the mqqt lib got updated.

That's exactly where the problem is.