arendst/Tasmota #3886: Sonoff SV

chris8837, 2018-09-23:
I'm using a sonoff SV and have my garage hooked up to the GPIO pin and reed sensor.  The problem I'm having is when it's not connected to WIFI or looses signal from WIFI when I open the garage it immediatly closes. 

I use home assisant and I can unplug it and no issues.  

Only when it can't connect to my Wifi router.

Any suggestions?  here is a log (At 09:04:21 is when I pushed the button on my sv)

09:04:06 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:13 WIF: Connect failed as AP cannot be reached
09:04:13 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:20 WIF: Connect failed as AP cannot be reached
09:04:20 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:21 RSL: stat/sonoffsvgarage/RESULT = {"POWER":"ON"}
09:04:21 RSL: stat/sonoffsvgarage/POWER = ON
09:04:22 RSL: stat/sonoffsvgarage/RESULT = {"POWER":"OFF"}
09:04:22 RSL: stat/sonoffsvgarage/POWER = OFF
09:04:27 WIF: Connect failed as AP cannot be reached
09:04:27 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:34 WIF: Connect failed as AP cannot be reached
09:04:34 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:41 WIF: Connect failed as AP cannot be reached
09:04:41 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:48 WIF: Connect failed as AP cannot be reached
09:04:48 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:04:55 WIF: Connect failed as AP cannot be reached
09:04:55 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:05:02 WIF: Connect failed as AP cannot be reached
09:05:02 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:05:09 WIF: Connect failed as AP cannot be reached
09:05:09 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:05:16 WIF: Connect failed as AP cannot be reached
09:05:16 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:05:23 WIF: Connect failed as AP cannot be reached
09:05:23 WIF: Connecting to AP1 sable88 in mode 11N as sonoffsvgarage-0217...
09:05:35 WIF: Connected
09:05:35 DNS: Initialized
09:05:35 MQT: Attempting connection...
09:05:37 MQT: Connected
09:05:37 MQT: tele/sonoffsvgarage/LWT = Online (retained)
09:05:37 MQT: cmnd/sonoffsvgarage/POWER = 
09:07:03 APP: Serial logging disabled
09:07:16 MQT: tele/sonoffsvgarage/STATE = {"Time":"2018-09-23T09:07:16","Uptime":"0T00:10:25","Vcc":3.160,"POWER":"OFF","Wifi":{"AP":1,"SSId":"sable88","RSSI":100,"APMac":"C4:6E:1F:56:65:6E"}}
09:07:16 MQT: tele/sonoffsvgarage/SENSOR = {"Time":"2018-09-23T09:07:16","Switch2":"OFF"}

wongnam, 2018-09-23:
Pls use WifiConfig 5
Disable wifi config but retry same AP without restart and flash writes

chris8837, 2018-09-24:
17:00:13 MQT: tele/sonoffsvgarage/SENSOR = {"Time":"2018-09-23T17:00:13","Switch2":"OFF"}
17:02:00 MQT: tele/sonoffsvgarage/UPTIME = {"Time":"2018-09-23T17:02:00","Uptime":"0T07:12:07"}
17:05:13 MQT: tele/sonoffsvgarage/STATE = {"Time":"2018-09-23T17:05:13","Uptime":"0T07:15:20","Vcc":3.162,"POWER":"OFF","Wifi":{"AP":1,"SSId":"sable88","RSSI":100,"APMac":"C4:6E:1F:56:65:6E"}}
17:05:13 MQT: tele/sonoffsvgarage/SENSOR = {"Time":"2018-09-23T17:05:13","Switch2":"OFF"}
17:10:13 MQT: tele/sonoffsvgarage/STATE = {"Time":"2018-09-23T17:10:13","Uptime":"0T07:20:20","Vcc":3.160,"POWER":"OFF","Wifi":{"AP":1,"SSId":"sable88","RSSI":100,"APMac":"C4:6E:1F:56:65:6E"}}
17:10:13 MQT: tele/sonoffsvgarage/SENSOR = {"Time":"2018-09-23T17:10:13","Switch2":"OFF"}
17:11:23 MQT: cmnd/2/POWER2 = ON
17:12:18 MQT: cmnd/2/POWER2 = OFF
17:14:09 CMD: wificonfig
17:14:09 MQT: stat/sonoffsvgarage/RESULT = {"WifiConfig":"5 (Wait)"}

I tested to see if wificonfig was set to 5 and it is.

I don't think this is on the HA side as I can completly disable HA and my wifi router and it does the same thing.  It's maybe some setting in tasmota that I'm not seeing?

I tried a few changes, but I dont' want to mess anything up

ascillato, 2018-09-24:
Ok, please provide us the troubleshooting template complete and also use the command weblog 4 in the console in order to see debug information and please provide the output of the console when your issue is manifested.
Thanks.

wongnam, 2018-09-24:
The last thing is how you read the sensor and process the logic flow for your door by using mqtt or using Rule in Támota.
I suggest using Rule in Tasmota local.

ascillato, 2018-09-24:
Hi

Have you managed to solve your issue?

chris8837, 2018-09-24:
No I’m just going to buy open garage and be done with it. Every other switch works flawless and I don’t think I have the experience to mess around with this. Tasmota is too advanced for me I think. Haha. I’ve heard good things about open garage.

closed 2018-09-24 by chris8837