arendst/Tasmota #3890: Command NtpServer restarts Tasmota

h-tro, 2018-09-24:
Used Tasmota Version 6.2.1.6:
Command NtpServer 1 (with space) shows the NtpServer and restarts Tasmota.
Is that correct?
Command NtpServer1 (without space) shows the NtpServer without restart.

ascillato, 2018-09-24:
Hi,

As designed.

closed 2018-09-24 by ascillato2

h-tro, 2018-09-24:
Why?

ascillato, 2018-09-24:
If the line 1046 in _sonoff.ino_ is commented out, the restart is not being performed. After some seconds, Tasmota Sync to the NTP server.

So, seems that it is not needed the restart there anymore.

I will propose a PR #3891 to Theo in order to let him evaluate this,

Thanks for pointing this out.

arendst, 2018-09-24:
ntpserver1 just shows the current ntpserver 1.
ntpserver does the same
ntpserver2 shows the current ntpserver 2
ntpserver 1 changes ntpserver 1 to the default ntpserver and does a restart as designed.

arendst, 2018-09-24:
Without the restarg ntpservers are not re-initialized.

ascillato, 2018-09-24:
Hi,

I did the following test using the Proposed PR #3891

* Put an invalid IP in ntpserver1, 2 and 3 (192.168.1.3)
* `Restart 1` -> After restart Tasmota obviously shows 00:00 as time
* Put the NTPserver1 as default by command `NTPserver1 1`

And after few seconds it start to show the right time.

Console Output:

```
00:00:00 CFG: Cargado desde flash en 3F8, Conteo 314
00:00:00 APP: Conteo Reinicios 157
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.6-2_4_2
00:00:00 CFG: Grabado a la flash en 3F7, Conteo 315, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:06 MQT: Intentando conectar...
00:00:08 MQT: Conectado
00:00:08 MQT: tele/living/LWT = Online (Grabado)
00:00:08 MQT: cmnd/living/POWER = 
00:00:08 MQT: Suscribir a cmnd/living/#
00:00:08 MQT: Suscribir a cmnd/sonoffs/#
00:00:08 MQT: Suscribir a cmnd/DVES_591719/#
00:00:08 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.2.1.6","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
00:00:08 MQT: tele/living/INFO3 = {"RestartReason":"External System"}
00:00:08 MQT: stat/living/RESULT = {"POWER1":"OFF"}
00:00:08 MQT: stat/living/POWER1 = OFF
00:00:15 CMD: ntpserver1
00:00:15 SRC: WebConsole from 192.168.1.123
00:00:15 RSL: Topic Recibido /ntpserver1, Tamaño de Datos 0, Datos 
00:00:15 RSL: Grupo 0, Índice 1, Comando NTPSERVER, Datos 
00:00:15 MQT: stat/living/RESULT = {"NtpServer1":"192.168.1.3"}
00:00:17 MQT: tele/living/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:17","Vcc":2.990,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":72}}
00:00:24 WIF: Probando conexión...
00:00:24 WIF: Conectado
00:00:29 CMD: ntpserver1 192.168.1.2
00:00:29 SRC: WebConsole from 192.168.1.123
00:00:29 RSL: Topic Recibido /ntpserver1, Tamaño de Datos 11, Datos 192.168.1.2
00:00:29 RSL: Grupo 0, Índice 1, Comando NTPSERVER, Datos 192.168.1.2
00:00:29 MQT: stat/living/RESULT = {"NtpServer1":"192.168.1.2"}
00:00:29 CFG: Grabado a la flash en 3F6, Conteo 316, Bytes 3584
00:00:44 WIF: Probando conexión...
00:00:44 WIF: Conectado
00:00:54 CMD: ntpserver1 1
00:00:54 SRC: WebConsole from 192.168.1.123
00:00:54 RSL: Topic Recibido /ntpserver1, Tamaño de Datos 1, Datos 1
00:00:54 RSL: Grupo 0, Índice 1, Comando NTPSERVER, Datos 1
00:00:54 MQT: stat/living/RESULT = {"NtpServer1":"pool.ntp.org"}
00:00:54 CFG: Grabado a la flash en 3F5, Conteo 317, Bytes 3584
00:01:04 WIF: Probando conexión...
00:01:04 WIF: Conectado
00:01:24 WIF: Probando conexión...
00:01:24 WIF: Conectado
00:01:30 APP: (UTC) Mon Sep 24 19:10:44 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
16:10:58 WIF: Probando conexión...
16:10:58 WIF: Conectado
16:11:18 WIF: Probando conexión...
16:11:18 WIF: Conectado
```

arendst, 2018-09-24:
I cannot explain as the sntp init code is only run at reboot time where the sdk ntp servers are being set with settings.ntpserevers.....

ascillato, 2018-09-24:
He he he,

So there is some of your Tasmota magic code there :wink: 

Tested again:

```
00:00:00 CFG: Cargado desde flash en 3FA, Conteo 328
00:00:00 APP: Conteo Reinicios 161
00:00:00 SRC: Restart
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.6-2_4_2
00:00:00 CFG: Grabado a la flash en 3F9, Conteo 329, Bytes 3584
00:00:00 WIF: Probando conexión...
00:00:00 WIF: Intentando conectar...
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:01 WIF: Probando conexión...
00:00:01 WIF: Intentando conectar...
00:00:02 WIF: Probando conexión...
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conexión...
00:00:03 WIF: Intentando conectar...
00:00:04 WIF: Probando conexión...
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:06 MQT: Intentando conectar...
00:00:09 MQT: Conectado
00:00:09 MQT: tele/living/LWT = Online (Grabado)
00:00:09 MQT: cmnd/living/POWER = 
00:00:09 MQT: Suscribir a cmnd/living/#
00:00:09 MQT: Suscribir a cmnd/sonoffs/#
00:00:09 MQT: Suscribir a cmnd/DVES_591719/#
00:00:09 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.2.1.6","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
00:00:09 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
00:00:09 MQT: tele/living/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/living/RESULT = {"POWER1":"OFF"}
00:00:09 MQT: stat/living/POWER1 = OFF
00:00:17 MQT: tele/living/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:17","Vcc":2.990,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":66}}
00:00:19 CMD: ntpserver1
00:00:19 SRC: WebConsole from 192.168.1.123
00:00:19 RSL: Topic Recibido /ntpserver1, Tamaño de Datos 0, Datos 
00:00:19 RSL: Grupo 0, Índice 1, Comando NTPSERVER, Datos 
00:00:19 MQT: stat/living/RESULT = {"NtpServer1":"192.168.1.3"}
00:00:24 WIF: Probando conexión...
00:00:24 WIF: Conectado
00:00:25 CMD: ntpserver2
00:00:25 SRC: WebConsole from 192.168.1.123
00:00:25 RSL: Topic Recibido /ntpserver2, Tamaño de Datos 0, Datos 
00:00:25 RSL: Grupo 0, Índice 2, Comando NTPSERVER, Datos 
00:00:25 MQT: stat/living/RESULT = {"NtpServer2":"192.168.1.3"}
00:00:28 CMD: ntpserver3
00:00:28 SRC: WebConsole from 192.168.1.123
00:00:28 RSL: Topic Recibido /ntpserver3, Tamaño de Datos 0, Datos 
00:00:28 RSL: Grupo 0, Índice 3, Comando NTPSERVER, Datos 
00:00:28 MQT: stat/living/RESULT = {"NtpServer3":"192.168.1.3"}
00:00:38 CMD: ntpserver1 1
00:00:38 SRC: WebConsole from 192.168.1.123
00:00:38 RSL: Topic Recibido /ntpserver1, Tamaño de Datos 1, Datos 1
00:00:38 RSL: Grupo 0, Índice 1, Comando NTPSERVER, Datos 1
00:00:38 MQT: stat/living/RESULT = {"NtpServer1":"pool.ntp.org"}
00:00:38 CFG: Grabado a la flash en 3F8, Conteo 330, Bytes 3584
00:00:44 WIF: Probando conexión...
00:00:44 WIF: Conectado
00:01:04 WIF: Probando conexión...
00:01:04 WIF: Conectado
00:01:24 WIF: Probando conexión...
00:01:24 WIF: Conectado
00:01:30 APP: (UTC) Mon Sep 24 19:52:37 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
16:52:51 WIF: Probando conexión...
16:52:51 WIF: Conectado
16:53:01 CMD: status 0
16:53:01 SRC: WebConsole from 192.168.1.123
16:53:01 RSL: Topic Recibido /status, Tamaño de Datos 1, Datos 0
16:53:01 RSL: Grupo 0, Índice 1, Comando STATUS, Datos 0
16:53:01 MQT: stat/living/STATUS = {"Status":{"Module":18,"FriendlyName":["Living"],"Topic":"living","ButtonTopic":"living","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:53:01 MQT: stat/living/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:54","StartupUTC":"2018-09-24T19:51:07","Sleep":0,"BootCount":161,"SaveCount":330,"SaveAddress":"3F8000"}}
16:53:01 MQT: stat/living/STATUS2 = {"StatusFWR":{"Version":"6.2.1.6","BuildDateTime":"2018-09-24T16:06:01","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:53:01 MQT: stat/living/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"192.168.1.104","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["04008009","55818000","00000000"]}}
16:53:01 MQT: stat/living/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":2580,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","0C0003B0","23021796","000000C0"]}}
16:53:01 MQT: stat/living/STATUS5 = {"StatusNET":{"Hostname":"living-5913","IPAddress":"192.168.1.33","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:59:17:19","Webserver":2,"WifiConfig":5}}
16:53:01 MQT: stat/living/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_591719","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:53:01 MQT: stat/living/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 24 19:53:01 2018","Local":"Mon Sep 24 16:53:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"07:17","Sunset":"19:30"}}
16:53:01 MQT: stat/living/STATUS10 = {"StatusSNS":{"Time":"2018-09-24T16:53:01"}}
16:53:01 MQT: stat/living/STATUS11 = {"StatusSTS":{"Time":"2018-09-24T16:53:01","Uptime":"0T00:01:54","Vcc":2.989,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":68}}}
```

ascillato, 2018-09-24:
The only change is that the **line 1046** in _sonoff.ino_ is commented out.

arendst, 2018-09-24:
It might be the sdk just needed the pointers to settings.ntpservers.

I'll merge now and will test tomorrow too ;-)

ascillato, 2018-09-24:
Thanks Theo!

ascillato, 2018-09-24:
@h-tro 

> Why?

Your **_why_** make all this :rofl: