arendst/Tasmota #3892: Extrange issue with GPIO14 (Sonoff Basic) when using a PIR.

gerardovf, 2018-09-25:
Everything works as expected BUT if the Sonoff is not able to connect to the MQTT broker, then when the PIR is triggered the relay is operated!

_Originally posted by @gerardovf in https://github.com/arendst/Sonoff-Tasmota/issue_comments#issuecomment-424265932_

ascillato, 2018-09-25:
Hi,

That is the way that it is designed, for a PIR would be better to use rules instead of switchtopic to make the behaviour you are expecting.

Please, see in the wiki the example:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

Then, after you make it work, you can go further and try the example 15 in order to add energy saving. 

Hope this helps. 

Thanks

gerardovf, 2018-09-25:
@ascillato Thanks a lot for the tip! I'm closing this issue

closed 2018-09-25 by gerardovf