arendst/Tasmota #3895: Setting Sleep back to 0 made my Sonoff Basic unresponsive

overrunner, 2018-09-25:
I have a Sonoff Basic flashed with Tasmota 6.2.1 compiled on a Mac high Sierra in Atom.
Everything apart from Timers and MQTT is disabled.

**To Reproduce**
_Steps to reproduce the behavior:_

Flash on Tasmota Basic and set everything you like to have.
Set Sleep to 50 to safe power.
Set back to Sleep 0 after (????) time, because the internal time (set by ntp) went wrong after a couple of minutes.

**Expected behavior**
No more sleep, but a correct time.

What happened?
After setting to Sleep to 0, the devices was very unresponsive via HTTP. Via MQTT the Power was only a empty response and even there was no "on/off" status on the start page of the device. the part of it was blank.
Reset 1 did not work, a reflash after many tries pressing F5 did not work.
Setting back to sleep 50 made the device work again.

andrethomas, 2018-09-25:
You did not complete the template as requested when you created the issue so its difficult to diagnose. Are you by chance compiling with 2.4.2 ?

overrunner, 2018-09-25:
Platform IDE is 2.1.7
You mean upgrading to 2.4.2?

Tried to put all information asked by the template in that issue. Sorry if I have not mentioned anything.

andrethomas, 2018-09-25:
Can you get a status 0 output?

I don't mean use 2.4.2 - I assumed from what you wrote that you self-compiled, which means you could be inadvertently running using the 2.4.2 core which has issues with sleep

ascillato, 2018-09-26:
Hi,

From the wiki at trobleshooting (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting):

If you use the ESP8266 Core Libraries v2.3.0, sleep is working.
If you use the ESP8266 Core Libraries v2.4.0, it has wifi issues. Do not use.
If you use the ESP8266 Core Libraries v2.4.1, it has wifi issues. Do not use.
If you use the ESP8266 Core Libraries v2.4.2, sleep is disabled due to problems in the Espressif SDK.

`Status 0` is a command to put in the Tasmota console. It gives a lot of information needed for troubleshooting. That is why is asked in the Troubleshooting template and needed to have the information we need in order to properly help you. Not using the template make us guess or spend time in asking for the information on the template.

Please, in order to solve your issue:
* erase all flash with esptool.py (as explained in the wiki)
* flash again with a precompiled bin from development branch (http://thehackbox.org/tasmota/)

If in that case, it works but you want to compile by yourself, use in your platformio.ini
* ESP8266 Core Libraries v2.3.0 if you want to use sleep feature
```
[common] ; ************************************************************
; *** Esp8266 core for Arduino version 2.3.0
platform = espressif8266@1.5.0
```
* ESP8266 Core Libraries v2.4.2 if you don't need sleep feature and want faster and better wifi.
[common] ; ************************************************************
```
; *** Esp8266 core for Arduino version 2.4.2
platform = espressif8266@1.8.0
```

We recommend the precompiled bins.

If you need further assistance, please, find us in Tasmota Discord Chat.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-09-26 by ascillato2

overrunner, 2018-09-26:
Sleep is set to zero and device does not show on/off status and reacts only "sometimes" on a request. I did a http request and got some output:

{"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://","RestartReason":"Software/System restart","Uptime":"0T01:48:45","StartupUTC":"2018-09-26T16:07:12","Sleep":0,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"},"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-08T16:22:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fs-IoT",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]},"StatusMEM":{"ProgramSize":409,"Free":592,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","07000680","24000000","00000094","00000000"]},"StatusNET":{"Hostname":"sonoff-4905","IPAddress":"172.16.200.160","Gateway":"172.16.200.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.200.1","Mac":"84:F3:","Webserver":2,"WifiConfig":5},"StatusMQT":{"MqttHost":"set.topic.before","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_48B329","MqttUser":"symcon_mosq","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Wed Sep 26 17:55:57 2018","Local":"Wed Sep 26 18:55:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:09","Sunset":"18:06"},"StatusSNS":{"Time":"2018-09-26T18:55:57"},"StatusSTS":{"Time":"2018-09-26T18:55:57","Uptime":"0T01:48:45","POWER":"OFF","Wifi":{"AP":1,"SSId":"fs-IoT","RSSI":32,"APMac":"92:2A:"}}}

ascillato, 2018-09-26:
Hi,

Please, in order to solve your issue:

* erase all flash with esptool.py (as explained in the wiki)
* flash again with a precompiled bin from development branch (http://thehackbox.org/tasmota/)

overrunner, 2018-09-28:
Tried a lot of things with another BASIC. It seems to be a problem with newer Versions. Even if I flash your compiled version after a flash erase, the led just blinks and the Basic is not reachable. On an S20 it works fine.
When I flash 6.11c the basic works fine. as far as I flash a newer version on it, it stops working.

andrethomas, 2018-09-28:
The RSSI level is concerning - it suggests bad wifi coverage - try changing your wifi channel to one which is not used or overlaps too closely to other people in your nearby area

andrethomas, 2018-09-28:
For example, my most remote sonoff device reports "RSSI":74 - so relative signal strength is ~74% on mine.. yours is 32%!

overrunner, 2018-09-28:
yes, you're right. thanks for that hint.
But I tried it now with another basic. it has 74 rssi in my office.

andrethomas, 2018-09-28:
As long as its > 50% you should not have ping issues - if you do it means there is something else amis with your wifi network. There's several hundred thousand devices running this firmware at the moment so unless there is something physically defective with your device it can only be your network setup. All my devices are currently running the latest development pre-compiled binaries without any problems... these include Sonoff Basic's, Sonoff POW's, Sonoff Dual's, and other custom boards with ESP8266 modules in them

overrunner, 2018-09-28:
Thanks for your quick feedback. I made some new tests and found out, that your compiled versions work indeed as long as I don't come from a "new" self-compiled version. 
I compiled on Windows now again and it works! Before I compiled on OSX and Linux. Both made problems...

andrethomas, 2018-09-28:
If you are using platformio then that could be the source of the problem - There is currently a process wherein we are trying to determine how the core imported by platformio since the release of 2.4.2 support was added for platformio but as of now this is not solved. If you really need to self-compile perhaps try to compile using Arduino and confirm if that is working - if it is then the platformio problem still persists. Like I said its a very complex issue to resolve (the platformio one) because the problems occur randomly across a few number of identified devices most especially the R2.

savage872, 2019-02-08:
I had the same problem on my sonoff basic r1.. done sleep 0, the device restarted and the led started flashing constantly and couldn't do anything with it. I connected serial adapter and it was saying it can't connect to AP1 which was online. restarted but it didn't help. When I typed in sleep 50 it went back to normal. I only used the command sleep 0 just so it stop the wifi disconnecting every couple of hours for no reason 
status 0 below
17:43:10 MQT: stat/fireplace/STATUS = {"Status":{"Module":1,"FriendlyName":["Fireplace"],"Topic":"fireplace","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:43:10 MQT: stat/fireplace/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:39","StartupUTC":"2019-02-08T16:38:31","Sleep":50,"BootCount":26,"SaveCount":48,"SaveAddress":"F6000"}}
17:43:10 MQT: stat/fireplace/STATUS2 = {"StatusFWR":{"Version":"6.4.1.15(953bcd2-sonoff)","BuildDateTime":"2019-02-08T17:15:23","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
17:43:10 MQT: stat/fireplace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home Sweet Home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000001"]}}
17:43:10 MQT: stat/fireplace/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":448,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
17:43:10 MQT: stat/fireplace/STATUS5 = {"StatusNET":{"Hostname":"fireplace-8119","IPAddress":"192.168.0.101","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:72:DF:B7","Webserver":2,"WifiConfig":3}}
17:43:10 MQT: stat/fireplace/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"fireplace","MqttClient":"fireplace","*":"*","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:43:10 MQT: stat/fireplace/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 08 16:43:10 2019","Local":"Fri Feb 08 17:43:10 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:58"}}
17:43:10 MQT: stat/fireplace/STATUS10 = {"StatusSNS":{"Time":"2019-02-08T17:43:10"}}
17:43:10 MQT: stat/fireplace/STATUS11 = {"StatusSTS":{"Time":"2019-02-08T17:43:10","Uptime":"0T00:04:39","Vcc":3.439,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Home Sweet Home","BSSId":"E4:F4:C6:79:C6:C8","Channel":6,"RSSI":100}}}

I guess this hasn't been fixed yet.

andrethomas, 2019-02-08:
Unfortunately the Basic's need some dynamic sleep (not wifi sleep) otherwise they go into thermal runaway because the pc board was not designed well enough to dissipate heat away from the ESP8286 chip... this became a problem after many new drivers were added which put more load on the ESP8266 leaving less room for the SDK to do its wifi work in the background.