arendst/Tasmota #3896: Shelly 2 Automatically turns ON/OFF with external switch connected

ricreis394, 2018-09-25:
_Originally posted by @ricreis394 in https://github.com/arendst/Sonoff-Tasmota/issues/2789#issuecomment-424422654_

I have a shelly 2 with Tasmota 6.2.1.6, I have connected a momentary switch to S1. Randomly the relay aggregated to the S1 turns On and Off.

ascillato, 2018-09-25:
Please, use the troubleshooting template. We need the information that is being asked there in order to help you properly.

ricreis394, 2018-09-25:
Where do I find the troubleshooting template?

ascillato2, 2018-09-25:
When you have clicked on New Issue, you have 3 options (Bug, Troubleshooting, Feature Request). And below there is a little link that said **open a regular issue**. You have clicked last one that don't have template. We made the template to save you and us time on asking for more basic information.

the template is:

> **IMPORTANT NOTICE**
> If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.
> 
> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> - [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
> - [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
> - [ ] Development/Compiler/Upload tools used :
> - [ ] Hardware used : 
> - [ ] Provide the output of command ``status 0`` :
> ```
> STATUS 0 OUTPUT HERE
> ```
> 
> **(Please, remember to close the issue when the problem has been addressed)**

Mainly, we need the output of the` status 0` command on the console.

ricreis394, 2018-09-25:
```
00:36:06 CMD: Status 0
00:36:06 MQT: casa/sonoff/stat/STATUS = {"Status":{"Module":47,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
00:36:06 MQT: casa/sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T09:00:40","StartupUTC":"2018-09-25T14:35:26","Sleep":0,"BootCount":6,"SaveCount":205,"SaveAddress":"F9000"}}
00:36:06 MQT: casa/sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.2.1.6","BuildDateTime":"2018-09-24T23:38:17","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:36:06 MQT: casa/sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["@HA",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
00:36:06 MQT: casa/sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":18,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","0FDAE794","240003A0","23B617CE","000003C0"]}}
00:36:06 MQT: casa/sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0228","IPAddress":"10.10.1.57","Gateway":"10.10.1.254","Subnetmask":"255.255.255.0","DNSServer":"10.10.1.254","Mac":"84:F3:EB:9F:60:E4","Webserver":2,"WifiConfig":5}}
00:36:06 MQT: casa/sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.1.248","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F60E4","MqttUser":"mqttbroker_sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:36:06 MQT: casa/sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 25 23:36:06 2018","Local":"Wed Sep 26 00:36:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:42","Sunset":"18:40"}}
00:36:06 MQT: casa/sonoff/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:36:06 MQT: casa/sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2018-09-26T00:36:06","Switch1":"OFF","Switch2":"OFF","ENERGY":{"Total":0.019,"Yesterday":0.017,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000,"Frequency":0}}}
00:36:06 MQT: casa/sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2018-09-26T00:36:06","Uptime":"0T09:00:40","Vcc":3.196,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"@HA","BSSId":"82:2A:A8:5A:E0:93","Channel":1,"RSSI":74}}}
```

ascillato, 2018-09-25:
Thanks,

Please, can you also use the command `weblog 4` in your console in order to see more debug information and try to catch that power on and off again?

ricreis394, 2018-09-25:
Okay, I think I caught the ghost off. 

```
00:44:51 SER: Received 06 19 DB 00 00 00 52 09 E1 00 00 00 0B 00 00 00 CA 01 00 00 1F C1 BE C5 6F
00:44:52 SER: Received 06 19 DB 00 00 00 53 09 E5 00 00 00 09 00 00 00 CA 01 00 00 00 C0 BE C5 52
00:44:52 SRC: Switch
00:44:52 MQT: casa/sonoff/stat/RESULT = {"POWER1":"OFF"}
00:44:52 MQT: casa/sonoff/stat/POWER1 = OFF (retained)
00:44:53 SER: Received 06 19 B2 00 00 00 53 09 92 00 00 00 11 00 00 00 89 01 00 00 73 D0 FC C5 5E
00:44:53 CFG: Saved to flash at FA, Count 212, Bytes 3584
00:44:54 SER: Received 06 19 AA 00 00 00 53 09 93 00 00 00 13 00 00 00 89 01 00 00 20 D0 E2 C5 EC
00:44:55 MQT: casa/sonoff/tele/SENSOR = {"Time":"2018-09-26T00:44:55","ENERGY":{"Total":0.019,"Yesterday":0.017,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000,"Frequency":0}}
```

ascillato, 2018-09-25:
Thanks again for the tests.

You can see there in your log that say: `00:44:52 SRC: Switch`
So, that means that the command to power off is coming from your switch. If you are not pressing your switch, that means that is electrical noise coming from your switch or bad electrical connections.
Is not software related.

Please, check your wirings.

mike2nl, 2018-09-26:
Stupid question:
You have a cat? 
Cats love to treat us humans and mine two love it to dance in the night on my table stuff. 
Got hem with a night view camera.

arendst, 2018-09-26:
Pls provide output of commands ``switchmode1`` and ``switchmode2``. Should be 0.

Consider disabling powerretain (``powerretain 0``) as it will most likely be the source of your next issue.

When connecting two switches in parallel as you do the second switch won't do much when the first switch is closed. It may also introduce noise when both switches are open.

ricreis394, 2018-09-26:
@mike2nl Thanks for the try. I don't have any cat.
@ascillato Thanks. I don't believe the noise comes from the switch, the switch is from a high quality brand. Is it possible to come from wires behind the switch? Well, low quality wires? Can it be to run in my test environment? This is my shceme 
[image]

arendst, 2018-09-26:
Have a look at my request just before you send your answer.

ricreis394, 2018-09-26:
@arendst My switches have a spring in each button allowing it to go back to initial position. This switches only does a pulse. Isn't constant the current.
Using switchmode 0 I'm not able to use the switch correctly, when I press the button and release the power goes On and Off instantly.

Output of the switches:
```
09:29:20 CMD: switchmode1
09:29:20 MQT: casa/sonoff/stat/RESULT = {"SwitchMode1":4}
09:29:25 CMD: switchmode2
09:29:25 MQT: casa/sonoff/stat/RESULT = {"SwitchMode2":4}
```

arendst, 2018-09-26:
OK. Makes sense.

In that case you might want to try a workaround using pull-up enabled on the switch input. For this test select module generic and assign the following GPIOs. This will disable energy monitoring but that's not a problem during testing.

[image]

The main change here is that you select ``Switch1`` for GPIO12 and NOT ``Switch1n``.

Give it a try for a bit longer than your ghost switches appeared and let me know the result.

ricreis394, 2018-09-26:
Ok, module is updated. I'll keep watching. I realized that when I restart the shelly, It needs longer time to see happening the ghosts. So, I hope having news until the end of the day.

arendst, 2018-09-26:
Just release a fix for switchmode PUSHBUTTON and default switch no-pullup.

closed 2018-09-27 by ascillato2

ricreis394, 2018-09-27:
Right now for me the latest development version is working great. And at any time you can OTA upgrade firmware.
I'm using hackbox sonoff-classic.bin 1M

TheEggi, 2018-09-30:
I have pretty much the same setup as you are using and I really cannot make it work properly. Are you using "GPIO_SWT1_NP" (the default as in the Shelly 1/2 module) or GPIO_SWT1?

ricreis394, 2018-09-30:
use the latest development version, and for this to work you need switches with spring that returns to initial state after pressed and use switchmode 4. Your setup seems correct.
If still don't work open a new issue

TheEggi, 2018-10-01:
It seems to be already working for me but I needed to change the function

```
void GpioSwitchPinMode(uint8_t index)
{
  if (pin[GPIO_SWT1 +index] < 99) {
//    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : bitRead(switch_no_pullup, index) ? INPUT : INPUT_PULLUP);

    uint8_t no_pullup = 0;
    if (bitRead(switch_no_pullup, index)) {
      no_pullup = (Settings.switchmode[index] < PUSHBUTTON);
    }
    pinMode(pin[GPIO_SWT1 +index], (16 == pin[GPIO_SWT1 +index]) ? INPUT_PULLDOWN_16 : (no_pullup) ? INPUT : INPUT_PULLUP);
  }
}
```
to not have the no_pullup set in case of PUSHBUTTONS.

fsilva80, 2018-12-18:
Hello!
I also have a Shelly2 with problems only on the second relay.
Here's my wiring diagram:
[IMG]http://i64.tinypic.com/30aa003.jpg[/IMG]

I already tried switching both wires from and switch and also for the lamps.
The problem is only on the second relay...

I also ha this message when it appens:
SRC: Switch

But, as i said i switched the wiring and the problem stays in relay 2.

I'm using the latest Sonoff-Tasmota 6.4.0!

Here's the status 0:

`03:33:40 CMD: STATUS 0
03:33:41 MQT: stat/shelly2_01/STATUS = {"Status":{"Module":47,"FriendlyName":["Interruptor entrada da sala","Interruptor fundo da sala"],"Topic":"shelly2_01","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
03:33:41 MQT: stat/shelly2_01/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T07:08:37","StartupUTC":"2018-12-17T19:25:04","Sleep":50,"BootCount":10,"SaveCount":120,"SaveAddress":"F4000"}}
03:33:41 MQT: stat/shelly2_01/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2018-12-16T14:31:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
03:33:41 MQT: stat/shelly2_01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Lost Soul",""],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
03:33:41 MQT: stat/shelly2_01/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
03:33:41 MQT: stat/shelly2_01/STATUS5 = {"StatusNET":{"Hostname":"shelly2_01-3091","IPAddress":"192.168.1.72","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"CC:50:E3:13:4C:13","Webserver":2,"WifiConfig":4}}
03:33:41 MQT: stat/shelly2_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_134C13","MqttUser":"fsilva80","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:33:41 MQT: stat/shelly2_01/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 18 02:33:41 2018","Local":"Tue Dec 18 03:33:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
03:33:41 MQT: stat/shelly2_01/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:33:41 MQT: stat/shelly2_01/STATUS10 = {"StatusSNS":{"Time":"2018-12-18T03:33:41","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2018-12-17T01:37:55","Total":0.015,"Yesterday":0.011,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
03:33:41 MQT: stat/shelly2_01/STATUS11 = {"StatusSTS":{"Time":"2018-12-18T03:33:41","Uptime":"0T07:08:37","Vcc":3.556,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Lost Soul","BSSId":"00:26:44:38:86:5A","Channel":1,"RSSI":80}}}`

kiwijunglist, 2018-12-18:
I had this problem too ^

My plan was to reflash the original firmware and if the issue was ongoing then to return the unit. I haven't got around to that bit yet