arendst/Tasmota #3897: MAX31855 and thermocouple

sirLeone, 2018-09-25:
Any chance to add support for thermocouple K via MAX31855 (very popular) ? There is already Adafruit library which works also with esp8266 so maybe it wont be a big problem ?
https://github.com/adafruit/Adafruit-MAX31855-library

I know that now MAX31850 is supported which is 1-wire thermocouple amplifier but as far as i know it is not produced anymore or at least hard to buy.

mike2nl, 2018-09-26:
Yes the MAIX, MAX31850 zero has ending lifetime some time ago so far i know.

Question:
- You are messaure so high temperatures that you need a thermocouple K type?

- It is not the solution but there are other very good sensors when you will messaure 
air or liquid media. Have you ever tried a PT100 or PT1000?

Official there is no I2C breakout board, SPI only, but there are chips and i had done 
some tests. No official driver yet and no support by tasmota right now.

Possible you are able to write a driver based on the Ladyada stuff for the newer MAX31855?

sirLeone, 2018-09-26:
I would like to make fireplace controler so that i need to measure temps up to let say 400C-600C (sonoff may control fans and air intake valve additionaly)
I think that thermocouples K type are so popular and easier to byu that it would be great addition for tasmota for many people.

Unfortunately i am not good at programming and i am afraid that my driver would not be acceptable quality :)

mike2nl, 2018-10-09:
@sirLeone 
Have ordered now a little breakout board with that MAX31855 and 
a K-Type sensor for -100...700 degress celsius.

Have to wait around two weeks then tey will be here and i can start to develop it.

mike2nl, 2018-10-24:
@sirLeone 
Short update for you:
- the sensor arrived from china yesterday in the afternoon.
and that one is from a very, very good quality so far i have messaured.
The shield around the teflon cable looks so good that you can't see any space at all.

- the amplifier is still missing because the douane will have 5,-Euro extra tax money. ;-)
I got that information yesterday. I hop the AMP arrives this week..

These guys are so s****d there. They think our local system loses money when 
i order it in China. Right, but no shop has that special amp because the zero
point compensation i need. So the import control system is wrong.

Tommy-LSA, 2018-11-09:
The approach reading both sensors is the same but MAX31855 has an other response than MAX6675. (MAX6675 outputs 12 bits of data. The MAX31855 outputs 14 bits) The bit operation (https://gist.github.com/sleemanj/059fce7f1b8087edfe7d7ef845a5d881) won't work for MAX31855. Instead this have to be used from Adafruit library double _Adafruit_MAX31855::readCelsius(void)_ https://github.com/adafruit/Adafruit-MAX31855-library/blob/master/Adafruit_MAX31855.cpp

Response of MAX31855 is 32bit while response of MAX6675 is 16bit. It's because  MAX31855  delivers it's own temperature aditional to the temperature of thermocouple. Also MAX31855 can handle negative temperatures (-270°C-1800°C) while MAX6675 handle 0°C to +700°C

Page 6 is the relevant page on MAX6675  Datasheet http://www.datasheetcatalog.com/datasheets_pdf/M/A/X/6/MAX6675.shtml

Page 10 is the relevant page on MAX31855 Datasheet http://www.datasheetcatalog.com/datasheets_pdf/M/A/X/3/MAX31855.shtml

However, both sensors can be implemented with the same code approach (without additional libraries) and a fork in reading function. Also multiple sensors are possible on same communication gpio's as long each sensor get it's own gpio for CS

ascillato, 2018-11-10:
@LakeCruiser

>  I have the MAX31855 working

using Tasmota ?

Tommy-LSA, 2018-11-10:
I have ordered both (not received yet) and put the things together in the meantime. Following code is untested but this way both Sensors could be implemented in one module without using any additional library.

`
#define SPI_SO 7
#define SPI_SCK 6

#define MAX6675_CS 8
#define MAX31855_CS 9

void setup() {
  Serial.begin(115200);
}

void loop() {
  // read MAX6675
  uint16_t max6675_temp;
  max6675_temp = readCelsiusMAX6675(MAX6675_CS);
  Serial.print("MAX6675  Temperature: ");
  Serial.println(max6675_temp);

  // read MAX31855
  int32_t max31855_temp;
  max31855_temp = readCelsiusMAX31855(MAX31855_CS);
  Serial.print("MAX31855 Temperature: ");
  Serial.println(max31855_temp);
  delay(1000);
}

double readCelsiusMAX6675(sensor_cs) {
  uint16_t v;
  v = spiread16(sensor_cs);

  if (v & 0x4){    
    return NAN;     
  }
  
  v >>= 3;
  return v*0.25;
}

double readCelsiusMAX31855(sensor_cs) {
  int32_t v;
  v = spiread32(sensor_cs);

  if (v & 0x7) {
    return NAN; 
  }

  if (v & 0x80000000) {
    // Negative value, drop the lower 18 bits and explicitly extend sign bits.
    v = 0xFFFFC000 | ((v >> 18) & 0x00003FFFF);
  }
  else {
    // Positive value, just drop the lower 18 bits.
    v >>= 18;
  }

  double centigrade = v;
  // LSB = 0.25 degrees C
  centigrade *= 0.25;
  return centigrade;
}

uint16_t spiread16(sensor_cs) { 
  int i;
  uint32_t d = 0;

  digitalWrite(sensor_cs, LOW);
  delay(1);
  digitalWrite(SPI_SCK, LOW);
  delay(1);

  for (i=15; i>=0; i--) {
    digitalWrite(SPI_SCK, LOW);
    delay(1);
    d <<= 1;
    if (digitalRead(SPI_SO)) {
      d |= 1;
    }
    digitalWrite(SPI_SCK, HIGH);
    delay(1);
  }
  digitalWrite(sensor_cs, HIGH);
  //Serial.println(d, HEX);
  return d;
}

uint32_t spiread32(sensor_cs) { 
  int i;
  uint32_t d = 0;

  digitalWrite(sensor_cs, LOW);
  delay(1);
  digitalWrite(SPI_SCK, LOW);
  delay(1);

  for (i=31; i>=0; i--) {
    digitalWrite(SPI_SCK, LOW);
    delay(1);
    d <<= 1;
    if (digitalRead(SPI_SO)) {
      d |= 1;
    }
    digitalWrite(SPI_SCK, HIGH);
    delay(1);
  }
  digitalWrite(sensor_cs, HIGH);
  //Serial.println(d, HEX);
  return d;
}
`

LakeCruiser, 2018-11-10:
Talk to me!  Share details!  Please!Sent from my Verizon, Samsung Galaxy Note 9

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Sonoff-Tasmota/issues/3897#issuecomment-437574080",
"url": "https://github.com/arendst/Sonoff-Tasmota/issues/3897#issuecomment-437574080",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [arendst/Sonoff-Tasmota] MAX31855 and thermocouple (#3897)",
"sections": [
{
"text": "",
"activityTitle": "**Adrian Scillato**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@ascillato",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"arendst/Sonoff-Tasmota\",\n\"issueId\": 3897,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"arendst/Sonoff-Tasmota\",\n\"issueId\": 3897\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/arendst/Sonoff-Tasmota/issues/3897#issuecomment-437574080"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 384184953\n}"
}
],
"themeColor": "26292E"
}
]

ascillato, 2018-11-10:
@LakeCruiser

I'm asking you. You said in your previous post that you have that working with mqtt. My question is: do you have that working using Tasmota ?

LakeCruiser, 2018-11-11:
I converted mine to the MAX31855 dual probe.  It is discovered and displayed in Home Assistant but I can't get it to work with climate.  I don't know how to share my HA cards or yaml views but I can paste section if you need them.

mike2nl, 2018-11-25:
Sorry that i had done nothing on the driver. Had some really heart health issues.
So had to take time and need some more with resting. After taht i will look again
in it. If the driver is ready then ican test a lot....

saikek, 2018-12-08:
Have anyone tried this library ? 
https://github.com/SirUli/MAX6675

ipm79, 2018-12-31:
I am also working on a project to control my fireplace and needed support for a thermocouple sensor that supports high temperatures. The MAX31855 seemed to be a very good solution for my problem so I wrote a driver and filed a PR (#4764). Currently the driver only supports one MAX31855 connected to 3 user configurable pins. Communication is currently only supported via embedded SW SPI (I will add HW-SPI later). I've done tests with a MAX31855 breakout board and a Sonoff Tasmota Basic module (with the new board layout as described in https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic).
[image]
[image]
The JSON response includes both temperatures (probe and reference) and the errorcode reported by the chip (TC open, short to GND and short to VCC).
[image]

andrethomas2, 2019-01-01:
Closing this issue as it the PR made for this request has been merged into development. For reference please see PR https://github.com/arendst/Sonoff-Tasmota/pull/4764

closed 2019-01-01 by andrethomas2

ropopow, 2020-02-13:
How did you connect Sensor and MAX31855 to Sonoff Basic? Thank you.