arendst/Tasmota #3898: Tasmota v6.2.1 on Sonof Basic, sensors stoped reading after few days.

easyyu, 2018-09-25:
Im using v6.2.1 on Sonoff Basic. On GPIO 3 I have one DHT22 sensor as well as on GPIO 14.
They are connected with not longer than 10cm cables.
It working nicely few days (4-5), when suddenly stop to read data from sensor. The output looks like:

`23:50:25 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-09-25T23:50:25","Uptime":"0T00:29:06","Vcc":3.222,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mywifi","RSSI":84,"APMac":"xx:29:xx:xx:xx:xx"}}
23:50:25 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-09-25T23:50:25","AM2301-03":{"Temperature":**nan**,"Humidity":**nan**},"AM2301-14":{"Temperature":**nan**,"Humidity":**nan**},"TempUnit":"C"}`

The reading values are nan.
The sensrs are configured as: `AM2301`
The biggest problem is that reboot do not helping to start to read again, neither the save of configuration in sensor menu do nothing, **only powering off from electricity help**. This is fourth time that it happen.
The problem is that I can access only remotly the Sonoff basic.
Why this happening?
So my question is is there any way to reboot it to start to read again?

Output od command status 0 is:

`23:56:36 MQT: stat/sonoff-Rome/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff-Rome","ButtonTopic":"sonoff","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:56:36 MQT: stat/sonoff-Rome/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:17","StartupUTC":"2018-09-25T22:21:19","Sleep":0,"BootCount":104,"SaveCount":327,"SaveAddress":"FA000"}}
23:56:36 MQT: stat/sonoff-Rome/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:56:37 MQT: stat/sonoff-Rome/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["mywifi",""],"TelePeriod":30,"SetOption":["00008009","55818000","00000001"]}}
23:56:37 MQT: stat/sonoff-Rome/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
23:56:37 MQT: stat/sonoff-Rome/STATUS5 = {"StatusNET":{"Hostname":"sonoff-Rome-6044","IPAddress":"192.168.1.134","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"xxxxxxxxxxxxxxx","Webserver":2,"WifiConfig":2}}
23:56:37 MQT: stat/sonoff-Rome/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxx","MqttPort":xxxxx,"MqttClientMask":"Rome","MqttClient":"Rome","MqttUser":"xxxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:56:37 MQT: stat/sonoff-Rome/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 25 22:56:37 2018","Local":"Tue Sep 25 23:56:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:41","Sunset":"18:42"}}
23:56:37 MQT: stat/sonoff-Rome/STATUS10 = {"StatusSNS":{"Time":"2018-09-25T23:56:37","AM2301-03":{"Temperature":nan,"Humidity":nan},"AM2301-14":{"Temperature":nan,"Humidity":nan},"TempUnit":"C"}}
23:56:37 MQT: stat/sonoff-Rome/STATUS11 = {"StatusSTS":{"Time":"2018-09-25T23:56:37","Uptime":"0T00:35:18","Vcc":3.222,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mywifi","RSSI":76,"APMac":"xxxxxxxxxx"}}}
`
Thx

ascillato, 2018-09-25:
Have you tried another sensor?
Have you checked wiring?

easyyu, 2018-09-25:
Of course that I am..
I have few sensors, and this even happend just with one DHT22 on GPIO14.

Frogmore42, 2018-09-26:
There have been reports of this before.  The DHT22 and similar sensors are picky and can go out to lunch.  If I had to use one I would wire it so that it's power could be controlled by a GPIO and then I would power cycle it regularly.  Since I don't have to use one, I choose to use the BME280, HTU21, and SHT15.  Any of them are far better sensors for not a lot more money and a lot less aggravation.

closed 2018-09-26 by ascillato2