arendst/Tasmota #3903: Changing module to ID 43 (Sonoff POW R2) seems to cause the device to stop taking in serial commands via hard wire.

DouglasSim, 2018-09-26:
Description:
Changing module to ID 43 (Sonoff POW R2) seems to cause the device to stop taking in serial commands via hard wire. Changing the module back however, gets the serial input working again!. Seems like there is something with module 43. Module 06 Sonoff POW seems to work.

**Describe the bug**
User is not able to send commands via hard wire serial connection if the Sonoff is set to module 43 (web server console still works!). Changing the module to other modules (so far tested module 01 and 06) via web server seems to allow the hard wire serial connection to start accepting commands again.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Rpi2 + Sonoff POW R2 with Tasmota 6.2.1
- [x] _Development/Compiler/Upload tools used_ : Followed the guide here on https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi
- [x] _Provide the output of command_``status 0`` :
```
07:14:07 CMD: status 0
07:14:07 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:14:07 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":1152000,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:31","StartupUTC":"2018-09-26T06:13:36","Sleep":0,"BootCount":15,"SaveCount":36,"SaveAddress":"F8000"}}
07:14:07 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-23T17:01:36","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:14:07 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Slacker's Network Black","Slacker Modem"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
07:14:07 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":470,"Free":532,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
07:14:07 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1672","IPAddress":"192.168.0.46","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"64.59.144.92","Mac":"84:F3:EB:4A:C6:88","Webserver":2,"WifiConfig":5}}
07:14:07 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4AC688","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:14:07 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 26 06:14:07 2018","Local":"Wed Sep 26 07:14:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:42","Sunset":"18:40"}}
07:14:07 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-26T07:14:07"}}
07:14:07 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-26T07:14:07","Uptime":"0T00:00:31","Vcc":3.138,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Slacker's Network Black","RSSI":100,"APMac":"00:23:69:50:D7:DE"}}}
```

**To Reproduce**
- Rpi2 connected to a Sonoff POW R2 as per https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi
- Flash Tastota v6.2.1 + set up WIFI connection
- Open serial connection to the Sonoff device via RPI
- Change module to ID 43 via command "module 43"
- Now attempt to issue any other serial command and it seems like nothing happens (i.e. mashing enter doesn't give me the "Command":"Unknown" anymore). Note that logs still print out just fine.
- Go onto the web server and change the module back to ID 01 (i.e. Sonoff basic)
- Go back to RPI serial connection and now it accepts serial command.

**Expected behavior**
- Hard wire serial connection should accept serial commands regardless of module

**Screenshots**
Not applicable

**Additional context**
I only tested on 1 Sonoff device, seems to happen 100% of the time for me even after a reboot and re-flash.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-09-26:
As designed.

Module 43 uses the serial interface for energy monitoring which will be interfered with your serial data.

DouglasSim, 2018-09-26:
Thanks for the response, I also noticed that model 6 (i.e. normal POW) also does energy monitoring. Is this accurate if used with R2? I was thinking of using an opto-isolator so I can get serial logs while being connected to main power.

arendst, 2018-09-26:
Model 6 uses different energy monitoring hardware so can't be used on module 43 hardware.

See wiki regarding warnings and possible opto-coupler connection.

Set weblog or syslog to 4 to get more logging.

ascillato, 2018-09-26:
> Any suggestions? Or should I pick up a regular pow?

As a Pow and Pow R2 are grounded to Mains AC and not isolated, any connection or hardware modification is discouraged due to the danger of electrocution involved in it.

closed 2018-09-26 by ascillato2

DouglasSim, 2018-09-26:
@ascillato and @arendst thank you very much for your support