arendst/Tasmota #3904: Tele- Rule, Switch as a Sensor

TLCary, 2018-09-26:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
On boot I want to be able to get the value of a switch status.   "Tele-switch1#state" doesn't appear to work.

**Describe the solution you'd like**
Add tele- functionality to Switch (like the other sensors).

**Describe alternatives you've considered**
Server side script side to pull out the values from the MQTT tele/ Json and push the values back to TASMOTA.

**Additional context**
Could be I'm just not referencing the Switch correctly in my "on tele-".  Or I may be asking for something unreasonable...  but, can't hurt to ask. Thanks

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-26:
Hi,

May be I'm not undertanding the whole idea you are proposing. Why you want that?

Switch are triggers, so they can change a lot of times between teleperiods. So, if you have a tele-switchx you will have only the switch status on that instant (a switches snapshot every teleperiod time)

That is why in rules, there are the triggers `switchX#state` to inmediatelly run a rule when a switch change its state.

If you want to send the switch status using MQTT, you have **switchtopic** or **rules and publish command**

If you want to send all your switches status as a periodical snapshot, you can use a ruletimer and publish command.

TLCary, 2018-09-26:
I have two switches on a garage door, one that closes when the door is fully open, one that closes when the door is fully closed.  The initial state of the door at boot is the issue.  Once the door status changes the rules pick it up fine and I can see it’s status and send the proper commands depending on its state.  The initial state, (and it could be days with one swished closed), can be seen in the telemetry data but isn’t available to the rules on the device.

TLCary, 2018-09-26:
A slide switch to toggle between Heating and Cooling on a typical thermostat...  to make a robust device that will work even when MQTT is unavailable the rules would need that value.  On Change triggers are great once there is a change... but again, on boot up the rules don’t know the state though it’s value is in the telemetry data.

ascillato, 2018-09-26:
> The initial state, (and it could be days with one switch closed), can be seen in the telemetry data but isn’t available to the rules on the device.

Yep, you are right. At boot time, there is no way to take decisions based on the initial state of the switches.

ascillato, 2018-09-26:
I'm not very convinced with the `tele-switch#state` rule. That is going to be triggered every teleperiod. For the switch changes, it is the `SWITCH1#STATE` trigger.

As the issue is that up to now, it is not possible to trigger a rule at boot time with the state of the switch (in order to take decisions), what about adding a new type of trigger like:

`ON SWITCH1#BOOT=0 DO .... ENDON`

That will follow the same structure type of the actual triggers `SYSTEM#BOOT` and `SWITCH#STATE`

ascillato, 2018-09-26:
Hi,

I'm not very convinced with the `tele-switch#state` rule. That is going to be triggered every teleperiod. For the switch changes, it is aready the `SWITCH1#STATE` trigger.

As the issue is that up to now, it is not possible to trigger a rule at boot time with the state of the switch (in order to take decisions), what about adding a new type of trigger like:

`ON SWITCH1#BOOT=0 DO .... ENDON`

That will follow the same structure type of the actual triggers `SYSTEM#BOOT` and `SWITCH1#STATE`

Also this `SWITCH1#BOOT` can be extended also for `POWER1#BOOT` and `BUTTON1#BOOT`.

@TLCary, @arendst What do you think of these new triggers type?

arendst, 2018-09-26:
switch info is in status 10. Not state

TLCary, 2018-09-26:
Would be an excellent solution.  @ascillato, can you translate Theo’s reply for me?  I’m not as familiar with the internals...

TLCary, 2018-09-27:
Yep, no way to get the value from status 10 into a rule on the device.  How crazy difficult would it be to add functionality for?
```
ON SWITCH1#BOOT DO ...  %value% ENDON
ON SWITCH1#BOOT=0 DO .... ENDON
ON SWITCH1#BOOT=1 DO .... ENDON

```
While I'm dreaming, could we parse a rule and treat newline character the same as a space?  It would make rules far more readable.  And if we change the textbox for the rule into a textarea...  Can't blame me for asking.

ascillato, 2018-09-27:
Hi,

I have proposed the Pull Request #3910, for Theo to review it, with the addition of the new triggers:

* `SWITCH1#BOOT`
* `POWER1#BOOT`

Example:

Using the rule
```
rule 1
rule rule on switch1#boot do event bootsw1=%value% endon on power1#boot do event bootpo1=%value% endon on switch2#boot do event bootsw2=%value% endon on system#boot do event bootsys endon
```
At startup gives the following output:
```
00:00:00 Proyecto sonoff Living (Topic living, Fallback DVES_591719, GroupTopic sonoffs) Versión 6.2.1.8-2_4_2
00:00:00 RUL: POWER1#BOOT performs "backlog event bootpo1=0"
00:00:00 RUL: SWITCH1#BOOT performs "backlog event bootsw1=1"
00:00:00 RUL: SWITCH2#BOOT performs "backlog event bootsw2=0"
00:00:00 WIF: Connectando a AP1 NetWireless en modo 11N como living-5913...
00:00:00 RSL: stat/living/RESULT = {"Event":"Done"}
00:00:00 RSL: stat/living/RESULT = {"Event":"Done"}
00:00:00 RSL: stat/living/RESULT = {"Event":"Done"}
00:00:04 WIF: Conectado
00:00:04 HTP: Servidor web activo en living-5913 con dirección IP 192.168.1.33
00:00:06 MQT: Intentando conectar...
22:59:19 MQT: Conectado
22:59:19 MQT: tele/living/LWT = Online (Grabado)
22:59:19 MQT: cmnd/living/POWER = 
22:59:19 MQT: tele/living/INFO1 = {"Module":"Generic","Version":"6.2.1.8","FallbackTopic":"DVES_591719","GroupTopic":"sonoffs"}
22:59:19 MQT: tele/living/INFO2 = {"WebServerMode":"Admin","Hostname":"living-5913","IPAddress":"192.168.1.33"}
22:59:19 MQT: tele/living/INFO3 = {"RestartReason":"External System"}
22:59:19 MQT: stat/living/RESULT = {"POWER1":"OFF"}
22:59:19 MQT: stat/living/POWER1 = OFF
22:59:19 RUL: SYSTEM#BOOT performs "backlog event bootsys"
22:59:20 MQT: stat/living/RESULT = {"Event":"Done"}
22:59:28 MQT: tele/living/STATE = {"Time":"2018-09-26T22:59:28","Uptime":"0T00:00:18","Vcc":2.993,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":52}}
22:59:28 MQT: tele/living/SENSOR = {"Time":"2018-09-26T22:59:28","Switch1":"ON","Switch2":"OFF"}
```

Hope this helps.

closed 2018-09-27 by ascillato2

TLCary, 2018-09-27:
Stunning.  You rock.  Thanks!