arendst/Tasmota #3911: How does Button code handle pull-up vs pull-down buttons

galak, 2018-09-27:
I've been trying to read through the `ButtonHandler` code and trying to understand how it handles the case of a button that is connected to a pull-up.  It seems that `digitalRead` will return 1 when the button isn't pressed and eventually we'll `holdbutton[]` will cause us to reset.

Seems like somewhere we should be inverted the value from `digitalRead` if the button is connected to a pull-up.

I feel like I'm missing something, just not sure what it is.

closed 2018-09-27 by ascillato2

galak, 2018-09-27:
> It only works with pull-ups
> `enum ButtonStates { PRESSED, NOT_PRESSED };`
> PRESSED is 0

what does one do if they have a pull-down button?

Frogmore42, 2018-09-27:
Get a new button or define it as a switch, which can work like a button except for the hold and multiple press functions.

galak, 2018-09-27:
> Get a new button or define it as a switch, which can work like a button except for the hold and multiple press functions.

Any reason we just don't support the ability to do the invert in software for that case, and just treat it as a button still?

Frogmore42, 2018-09-27:
Since you said "we", feel free to implement it and propose a PR 😉.

You are correct that pull downs can be good, but the GPIOs on the esp8266 have built-in pull-up support, which makes it easier for people to attach just a switch and have it work.