arendst/Tasmota #3912: Transmitting command by mqtt.fx not succesfull

JoergZ2, 2018-09-27:
I'm trying to send ON and OFF commands directly by using mqtt.fx but without success. For example:
Publishing of the statement "cmnd/S20neu/power" shows the result stat/S20neu/POWER = OFF at the S20neu console. But there is no way to give the command for ON. Everytime I tried to transmit one more character than the simple status query is quitted by {Command:Unknown}. Even a blank brings this messages. What is wrong with the publishing statement like cmnd/S20neu/power = "ON"?

roguestreak, 2018-09-27:
Don't know about mqtt.fx, but mqtt is typically published as 
cmnd/S20neu/POWER on
you might need to research mqtt.fx to find out how to publish a topic (cmnd/S20neu/POWER) and a payload for that topic (on)

closed 2018-09-27 by ascillato2

curzon01, 2018-09-27:
Often make mistake when beginner using mqtt.fx:

The payload must be entered in the extra window below the topic and NOT as postfix of the topic.

This is correct:
[image]

JoergZ2, 2018-09-27:
@curzon01
Thanks!