arendst/Tasmota #3913: DS18B20 connectoed to Sonoff DUAL R2

jiripaskov, 2018-09-27:
Hello could I use Tasmota if I would like to connect DS18B20 to sonoff dual R2? I know it runs with TH version but not sure if it will be posiible with DUAL.

 Thank you.

ascillato, 2018-09-27:
Yes, it is possible.

closed 2018-09-27 by ascillato2

jiripaskov, 2018-09-27:
OK :) And how please? Tasmota is ready for this? Which pin to use for connecting the sensor?

ascillato, 2018-09-27:
All the information is on the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki):

* Connecting sensors: https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs
* Available pins in Sonoff Dual: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual-and-Dual-R2
* Support at Tasmota Discord Chat: https://discord.gg/Ks2Kzd4