arendst/Tasmota #3916: Error controlling MagicHome RGB with Home Assistant

brinzlee, 2018-09-27:
Can anyone help me decipher this error and any obvious solutions I’m missing…The RGB strip works fine but just errors in the Home Assistant log. I have asked on Home Assistant forum but no obvious solutions
This is the log from Home Assistant

Error doing job: Exception in callback <bound method MQTT._mqtt_handle_message of <homeassistant.components.mqtt.MQTT object at 0x7585adf0>>
Traceback (most recent call last):
File “uvloop/cbhandles.pyx”, line 66, in uvloop.loop.Handle._run
File “/usr/local/lib/python3.6/site-packages/homeassistant/components/mqtt/ init .py”, line 691, in _mqtt_handle_message
msg.topic, payload, msg.qos)
File “/usr/local/lib/python3.6/site-packages/homeassistant/core.py”, line 300, in async_run_job
target(*args)
File “/usr/local/lib/python3.6/site-packages/homeassistant/components/light/mqtt.py”, line 250, in brightness_received
device_value = float(templatesCONF_BRIGHTNESS)
ValueError: could not convert string to float:

I’m using this in my configuration.yaml to control some RGB strip using a MagicHome thats been Tasmota’d ver 6.1.1c

> - platform: mqtt
    name: "MagicHome RGB Strip"
    state_topic: "stat/MagicHome/RESULT"
    command_topic: "cmnd/MagicHome/POWER"
    state_value_template: "{{ value_json.POWER }}"
    brightness_state_topic: "stat/MagicHome/RESULT"
    brightness_command_topic: "cmnd/MagicHome/Dimmer"
    brightness_scale: 100
    brightness_value_template: "{{ value_json.Dimmer }}"
    white_value_state_topic: "stat/MagicHome/RESULT"
    white_value_command_topic: "cmnd/MagicHome/Channel4"
    white_value_scale: 100
    white_value_template: "{{ value_json.Channel[3] }}"
    rgb_command_topic: "cmnd/MagicHome/Color"
    rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
    rgb_state_topic: "stat/MagicHome/RESULT"
    rgb_value_template: "{{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}}"
    effect_state_topic: "stat/MagicHome/RESULT"
    effect_command_topic: "cmnd/MagicHome/Scheme"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    retain: false
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"

>

ascillato2, 2018-09-27:
Hi,

Sorry but that is a crash from your mqtt broker.

If you had no answer in the HA forum, please open an issue directly to Home Assistant.

Thanks

blakadder, 2018-11-25:
This is the response from HA github [https://github.com/home-assistant/home-assistant/issues/16933#issuecomment-425891049](url)

They claim it is Tasmota's fault

ascillato, 2018-11-25:
Hi, I use the same configuration explained in the Tasmota wiki for ws2812 and I use last Home Assistant (Hassio) with Mosquitto and works very good. No issue.

The user that claims that is a Tasmota error is not a HA developer/supporter. Is an Espurna user.

Sorry, I can not reproduce the issue with Latest Tasmota (6.3.0.12) and Latest Hassio (HassOs 2.3 + HA 0.82.1) with Latest Mosquitto AddOn from Official HA Respository.

If you are having similar MQTT Broker issue, try first to reinstall your HomeAssistant. Sometimes a faulty SD Card or a not appropiated power source for the RPI, makes this weird issues.

ascillato, 2018-11-25:
Hi

I use the same WS2812 template explained in the Tasmota [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#led-ws2812b) and works fine:

```yaml
light:
    # LED WS2812b
    - platform: mqtt
      name: "WS2812LED"
      retain: false
      state_topic: "stat/led/POWER1"
      command_topic: "cmnd/led/POWER1"
      availability_topic: "tele/led/LWT"
      qos: 1
      payload_on: "ON"
      payload_off: "OFF"
      payload_available: "Online"
      payload_not_available: "Offline"
      optimistic: false
      brightness_state_topic: "stat/led/Dimmer"
      brightness_command_topic: "cmnd/led/Dimmer"
      brightness_value_template: "{{ value_json.Dimmer }}"
      brightness_scale: 100      
      rgb_state_topic: "stat/led/Color"
      rgb_command_topic: "cmnd/led/Color2"
      rgb_command_mode: hex
      rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
      effect_state_topic: "stat/led/Scheme"
      effect_command_topic: "cmnd/led/Scheme"
      effect_value_template: "{{ value_json.Scheme }}"
      effect_list:
          - 0
          - 1
          - 2
          - 3
          - 4
          - 5
          - 6
          - 7
          - 8
          - 9
          - 10
          - 11
          - 12
```

It is using: `rgb_state_topic: "stat/led/Color"`
You are using: `stat/led/RESULT`

Using stat/led/RESULT can give several types of outputs. Was not meant to be used like that. This is a misconfiguration.

It is not a home assistant issue because it is receiving unexpected data.
And it is not a Tasmota Bug because that topic was not designed to be used like that.

Let me check the right way to make Tasmota give the feedback to Home Assistant for color, since color doesn't actually return the state from tasmota using the template from the wiki.

@blakadder 

Thanks for reporting this.

emontnemery, 2018-11-26:
> Using stat/led/RESULT can give several types of outputs. Was not meant to be used like that. This is a misconfiguration.

The data in `stat/led/RESULT` is a JSON dictionary though and is fine to use together with home assistant to get state (this is that the Hass auto discovery option willdo, and it works fine).
I suggest to clean up the wiki entry to align with what the Hass auto discovery option is using.

@ascillato2 Does that make sense?
Also, what's the difference between @ascillato2 and @ascillato ? :)

emontnemery, 2018-11-26:
@brinzlee, @blakadder What version of HomeAssistant are you using? The error you saw should be fixed by home-assistant/home-assistant#16720 which was merged in Hass 0.80

emontnemery, 2018-11-26:
@blakadder Can you please post the exact error you see as well as your Hass configuration for the light?

blakadder, 2018-11-26:
I'm using HA 0.82.0 installed in python on Linux 4.14.70-sunxi armv7l GNU/Linux.

The device is H801 with Sonoff-Tasmota 6.3.0.12 with an RGBCCT led strip connected.

Gonna put the stuff at the bottom but examining the [fix](https://github.com/home-assistant/home-assistant/pull/16720) and the log output seems that it only fixes the issue when there's no brightness value but doesn't account for the rgb_state not existing in json.

I also tried changing the config as per HA autodiscovery sent settings to:

`rgb_value_template: "{{ value_json.Color }}"`

but i get this error:

> TypeError: color_RGB_to_hs() takes 3 positional arguments but 5 were given

That's definitely HA being confused by 5 RGB values!!!

error log:
```
2018-11-26 16:17:47 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on stat/h801/RESULT: b'{"POWER":"ON"}'
2018-11-26 16:17:47 DEBUG (MainThread) [homeassistant.components.light.mqtt] Ignoring empty brightness message from 'stat/h801/RESULT'
2018-11-26 16:17:47 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'Channel' (value: {"POWER":"ON"}, template: {{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}})
2018-11-26 16:17:47 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...at 0xa1f9bfb0>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/mqtt/__init__.py", line 775, in _mqtt_handle_message
    subscription.callback, msg.topic, payload, msg.qos)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/core.py", line 327, in async_run_job
    target(*args)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/light/mqtt.py", line 331, in rgb_received
    rgb = [int(val) for val in payload.split(',')]
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/light/mqtt.py", line 331, in <listcomp>
    rgb = [int(val) for val in payload.split(',')]
ValueError: invalid literal for int() with base 10: '{"POWER":"ON"}'
```

mqtt light in configuration.yaml:
```yaml
  - platform: mqtt
    name: magic
    state_topic: "stat/h801/RESULT"
    command_topic: "cmnd/h801/POWER"
    state_value_template: "{{ value_json.POWER }}"
    brightness_state_topic: "stat/h801/RESULT"
    brightness_command_topic: "cmnd/h801/Dimmer"
    brightness_scale: 100
    brightness_value_template: "{{ value_json.Dimmer }}"
    color_temp_state_topic: "stat/h801/RESULT"
    color_temp_command_topic: "cmnd/h801/CT"
    color_temp_value_template: "{{ value_json.CT }}"
    rgb_command_topic: "cmnd/h801/Color2"
    rgb_command_template: "{{ '%02x%02x%02x0000' | format(red, green, blue)}}"
    rgb_state_topic: "stat/h801/RESULT"
    rgb_value_template: "{{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}}"
    hs_state_topic: "stat/h801/RESULT"
    hs_value_template: "{{ value_json.HSBColor.rsplit(',', 1)[0] }}"

    effect_state_topic: "stat/h801/RESULT"
    effect_value_template: "{{ value_json.Scheme }}"
    effect_command_topic: "cmnd/h801/Scheme"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    retain: false
    optimistic: false
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
```
tasmota console for good measure:
```txt 16:17:46 MQT: stat/h801/RESULT = {"POWER":"ON","Dimmer":40,"Color":"0000000066","HSBColor":"0,0,0","Channel":[0,0,0,0,40],"CT":500}
16:17:46 MQT: stat/h801/RESULT = {"POWER":"ON"}
16:17:46 MQT: stat/h801/POWER = ON (retained)```

emontnemery, 2018-11-26:
>  i get this error:
>> TypeError: color_RGB_to_hs() takes 3 positional arguments but 5 were given

That has very recently been fixed in Tasmota auto discovery, you need to tell Hass to discard the additional values for RGBW or RGBWW lights by saying: 
`rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"`

So, in total:
``` yaml
 - platform: mqtt
    name: magic
    state_topic: "stat/h801/RESULT"
    command_topic: "cmnd/h801/POWER"
    state_value_template: "{{ value_json.POWER }}"
    brightness_state_topic: "stat/h801/RESULT"
    brightness_command_topic: "cmnd/h801/Dimmer"
    brightness_scale: 100
    brightness_value_template: "{{ value_json.Dimmer }}"
    color_temp_state_topic: "stat/h801/RESULT"
    color_temp_command_topic: "cmnd/h801/CT"
    color_temp_value_template: "{{ value_json.CT }}"
    rgb_command_topic: "cmnd/h801/Color2"
    rgb_state_topic: "stat/h801/RESULT"
    rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"
    effect_state_topic: "stat/h801/RESULT"
    effect_value_template: "{{ value_json.Scheme }}"
    effect_command_topic: "cmnd/h801/Scheme"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    retain: false
    optimistic: false
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
```

There are still errors in the Hass log though, I'll have another look at what's going on here.

blakadder, 2018-11-26:
Allright! When those issues get resolved I'll clean up the wiki and make a single general HA tasmota mqtt light guide with various options

blakadder, 2018-11-27:
I tried the  `"rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"`

but i still get errors 

```
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...at 0xa607c8f0>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/mqtt/__init__.py", line 775, in _mqtt_handle_message
    subscription.callback, msg.topic, payload, msg.qos)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/core.py", line 327, in async_run_job
    target(*args)
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/light/mqtt.py", line 331, in rgb_received
    rgb = [int(val) for val in payload.split(',')]
  File "/usr/local/lib/python3.5/dist-packages/homeassistant/components/light/mqtt.py", line 331, in <listcomp>
    rgb = [int(val) for val in payload.split(',')]
ValueError: invalid literal for int() with base 10: '000000005B'
```

aren't you supposed to `"{{value_json.Color|truncate(6,false,'',0)}}"` to get 3 hex values instead of 5?

emontnemery, 2018-11-27:
If the last colour command was sent as hex, Tasmota responds in hex.
Do `setoption17 1` or send decimal colour from the console

blakadder, 2018-11-27:
Ah I see now, I was using a custom command sending FFFFFFFFFF for full white strip and that changes the json color message. Didn't know about that behaviour

emontnemery, 2018-11-28:
Ok, so it's working now?

blakadder, 2018-11-28:
All fine. I also noticed there's no SetOption17 entry in the wiki, it just says Reserved. Looking at the code its used for `Respond with decimal color values` Is there any other value it refers to or can I change it in the wiki

emontnemery, 2018-11-28:
Yes, I noticed that to.

It's an on/off setting where:
0: Respond with hex color string
1: Respond with comma-separated decimal string

ascillato2, 2018-11-28:
@emontnemery @blakadder 

Hi,

The issue is solved? Can you update the wiki? Thanks

emontnemery, 2018-11-28:
I started updating the wiki, but I was waiting for @blakadder to confirm everything working before finalizing. I'll finalize it, then it would be great with a review of the changes and configuration examples.

ascillato2, 2018-11-28:
Thanks a lot for sharing your tests, ideas and for solving this and updating the wiki :+1:

closed 2018-11-28 by ascillato2

xbmcnut, 2020-05-11:
FYI, I'm running HassOS 109.6 with Tasmota 8.2.0.6 set as an AI Light (27), RGBW Bulb with auto discovery set on for HA and although the light is working fine, I still get the error discussed in this thread and I've noted just now that setoption17 is off. Shouldn't this get activated when auto-discovery is enabled?
```
Exception in rgb_received when handling msg on 'tele/office_rgb/STATE': '{"Time":"2020-05-11T05:17:29","Uptime":"5T00:34:30","UptimeSec":434070,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":100,"Color":"0000FF00","HSBColor":"240,100,100","White":0,"Channel":[0,0,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IoT-GV40","BSSId":"02:9F:C2:F7:D8:B3","Channel":11,"RSSI":60,"Signal":-70,"LinkCount":2,"Downtime":"0T00:02:17"}}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 35, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 360, in rgb_received rgb = [int(val) for val in payload.split(",")] File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 360, in <listcomp> rgb = [int(val) for val in payload.split(",")] ValueError: invalid literal for int() with base 10: '0000FF00' 
```

effelle, 2020-05-11:
It is:  `Settings.flag.decimal_text = 1;` 
To check it I disabled `SetOption17` then after running again `SetOption19 1` it was correctly enabled, same after a reboot.

xbmcnut, 2020-05-11:
I have enabled `SetOption17 1` now (along with `SetOption 19 1` and still getting:
```
Exception in rgb_received when handling msg on 'tele/office_rgb/STATE': '{"Time":"2020-05-12T00:27:30","Uptime":"5T19:44:31","UptimeSec":503071,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":100,"Color":"0000FF00","HSBColor":"240,100,100","White":0,"Channel":[0,0,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IoT-GV40","BSSId":"02:9F:C2:F7:D8:B3","Channel":11,"RSSI":60,"Signal":-70,"LinkCount":2,"Downtime":"0T00:02:17"}}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 35, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 360, in rgb_received rgb = [int(val) for val in payload.split(",")] File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 360, in <listcomp> rgb = [int(val) for val in payload.split(",")] ValueError: invalid literal for int() with base 10: '0000FF00' 
```
Is there another option I need to check?

blakadder, 2020-05-12:
your color is still reported in hex `"Color":"0000FF00"`

xbmcnut, 2020-05-12:
Thanks for the feedback. How do I change that? I thought that's what `SetOption 17 1` did?

blakadder, 2020-05-12:
yes but if you issue a color command in hex it reverts to reporting in hex again

xbmcnut, 2020-05-12:
Thanks. I do have a couple of scripts that flash the light but I don't know if those are sending hex.
```
# alarm_light_flash_blue:
  alias: 'Blue Light using Flash'
  sequence:      
    - service: mqtt.publish
      data:
        topic: "cmnd/office_rgb/backlog"
        payload: "color 3; power 3"
        retain: false
    - delay: 00:00:07
    - service: light.turn_off
      entity_id: light.rgb_w_light
```

blakadder, 2020-05-13:
yes, color 1-12 send hex commands

xbmcnut, 2020-05-13:
Thank you. I'll have to find another way to flash for 7 secs then.

Regards,

Pete Stothers
Beach Haven, Auckland
+64.21.585.737
https://xbmcnut.blogspot.co.nz
www.youtube.com/c/PeteStothers
Sent from my Nokia 7.1. All grammatical and spelling mistakes are a feature.

blakadder, 2020-05-13:
just use decimal values instead of 3 in the `color` command

xbmcnut, 2020-05-13:
Awesome, I'll give that a try, thank you. I'm assuming that when adding to
Home Assistant using auto discovery, the standard commands in the GUI send
RGB?

Regards,

Pete Stothers
Beach Haven, Auckland
+64.21.585.737
https://xbmcnut.blogspot.co.nz
www.youtube.com/c/PeteStothers
Sent from my Nokia 7.1. All grammatical and spelling mistakes are a feature.

effelle, 2020-05-13:
With or without Discovery Tasmota will use `HSBCOLOR` command when you change the color from the GUI. What changes is `Color` format in the `/STATE` topic :
- Setoption17 enabled:
```
CMD: Group 0, Index 1, Command "HSBCOLOR", Data "227"
{"POWER":"ON","Dimmer":22,"Color":"34,39,56,0","HSBColor":"227,39,22","White":0,"Channel":[13,15,22,0]}
```
- SetOption17 disabled:
```
CMD: Group 0, Index 1, Command "HSBCOLOR", Data "201"
{"POWER":"ON","Dimmer":22,"Color":"22303800","HSBColor":"201,39,22","White":0,"Channel":[13,19,22,0]}
```
You can obviously issue directly the `color` command from mqtt or console, like `color 34,39,56,0`.

emontnemery, 2020-05-13:
I think the Tasmota functionality to change the color format in responses according to last format of data in color command is both error prone and confusing.
Maybe change or update `setoption17` to disable the autodetect?

blakadder, 2020-05-13:
That issue has been raised and closed a few times already