arendst/Tasmota #3918: "SaveData Off" causes CRC config errors.

KPK75, 2018-09-28:
**To Reproduce**
1) `savedata off`
2) Now make change that requires flash save and restart ( for example change module type )

After restart you get CRC error from decode-config.py script and saved configs won't restore ( "file invalid" ).
After `savedata on` command script and restore working fine

arendst, 2018-09-28:
@curzon01 Norbert a nice one for you...

curzon01, 2018-09-28:
Theo, pls have a look in parallel - decode-config.py only reads the data and in situation above the crc calculated by decode-config.py does not match the crc from `struct SYSCFG` `uint16_t cfg_crc` field.

Example (situation SaveData ON)
1. decode-config.py calculated crc_data = struct crc_data - OK
2. `SaveData OFF`
3. decode-config.py calculated crc_data = struct crc_data - OK
4. restart Sonoff
5. decode-config.py calculated crc_data &ne; struct crc_data - ERROR (e.g. crc_data in struct is 0x6293, calculated is 0x62a0)
6. restart Sonoff again (!) - be aware, this step without any changes, SaveData is still OFF
7. decode-config.py calculated crc_data = struct crc_data - OK

arendst, 2018-09-29:
- the crc over the current settings is calculated when the settings need to be saved to flash.
- if the calculated crc differs from the stored crc the settings will be saved in the next flash slot.
- this check is perfomed every savedata seconds.
- when savedata is set to 0 no more crc's are calculated or written to flash.
- decode_config.py pulls a config from the device using webserver backup configuration. This config is a copy of the active settings.
- when savedata is set to 0 and a change is made like switchmode 5 the crc is not updated and decode_config.py will fail.
- with every reboot the bootcount is incremented and therefor will crc error at next reboot.
- the next reboot will increment bootcount too and will not crc error probably because the crc code does not detect double bits in same byte.

As a solution I will add updated crc code to webserver config backup but keep it out of the current settings as it interferes with setting savedata to >0 again which could then fail to store changes.

curzon01, 2018-09-29:
Theo, I thought something like that.

New version of 'decode-config.py' handle CRC and data len mismatch as warning and a new arg '--exit-on-error-only'  allows to proceed even when a warning (like CRC error) happens.

Just made a PR...

closed 2018-09-29 by arendst