arendst/Tasmota #3922: Device Relay keeps turning ON and OFF when loss to broker

fals3illusion, 2018-09-28:
I have tried to search for this but came up with nothing.  

 I created a Garage Door opener with a relay /door sensors/ etc and whenever there is loss to my MQTT broker ( restart raspberry pi) the relay keeps turning on and off .  This causes the garage door to start opening and closing until the connection is established again. 

```
16:33:22 MQT: Attempting connection...
16:33:22 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:33:25 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:33:25 RSL: tasmota/stat/ESP1/POWER = ON
16:33:26 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:33:26 RSL: tasmota/stat/ESP1/POWER = OFF
16:33:30 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:33:30 RSL: tasmota/stat/ESP1/POWER = OFF
16:33:33 MQT: Attempting connection...
16:33:33 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:33:35 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:33:35 RSL: tasmota/stat/ESP1/POWER = ON
16:33:36 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:33:36 RSL: tasmota/stat/ESP1/POWER = OFF
16:33:44 MQT: Attempting connection...
16:33:44 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:33:47 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:33:47 RSL: tasmota/stat/ESP1/POWER = OFF
16:33:55 MQT: Attempting connection...
16:33:55 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:04 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:34:04 RSL: tasmota/stat/ESP1/POWER = ON
16:34:05 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:05 RSL: tasmota/stat/ESP1/POWER = OFF
16:34:06 MQT: Attempting connection...
16:34:06 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:09 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:09 RSL: tasmota/stat/ESP1/POWER = OFF
16:34:17 MQT: Attempting connection...
16:34:17 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:28 MQT: Attempting connection...
16:34:28 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:36 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:34:36 RSL: tasmota/stat/ESP1/POWER = ON
16:34:37 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:37 RSL: tasmota/stat/ESP1/POWER = OFF
16:34:39 MQT: Attempting connection...
16:34:39 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:41 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:41 RSL: tasmota/stat/ESP1/POWER = OFF
16:34:45 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:34:45 RSL: tasmota/stat/ESP1/POWER = ON
16:34:46 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:46 RSL: tasmota/stat/ESP1/POWER = OFF
16:34:50 MQT: Attempting connection...
16:34:50 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:34:58 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:34:58 RSL: tasmota/stat/ESP1/POWER = OFF
16:35:01 MQT: Attempting connection...
16:35:01 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:35:01 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:35:01 RSL: tasmota/stat/ESP1/POWER = ON
16:35:02 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:35:02 RSL: tasmota/stat/ESP1/POWER = OFF
16:35:03 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:35:03 RSL: tasmota/stat/ESP1/POWER = OFF
16:35:12 MQT: Attempting connection...
16:35:12 MQT: Connect failed to 192.168.1.108:1883, rc -2. Retry in 10 sec
16:35:20 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"ON"}
16:35:20 RSL: tasmota/stat/ESP1/POWER = ON
16:35:21 RSL: tasmota/stat/ESP1/RESULT = {"POWER":"OFF"}
16:35:21 RSL: tasmota/stat/ESP1/POWER = OFF
16:35:23 MQT: Attempting connection...
```
^^^^^^^
This keeps going on until connection is established.  

I have my settings as PulseTime1 10.  I have tried SwitchRetain on/off, PowerRetain on/off , PowerOnState 0/1 and nothing is helping. 

It keeps on turning on and off and I am stumped. 

 Any Help would be greatly appreciated!
Thank You!

ascillato, 2018-09-28:
hehehe, you should make a video of that hehehe

Ok, That is because you are using switchtopic and that feature is not intended for your reeds.

Switchtopic was designed for actual wall switches, so in the case you lost the broker, you could continue to use your relays.

You need to use rules in order to do what you expect.

Adapt the following example to your situation. Go to your console and type:

switchtopic1 0
rule1 1
rule1 on switch1#state do publish cmnd/sonofftopic/power %value% endon

More information is available in the wiki at rules (with examples)

If you need more help, please find us in the Tasmota Discord Chat. Thanks

meingraham, 2018-09-28:
I was having this exact issue. For me it was a weak Wi-Fi signal. It was actually that the device was connecting to the wrong AP in our house (the one at the far end of the house from where the Sonoff is). That still happens occasionally, but now I monitor which AP each device is connected to so I can be alerted when it connects to the wrong one. If you address the cause, the symptom of the toggling (probably due to the retained MQTT messages). Check out this video for the retain issue - https://www.youtube.com/watch?v=31IyfM1gygo

ascillato, 2018-09-28:
You are welcome.

Please, also check the video in the link that @meingraham has posted. If you have used powerretain, you could have the retained messages issue also.

closed 2018-09-28 by fals3illusion