arendst/Tasmota #3925: I2cScan command is returning invalid JSON and an unknown error

andrethomas, 2018-09-28:
**Describe the bug**
Incorrect formatting in JSON message for i2cscan response.

The response is not formatted correctly - it outputs as:
`23:38:47 RSL: stat/sonoff/RESULT = {"I2CScan":"Unknown error at 0x7f"}"}`

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : 
- [x] _Development/Compiler/Upload tools used_ :
- [x] _Provide the output of command_``status 0`` :
```
23:44:01 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:44:01 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:56","StartupUTC":"2018-09-28T21:37:05","Sleep":0,"BootCount":3,"SaveCount":7,"SaveAddress":"3F5000"}}
23:44:01 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.9","BuildDateTime":"2018-09-28T23:32:44","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:44:01 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wireless",""],"TelePeriod":300,"SetOption":["00008008","55818000","00000000"]}}
23:44:01 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":439,"Free":2632,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F000684","00000001","20000596","000003D1"]}}
23:44:01 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4052","IPAddress":"192.168.42.42","Gateway":"192.168.32.1","Subnetmask":"255.255.240.0","DNSServer":"192.168.32.1","Mac":"DE:AD:00:00:BE:EF","Webserver":2,"WifiConfig":2}}
23:44:01 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_102FD4","MqttUser":"sensor","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:44:01 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 28 21:44:01 2018","Local":"Fri Sep 28 23:44:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":2,"Sunrise":"05:44","Sunset":"17:59"}}
23:44:01 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-28T23:44:01"}}
23:44:01 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-28T23:44:01","Uptime":"0T00:06:56","Vcc":2.737,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"DE:AD:00:00:BE:EF","Channel":3,"RSSI":86}}}
```

**To Reproduce**
Connect PN532 I2C slave and perform i2cscan from the console - This may also happen in other cases of i2c devices but I am unable to reproduce with a different device.

**Expected behavior**
i2cscan output should be
`23:38:47 RSL: stat/sonoff/RESULT = {"I2CScan":"Unknown error at 0x7f"}`
But in addition to that it is not immediately clear what the unknown error part means, so I guess we'll need to dig as to why this specific sensor is responding in a way that Tasmota is recognising as an error.

andrethomas, 2018-09-28:
I think the error is as a result of no pull-up resistors on the PN532 board because it should be on 0x24 and when I add a MCP230xx to the same I2C bus I get
`00:24:05 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x20 0x24"}`
So then the pullups provided by the MCP chip is working fine and causing the I2C bus to behave like it should.

The issue of the extra appendage on the JSON if an error is detected should be fixed though, I guess.

andrethomas, 2018-09-28:
So the appendage is added in
void I2cScan(char *devs, unsigned int devs_len)
{
  byte error;
  byte address;
  byte any = 0;
  char tstr[10];

  snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_DEVICES_FOUND_AT));
  for (address = 1; address <= 127; address++) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (0 == error) {
      snprintf_P(tstr, sizeof(tstr), PSTR(" 0x%2x"), address);
      strncat(devs, tstr, devs_len);
      any = 1;
    }
    else if (4 == error) {
      snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_UNKNOWN_ERROR_AT " 0x%2x\"}"), address);
    }
  }
  if (any) {
    strncat(devs, "\"}", devs_len);
  } else {
    snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_NO_DEVICES_FOUND "\"}"));
  }
}

@arendst How would you like to solve this? I do not want to set any to 0 because that would exclude other I2C devices which were detected correctly so I think its just
```
snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_UNKNOWN_ERROR_AT " 0x%2x\"}"), address);
```

that needs to be reviewed.

Otherwise, if we find an error on the I2C bus just reports nothing found?

mike2nl, 2018-09-29:
I think a new sensor agan.

arendst, 2018-09-29:
Why not set any to 0?

What happened here is that a I2C scan from 1 to 127 (=0x7f) went fine but device 127 returned a 4 on wire.endtransmission. I will have to look up what that means in the wire library but it's probably something like UNKNOWN ERROR ;-)

As you noticed it might be a pull-up problem and in that case the whole I2C bus is in jeopardy so it makes no sense to report valid detected addresses as they won't be able to get used due to bad bus impedance.

arendst, 2018-09-29:
The error is probably I2C_SDA_HELD_LOW_AFTER_INIT as defined in twi.h:
```
#define I2C_OK                      0
#define I2C_SCL_HELD_LOW            1
#define I2C_SCL_HELD_LOW_AFTER_READ 2
#define I2C_SDA_HELD_LOW            3
#define I2C_SDA_HELD_LOW_AFTER_INIT 4
```

arendst, 2018-09-29:
And then it might be different...
```
unsigned char twi_writeTo(unsigned char address, unsigned char * buf, unsigned int len, unsigned char sendStop){
  unsigned int i;
  if(!twi_write_start()) 
    return 4;//line busy
  if(!twi_write_byte(((address << 1) | 0) & 0xFF)) {
    if (sendStop) 
      twi_write_stop();
    return 2; //received NACK on transmit of address
  }
  for(i=0; i<len; i++) {
    if(!twi_write_byte(buf[i])) {
      if (sendStop) 
        twi_write_stop();
      return 3;//received NACK on transmit of data
    }
  }
  if(sendStop) 
    twi_write_stop();
  i = 0;
  while(SDA_READ() == 0 && (i++) < 10){
    SCL_LOW();
    twi_delay(twi_dcount);
    SCL_HIGH();
    twi_delay(twi_dcount);
  }
  return 0;
}
```
Although what is in a name. A busy might well be caused by I2C_SDA_HELD_LOW_AFTER_INIT...

arendst, 2018-09-29:
And final info from core_esp8266_si2c.c
```
uint8_t twi_status() {           
    if (SCL_READ()==0)     
        return I2C_SCL_HELD_LOW;             //SCL held low by another device, no procedure available to recover
    int clockCount = 20;                   

    while (SDA_READ()==0 && clockCount>0) {                     //if SDA low, read the bits slaves have to sent to a max
        --clockCount;
        twi_read_bit();                    
        if (SCL_READ()==0) 
            return I2C_SCL_HELD_LOW_AFTER_READ;  //I2C bus error. SCL held low beyond slave clock stretch time
    }

    if (SDA_READ()==0)     
        return I2C_SDA_HELD_LOW;             //I2C bus error. SDA line held low by slave/another_master after n bits.

    if(!twi_write_start()) 
        return I2C_SDA_HELD_LOW_AFTER_INIT;  //line busy. SDA again held low by another device. 2nd master?

    return I2C_OK;                       //all ok

}
```

arendst, 2018-09-29:
Concluding. 

I'll update the code and report the error number as defined in twi.h:
```
void I2cScan(char *devs, unsigned int devs_len)
{
  // Return error codes defined in twi.h and core_esp8266_si2c.c
  // I2C_OK                      0
  // I2C_SCL_HELD_LOW            1 = SCL held low by another device, no procedure available to recover
  // I2C_SCL_HELD_LOW_AFTER_READ 2 = I2C bus error. SCL held low beyond slave clock stretch time
  // I2C_SDA_HELD_LOW            3 = I2C bus error. SDA line held low by slave/another_master after n bits
  // I2C_SDA_HELD_LOW_AFTER_INIT 4 = line busy. SDA again held low by another device. 2nd master?

  byte error;
  byte address;
  byte any = 0;
  char tstr[10];

  snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_DEVICES_FOUND_AT));
  for (address = 1; address <= 127; address++) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (0 == error) {
      snprintf_P(tstr, sizeof(tstr), PSTR(" 0x%2x"), address);
      strncat(devs, tstr, devs_len);
      any = 1;
    }
    else {
      any = 2;
      snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"Error %d at 0x%2x"), error, address);
      break;
    }
  }
  if (any) {
    strncat(devs, "\"}", devs_len);
  }
  else {
    snprintf_P(devs, devs_len, PSTR("{\"" D_CMND_I2CSCAN "\":\"" D_JSON_I2CSCAN_NO_DEVICES_FOUND "\"}"));
  }
}
```

andrethomas, 2018-09-29:
Confirm fixed

```
14:29:22 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x24"}       // Correct
14:29:23 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x20 0x24"}  // Correct
14:29:24 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x24"}       // Correct
14:29:25 MQT: stat/sonoff/RESULT = {"I2CScan":"Error 4 at 0x26"}               // Correct, error !;)
```

closed 2018-09-29 by andrethomas

andrethomas, 2018-09-29:
@arendst I'm surprised you didn't see the MAC address in the status 0 ;)

arendst, 2018-09-29:
I'm glad you did ;-). Wonder what's in all those other statuses....