arendst/Tasmota #3928: Electrodragon(dual relay) with Capacitive touch sensor on gpio

amitubale, 2018-09-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : Electrodragon Wifi Switch9dual relay)
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
``03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS = {"Status":{"Module":15,"FriendlyName":["Master Bath Toilet Light","Master Bath Toilet Fan"],"Topic":"masterbathtoiletlightfan","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff-rf1.ino.bin","RestartReason":"External System","Uptime":"0T00:00:31","StartupUTC":"2018-09-29T01:24:29","Sleep":0,"BootCount":16,"SaveCount":54,"SaveAddress":"F8000"}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["MyHomeWifiSSID","MyHomeWifiSSID_Ext"],"TelePeriod":300,"SetOption":["000A0109","55A18000","00000001"]}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS5 = {"StatusNET":{"Hostname":"masterbathtoiletlightfan-5707","IPAddress":"192.168.5.250","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.5.1","Mac":"C0:20:C6:12:76:CB","Webserver":2,"WifiConfig":2}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.36","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32764B","MqttUser":"username","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 29 01:25:00 2018","Local":"Sat Sep 29 03:25:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:46","Sunset":"19:34"}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS10 = {"StatusSNS":{"Time":"2018-09-29T03:25:00","Switch1":"ON","AM2301":{"Temperature":75.9,"Humidity":62.3},"TempUnit":"F"}}
03:25:00 MQT: stat/masterbathtoiletlightfan/STATUS11 = {"StatusSTS":{"Time":"2018-09-29T03:25:00","Uptime":"0T00:00:31","Vcc":2.914,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"MyHomeWifiSSID","RSSI":96,"APMac":"30:FD:38:FA:C8:38"}}}`

**(Please, remember to close the issue when the problem has been addressed)**

I have connected the capacitive touch sensors on GPIO 4 and 5. and configured it to "09 Switch1" on the wes console -->configure-->configure module. for both GPIO 4 and 5. 
capacitive touch i am using is from ali express -"Blue HTTM-SCB 2.7V-6V Capacitive Touch Switch Module Strong anti-interference"
the OUT pins of 2 sensors are  connected to GPIO 4 and 5 resp.

The sensor light is turin on and off when touched. What should I configure theGPIOs to or what command i need to type on the console. DrZZZ has a video on the sonoff basic with the same sensor but i tried 'switchmode1 2' and it did not work.

since there are 2 relays, they show up in Mqtt as POWER1 and POWER2. If i need to type the command in the console do i need to change the mode for both ? please let me know the commands. thank you!

ascillato2, 2018-09-29:
Hi,

> The sensor light is turin on and off when touched. What should I configure theGPIOs to or what command i need to type on the console. 

What exactly you want to do? Please explain what is the issue.

amitubale, 2018-09-29:
the sensor is working that is it turns on and off but it does not have any effect on the 2 relays.
Can i use the GPIO's 4 and 5 to drive the relay1 and relay2.
what shoudl be the configuration on the GPIOs 4 and 5 to drive the relays. thanks

ascillato, 2018-09-29:
Hi, I'm so sorry. I don't understand.

You have 2 capacitive sensors connected to gpio 4 and gpio 5. It is Right? Have you also connected gnd? 

You have set gpio 4 as button1 in module config (tasmota web menú). It is Right?
You have set gpio 5 as button2 in module config (Tasmota web menú). It is Right?

Frogmore42, 2018-09-29:
The Electrodragon already has two buttons (KEY1 and KEY2) defined.  I don't know if you can have two GPIOs with the same definition or not.  If you can, that might work.

ascillato, 2018-09-29:
No, you can't have two gpio with the same definition.

I'm asking @amitubale how it is connected and configured because it is not clear in the description. 

Please, try the generic module instead and set all your pins there

amitubale, 2018-09-29:
the capasitive sensor had +,- and OUT. all 3 wires are connected both OUT pins of 2 sensors are connected to GPIO 4 and GPIO5. when i touch the sensors the light on the sensor does show that it turning on and off.
but it does not have any effect on the Relays. i have configures GPIO4 to '09 switch1' and GPIO5 to '10 switch2' and the relays are not turing on or off. To test i have tried the push button on the PCB and it works.  Is there a switchmode command i need to configure the GPIOs to control the relays ?

amitubale, 2018-09-29:
HI ascillato, what do you mean by 'As you have a hardware modification" ?

I have not modified the hardware. 

i am using the gpio4, 5 (to touch sensor) and gpio 15 to PIR wich are on the PCB. i also connected dht22 on the connecor provided on the pcb( gpio14- and its working )

Do you mean that the version of electrodragon i have is not he same as the the one thats in the 6.2.1 version of tasmota code ?....so it was coded for a previous version of the the eletrodragon hardware ?

ascillato, 2018-09-29:
> HI ascillato, what do you mean by 'As you have a hardware modification" ?

I meant that Chosing ELECTRODRAGON from module config will work for you ONLY if you use ELECTRODRAGON as standard (as it cames from factory)
Adding switches to your device is a Hardware modification, so you need to chose module generic and set the GPIO as:

GPIO4 as Switch1
GPIO5 as Switch2
GPIO12 as Relay2
GPIO13 as Relay1
GPIO16 as Led1

Please, Try that.

ascillato, 2018-09-29:
Sorry but you need to be more specific in order to let me help you. The **_didn't work phrase_** lack a lot of information. What happened?, what you see in the console output? what switchmode and switchtopic did you use?

amitubale, 2018-09-29:
sorry i did not use the console commands. I configured the following
GPIO4 as Switch1
GPIO5 as Switch2
GPIO12 as Relay2
GPIO13 as Relay1
GPIO16 as Led1

in the web configure module and then tested the touch sensors. the touch senors aer working but did not drive the relay. is there a swithmode1 x (& swithmode2 x) command i need to type in the console window ?

ascillato, 2018-09-29:
Are you connecting GND besides GPIO4 and GPIO5?

amitubale, 2018-09-29:
there are 2 GND pins on the 12 pins on the pcb and yes i am using the GND pin not the one next to 4 and5  but the one next to 3V3 and 5V pins...i am guessing the GND pin for both are the same. I will ceheck it with a multimeter on another spare electrodragon module. I appreciate you helping me out with this issue.

ascillato, 2018-09-29:
May be there is something misconfigured. Try `switchtopic1 0` and `switchtopic2 0` in the console

amitubale, 2018-09-29:
10:41:05 CMD: switchtopic1 0
10:41:05 MQT: stat/masterbathtoiletlightfan/RESULT = {"SwitchTopic":""}
10:41:25 CMD: switchtopic2 0
10:41:25 MQT: stat/masterbathtoiletlightfan/RESULT = {"SwitchTopic":""}
did that no luck.
I need to lookup the diffrence between switchmode and switchtopic.
I had tried it before with "switchmode1 2" as it was on the youtube tutorial for sonoff basic module.
i had a PIR on GPIO15 and i had set is as switch3 and used switchtopic3 pirmotion....but could not see anyactivity on the console ..but that i will tackle once i get the touch switch to work

amitubale, 2018-09-29:
10:41:38 MQT: tele/masterbathtoiletlightfan/STATE = {"Time":"2018-09-29T10:41:38","Uptime":"0T00:31:35","Vcc":3.103,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"myssid","RSSI":98,"APMac":"31:CA:38:AC:C1:31"}}
10:41:38 MQT: tele/masterbathtoiletlightfan/SENSOR = {"Time":"2018-09-29T10:41:38","Switch1":"ON","Switch2":"ON","Switch3":"OFF","AM2301":{"Temperature":73.8,"Humidity":65.3},"TempUnit":"F"}
10:41:59 MQT: stat/masterbathtoiletlightfan/RESULT = {"POWER2":"ON"} --> this is with push buttons on PCB(and also work from web main page)
10:41:59 MQT: stat/masterbathtoiletlightfan/POWER2 = ON
10:42:01 MQT: stat/masterbathtoiletlightfan/RESULT = {"POWER1":"ON"} --> this is with push buttons on PCB and (also work from web main page) 
10:42:01 MQT: stat/masterbathtoiletlightfan/POWER1 = ON

ascillato, 2018-09-29:
if the pushbuttons are working, you didn't use the generic module nor the config you told me:

GPIO4 as Switch1
GPIO5 as Switch2
GPIO12 as Relay2
GPIO13 as Relay1
GPIO16 as Led1

Buttons from the PCB are not configured there.

So, what is the REAL configuration are you using?

amitubale, 2018-09-29:
after i tried your config ( the pcb buttons did not work) so after it failed, i put it back to Electrodragon module. Let me put it back on generic and try the switctopic command

ascillato, 2018-09-29:
Ok, let's continue this in the [Tasmota Discord Chat](https://discord.gg/Ks2Kzd4). The communication will be more dynamic.

amitubale, 2018-09-29:
sounds good

ascillato, 2018-09-29:
From the Chat:

Measuring with a DMM the output of the Capacitive sensors, those gives 2.2V(when off) and 3.3v(when on). So, adding a 4.7Kohm resistor between the GPIO4 and GND, solves the issue.

closed 2018-09-29 by ascillato2

TLCary, 2019-04-06:
The output from the capacitive touch is +3V3, (connected to an input pin configured as a switch).  Switches work by pulling the input pin to GND, not by pulling them UP.  So, what's the easiest way to fix this?  I guess you could invert the signal coming out of the sensor with a single-gate inverter, (one NPN transistor and two resistors), but if there are suggestions on an easier approach, I'd love to hear them.

TLCary, 2019-05-05:
I found that the knock-off Heltec HTTM Capacitive Touch sensors (blank white back with “touch button” printed on the back) have this problem.  The real Heltec with green screen printed logo on the back work fine.