arendst/Tasmota #3930: T1 StoreProhibited exception afer successfull fingerprint TLS verify

s00500, 2018-09-29:
**Describe the bug**
I just flashed a Sonoff T1 (EU) with the Version 6.2.1.9-2_3_0 and entered my mqtt server and fingerprint. It boots, connects to wifi, verifys the fingerprint and then fails with the following exception

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X] _Searched the problem in issues and in the wiki_
- [ X] _Hardware used_ : Sonoff T1 and T1 Dual EU
- [X ] _Development/Compiler/Upload tools used_ : PlatformIO
- [X ] _Provide the output of command_``status 0`` : **(This is before it could connect)**
```
00:00:02 CMD: status 0
00:00:02 RSL: stat/sonoff/STATUS = {"Status":{"Module":28,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:02 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:00:02","StartupUTC":"","Sleep":0,"BootCount":83,"SaveCount":87,"SaveAddress":"F8000"}}
00:00:02 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.9","BuildDateTime":"2018-09-29T12:48:55","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:02 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BachiNet",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
00:00:02 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":455,"Free":548,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE754","240003A0","23B617CE","000003C0"]}}
00:00:02 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0317","IPAddress":"0.0.0.0","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.245","Mac":"60:01:94:AD:A1:3D","Webserver":0,"WifiConfig":5}}
00:00:02 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"roomserver.lbs","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ADA13D","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:02 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:02 1970","Local":"Thu Jan 01 00:00:02 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:02 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:02"}}
00:00:02 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:02","Uptime":"0T00:00:02","Vcc":3.122,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BachiNet","BSSId":"44:D1:FA:1E:A9:85","Channel":12,"RSSI":100}}}
```

**To Reproduce**
Flash a Sonoff T1 with the Version 6.2.1.9-2_3_0 and try a mqtt TLS connection

**Expected behavior**
Stacktrace

```
Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff4410 end: 3fff49a0 offset: 01a0

>>>stack>>>
3fff45b0:  00000040 00000082 3fff602c 40263785  
3fff45c0:  3fffb374 3fff7cb4 00000000 00000041  
3fff45d0:  00000040 0000003f 00000000 00000041  
3fff45e0:  3fffbcac 00000000 3fff6044 3fff5fac  
3fff45f0:  3fffbda8 3fff605c 3fff7cb4 3fff7cb4  
3fff4600:  00000040 3fff7cb4 3fff602c 3fff7cb4  
3fff4610:  00000040 3fff7cb4 3fff602c 402642ba  
3fff4620:  3fff8754 0000003f f0452560 00000020  
3fff4630:  3fff600c 3fff7cb4 3fff5fdc 00000020  
3fff4640:  3fff600c 3fff7cb4 00000010 402645a0  
3fff4650:  3fff87cc 3fff8784 3fff5fdc 00000001  
3fff4660:  00000001 3fff5fdc 3fff8ede 40263378  
3fff4670:  00000100 3fff8724 3fff8e57 00000000  
3fff4680:  00000100 3fff8724 3fff8e57 402655a1  
3fff4690:  3fff46c0 00000000 000000d0 00000030  
3fff46a0:  2c3aa14f 2f6a91ea 6edfc2f8 00000004  
3fff46b0:  00000004 3fff7d14 3fff8e51 402616ac  
3fff46c0:  47520203 7fab84fc 47181f7c 9134fcbf  
3fff46d0:  12691f9d b8e09fa0 5c9acd7a e0c6fb4b  
3fff46e0:  ae232939 02d995b6 676341b5 a14f1994  
3fff46f0:  00000004 3fff8e51 3fff7d14 40261aad  
3fff4700:  6a5d98fe b7c5f583 f8fe0b95 3fff5d74  
3fff4710:  00000000 00000004 00000004 402031ef  
3fff4720:  3fff8e55 3fff8e51 3fffa93c 00000300  
3fff4730:  00000398 3fff91be 0000002a 00000004  
3fff4740:  3fff7d14 3fff8e51 00000004 00000004  
3fff4750:  00000004 3fff8e51 3fff7d14 402614a4  
3fff4760:  00000000 3fff8e51 3fff7d14 4026184c  
3fff4770:  0000000e 3fff2b21 0000000f 4020b518  
3fff4780:  3fff47d0 0000000e 00000010 00000000  
3fff4790:  3fff60dc 3fff2b21 3fff7d14 01000000  
3fff47a0:  3fff47d0 3fff2850 3fff7d14 402615fc  
3fff47b0:  40202fd0 20423020 3fff3980 00001387  
3fff47c0:  00001d6f 3fff2850 3fff879c 4020368d  
[... 54 lines cut ...]
0x40202fd0: WiFiClient::_s_connected(void*, void*, signed char) at ?? line ?
0x4020368d: WiFiClientSecure::_connectSSL(char const*) at ?? line ?
0x40202a46: WiFiClient::connect(IPAddress, unsigned short) at ?? line ?
0x402039d9: WiFiClientSecure::connect(char const*, unsigned short) at ?? line ?
0x4020181d: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at ?? line ?
0x40201cd5: std::enable_if   , std::is_convertible   &>)()))((declval )(), (declval )(), (declval )())), void> >::value, std::function &>::type std::function ::operator=  &>(std::function &) at ?? line ?
0x40222c2f: MqttReconnect() at ?? line ?
0x40233c66: wifi_get_opmode at ?? line ?
0x40107642: __digitalRead at ?? line ?
0x402024f7: ESP8266WiFiSTAClass::localIP() at ?? line ?
0x40220654: ButtonHandler() at ?? line ?
0x4021ea40: MqttDataHandler(char*, unsigned char*, unsigned int) at ?? line ?
0x40223220: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ?? line ?
0x40223208: std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, unsigned char*, unsigned int) at ?? line ?
0x40234abf: wifi_station_get_connect_status at ?? line ?
0x40222c78: MqttCheck() at ?? line ?
0x402231a0: StateLoop() at ?? line ?
0x4020fbac: MqttLoop() at ?? line ?
0x40209a04: esp_yield at ?? line ?
0x4021e8f6: Xdrv01(unsigned char) at ?? line ?
0x402231e8: loop at ?? line ?
0x40209a50: loop_wrapper() at core_esp8266_main.cpp line ?
0x4020b7bc: cont_norm at cont.S.o line ?

```

arendst, 2018-09-29:
Most likely running out-of-memory as TLS is very memory hungry.

s00500, 2018-09-29:
Damn I am sorry, just found this..... is there really no way at all to get around this? what measures can I take to remove as much stuff from ram as possible? #2845

ascillato, 2018-09-29:
Try to disable all sensors and all features in user_config.h

ascillato, 2018-09-29:
If that does not work for you, would be better to change to a local MQTT Broker. That is better suited for home automation. Web Broker is not reliable.

Jason2866, 2018-09-30:
Try core 2.4.2 it gives you more free RAM. Just sleep isnt working anymore.
What have you disabled?

s00500, 2018-09-30:
Ok, so I got it working after all, disabeled everything that makes no sense in a wall switch anyway: ws2818, all kinds of sensor support, the KNX stuff (since I do not need that) and more...

@ascillato Well I do use a local broker but SSL should be in place still in my opinion...

Closing this as it is kind of solved.... In case anyone runs into this disable as much as you can in the user_config or override

closed 2018-09-30 by s00500

eneghawi, 2018-10-07:
Hey S00500,

Can you please provide you userconfig.h so I know what you have removed... Thanks!

s00500, 2018-10-07:
@eneghawi 
[user_config_s00500.txt](https://github.com/arendst/Sonoff-Tasmota/files/2454190/user_config_s00500.txt)

eneghawi, 2018-10-10:
Thanks! I will test it. I was using my own development for this on different hardware but more expensive. This one looks cheaper and easier :) 

Thanks!