arendst/Tasmota #3934: BUG: Build 6.2.1.9 20180928 does not work with Sonoff Basic

ChrSch84, 2018-09-29:
**Describe the bug**
_With the Update to Build 6.2.1.9 20180928 Sonoff.bin my Sonoff Basic does not start anymore, with 6.2.1.7 all was ok.  My Sonoff Pow work with 6.2.1.9. Only Responds i get is a Blinking LED. With the Build "sonoff-classic" the Basic Sonoff is working fine.

With a Reflash with 6.2.1 Sonoff.bin all is Normal again. Used for all of this Problems ESPTool Version  2.5.1.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ :Sonoff Basic Revision 1 
- [ ] _Development/Compiler/Upload tools used_ : esptool 2.5.1
- [ ] _Provide the output of command_``status 0`` :
```
Could not get a output because device does not work.
```

**To Reproduce**
_Flash with Sonoff.bin from "sonoff.bin | http://thehackbox.org/tasmota/sonoff.bin | 479k | 20180929 15:00 GMT+2 | 53acb34" 

**Additional context**
esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 "sonoff (6.2.1).bin"
esptool.py v2.5.1
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: ec:fa:bc:86:a2:08
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 482512 bytes to 330006...
Wrote 482512 bytes (330006 compressed) at 0x00000000 in 29.3 seconds (effective 131.6 kbit/s)...
Hash of data verified.

Went Smooth but no funktion from the Device.

esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff-classic.bin
esptool.py v2.5.1
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: ec:fa:bc:86:a2:08
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 492400 bytes to 340158...
Wrote 492400 bytes (340158 compressed) at 0x00000000 in 30.2 seconds (effective 130.3 kbit/s)...
Hash of data verified.

Is Working fine on the Basic.

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-09-29:
How are you powering it?
The Basic devices seem to be particularly limited in their power supply and hard to power.

You will need to capture the serial output from the device while you power it up (without plugging it into the wall).  This might show something.

If you are using an FTDI to power it that doesn't have an external regulator, that is likely part of the problem.

Since the same software works fine on other devices, it points to a problem on this device and the Basic is known to have a marginal at best power supply.

Jason2866, 2018-09-29:
Cant reproduce. Works fine.
[image]

ascillato, 2018-09-29:
Hi

Sorry, I can't reproduce your issue. Working fine for me.

For help visit us as https://discord.gg/Ks2Kzd4

closed 2018-09-29 by ascillato2

ChrSch84, 2018-09-29:
I could reproduce this on this device every time. If I update from 6.2.1 or 6.2.1.8 to. 9 the device stops working.
I could not get any serial output from the device after the update.

It does not matter if I use serial update with rap tool or the build in update from tasmota. As I say after update to. 9 the device bricks. My s20 and 3 pow work fine with the new version.

ascillato, 2018-09-29:
Sorry,

I'm compiling right now, and also the automatic builds shows:

So, please, check your build environment or use the precompiled bins

ChrSch84, 2018-09-30:
I never wrote that I built the firmware myself I use the bin file from the link. As I said sonoff classic and sonoff minimal is working from. 9, but if I update to sonoff.bin with seriell or build in update the device fails. If I use. 8 all is fine. And it only effects the basic model my th pow and s 20 is working with. 9. But OK I wait for. 10 and test again.

Jason2866, 2018-09-30:
Have you erased the whole flash before? Use esptool 
esptool.py --port COM5 erase_flash
After that flash your firmware
esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff-classic.bin

ChrSch84, 2018-09-30:
yes, as i say sonoff-classic works but not sonoff.bin. thats the part what buged me :(

For now i stay on 6.2.1 for this sonoff basic

Jason2866, 2018-09-30:
Sorry, since we cant reproduce your issue we cant search for.

ChrSch84, 2018-10-13:
Test with the third Sonoff Basic from Stock with ESPTool. No luck. Same Problem no Wifi Connection ..Nothing". Then Flash with "6.2.1" all is fine. Upgrade to 6.2.1.14 or .15 no connection to wifi. 

BUT i got on my sonoff because it is in a boot loop and i connect direct to the device with 192.168.4.1 with "192.168.4.1/up?" i could go to the Firmware Update Page and could flash BACK to 6.2.1 and TATA all is fine again.

So i dont know what it is. I Stay on 6.2.1 on my critical Sonoffs. Maybe someone else got the same Problem or could "SEE" it.