arendst/Tasmota #3935: Compiling own version of 6.2.1 from source seems to fail sonoff basic

mcc05, 2018-09-29:
Hi I have flashed the 6.2.1 sonoff.bin file successfully and all seems to work fine.

However if I compile from the source code with no change (as I do want to change something), then flash the results, the sonoff basic flashes the LED and presents an AP hotspot but I can connect to it to enter the wifi credentials. I have tried erasing the rom first etc.

I have check the settings I have against what is specified and used the 2.3.0 board manger, any ideas or help would be great.

Jason2866, 2018-09-29:
Please use latest release version from here. If this works, you have something misconfigured in your self compiled version.
For help visit us as https://discord.gg/Ks2Kzd4

ascillato, 2018-09-29:
Hi, Sorry, I can't reproduce your issue. Is compiling ok for me and working fine in my sonoff basic.

closed 2018-09-29 by ascillato2

mcc05, 2018-09-30:
So its not for me, I have cleared out the lib folder, re-downloaded the source and the 2.3.0 lib.
So the compiled version I get is 473K(484,640 bytes) and the downloaded version is 471KB (482,512 bytes) and I have changed nothing in the code.