arendst/Tasmota #3936: Did somebody already look into the SP108E LED controller?

FauthD, 2018-09-29:
Hi,
This is a controller for the WS2812 and similar LED chips (stripes). Prise is ~13€ and up.
So far I analyzed it is equipped with these chips:
- ESP-12F
- STM32F-030C8T (Cortex M0)
- Winbond 25Q32 (EEPROM)
- 74HC245
- XL1509 3.3E1 (Step down regulator)

RX/TX of STM is connected to ESP.
74HC245 buffers the outputs

We can assume there is some firmware in the STM that does the low level connection to the LEDs.
Question is how does the ESP communicate with the STM and what exactly does the STM at all.
In worst case we can mod the hardware so the ESP can talk to the LEDs. Should be simple.
I made some pictures with a microscope and can add this info to the wiki.
Dieter Fauth

ascillato, 2018-09-29:
Hi, 

Please, for general question go to Tasmota Discord Chat at https://discord.gg/Ks2Kzd4

closed 2018-09-29 by ascillato2

G0ldi2222, 2019-03-18:
> Hi,
> This is a controller for the WS2812 and similar LED chips (stripes). Prise is ~13€ and up.
> So far I analyzed it is equipped with these chips:
>
> ESP-12F
> STM32F-030C8T (Cortex M0)
> Winbond 25Q32 [...]

Hi, what was the result of the hardware mod? was it possible to flash the SP108E? don't want to mod the hardware, when it doesn't work... I just bought a new SP108E and unfortunately they changed the Layout of the PCB. but I still do have two controller here with the old layout if the mod would work.

G0ldi2222, 2019-03-18:
[image]

FauthD, 2019-03-18:
Hi,
I had not the time to proceed, the only mod I tried is to add another ESP (keeping the original one in reset).
In the new layout, is the GPIO0 usable? Or still directly connected to VCC?

G0ldi2222, 2019-03-18:
Hi... pity to hear that... was hoping to use my existing one with a useful software:) the installed version is impossible to add to WiFi networks.
As it seems easier for my purpose, I am going to use the esp01 Modul with the led plc. It’s cheaper and I hope do-able for a noob like me:)
That’s why I also can not tell you more about the layout. But I could do some closer pictures, if that would help or test something on it for you, if you instruct me...

Thank you very much for your quick reply and the effort you put in that project!

Best regards, Jan