arendst/Tasmota #3937: release 6.2.1 dropping MQTT messages

jbreda007, 2018-09-29:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : laptop, RPi
- [ x] _Development/Compiler/Upload tools used_ : ESPtool
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Upgraded to 6.2.1 from 6.2.0 and experienced problems with mqtt dropping messages.

**Expected behavior**
_A clear and concise description of what you expected to happen._
Simple on/off commands from any mqtt client and any broker lose messages intermittently, including node-red .  For example I ran a script in MQTTfx turning a sonoff basic on and off in a loop.  Occasionally a cmnd would not be executed, perhaps 1 out of 3 times.  Tried multiple brokers, multiple node-reds, multiple clients, termial commands and all experienced the problem.  Issuing commands from sonoff console works fine.  Tried this on 6 sonoffs, 4 pc's, 45 brokers and 3 clients and all behaved the same.  Reverted to 6.2.0 and all works ok.  Commands used:
 mosquitto_sub -h iot.eclipse.org -t cmnd/GarageDoor/POWER 
 mosquitto_pub -h iot.eclipse.org -t cmnd/GarageDoor/POWER -d -m 0
 mosquitto_pub -h iot.eclipse.org -t cmnd/GarageDoor/POWER -d -m 1

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
I did notice that for some reason there would be a skip in the on/off on the sonoff when going thru the on/off loop, and it wouldnt recover.  Reverting to 6.2.0 I noticed a skip but it seemed to recover wnd catch up with the on/off sequence.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-29:
Hi,

Sorry, I can't reproduce your issue.

Mqtt is working fine for me. no messages lost.

seems that you are having wifi issues.

Frogmore42, 2018-09-29:
You did not include the requested information to truly diagnose your issue.

It looks like you are using an external broker, I know other people have had issues with that.  I use a local broker and have not had any issues with it.

I would suggest setting up a local broker for MQTT.  You said you have a RPi, I have heard it is easy to set up a broker on one, and there are lots of tutorials on it.

You will also need to be able to capture the serial output from the Sonoff.  It provides a wealth of information as to what is going on with it.

closed 2018-09-29 by ascillato2