arendst/Tasmota #3939: Support push buttons for Shelly 1/2

TheEggi, 2018-09-30:
It seems that currently only switches are supported for the Shelly1/2 and the user is not able to configure this. Also the default option for the Shelly should imho be a push button as you cannot use the up/down paddle information of a switch anymore (it does not reflect the real state).

arendst, 2018-09-30:
Shelly supports pushbuttons out-of-the-box in the latest development releases

Have look at command switchmode and issue #3896

Maybe try ``switchmode 4``.

TheEggi, 2018-09-30:
I commented in the other issue - will close this one.

ascillato2, 2018-09-30:
Hi,

Closing. Issue continued at #3896

closed 2018-09-30 by ascillato2

TheEggi, 2018-10-03:
Could you please reopen the other issue as the solution seems to cause issues.