arendst/Tasmota #3940: Add MQTT Last Will

HakamSaffour, 2018-09-30:
MQTT Last Will is a nice feature in MQTT that allows you to know if the connected devices are still online or not. 
Also it allows you to perform a pre-defined actions when the ESP goes offline (from the side of home automation software).

As example:
In 'AsyncMqttClient' library, the last will can be programmed like that:

`mqttClient.setKeepAlive(15).setCleanSession(true).setWill(mqtt_topic_lastwill, 1, true, "no").setCredentials(MQTT_userName, MQTT_pass).setClientId(MQTT_clientID);`

`setKeepAlive(x)` :no of seconds to wait before executing 'the last will'
`setWill(mqtt_topic_lastwill, 1, true, "no")`: "no": is the 'Last Will' value if ESP goes offline (will be sent by the MQTT broker)

arendst, 2018-09-30:
It's already implemented as 
```
tele/wemos6/LWT Online
tele/dualr2/LWT Offline
```
Keep Alive is currently set 15 seconds.

See ``status 6``

ascillato, 2018-09-30:
As this is already implemented. Please close this issue. Thanks

HakamSaffour, 2018-09-30:
Ok....I searched the WiKi before opening the issue but I did not find it

closed 2018-09-30 by HakamSaffour