arendst/Tasmota #3942: Garage Door opener with sensors, publish Rule not working.

svh1985, 2018-09-30:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Visual Studio
- [x] Hardware used : WeMos D1 Mini
- [X] Provide the output of command ``status 0`` :
```
15:45:57 MQT: stat/GarageDoor/STATUS = {"Status":{"Module":18,"FriendlyName":["Garage Door","WeMos D1 Mini2","WeMos D1 Mini3"],"Topic":"GarageDoor","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:45:57 MQT: stat/GarageDoor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://ota.bla.com/tasmota/svh/wemosd1mini.bin","RestartReason":"Software/System restart","Uptime":"0T01:00:16","StartupUTC":"2018-09-30T12:45:41","Sleep":0,"BootCount":9,"SaveCount":96,"SaveAddress":"3F4000"}}
15:45:57 MQT: stat/GarageDoor/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-09-22T20:43:17","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:45:57 MQT: stat/GarageDoor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
15:45:57 MQT: stat/GarageDoor/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":2580,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F082790","00000001","21021796","00000000"]}}
15:45:57 MQT: stat/GarageDoor/STATUS5 = {"StatusNET":{"Hostname":"GarageDoor-5145","IPAddress":"172.16.0.21","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.2","Mac":"A0:20:A6:2F:74:19","Webserver":2,"WifiConfig":5}}
15:45:57 MQT: stat/GarageDoor/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2F7419","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:45:57 MQT: stat/GarageDoor/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 30 13:45:57 2018","Local":"Sun Sep 30 15:45:57 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:39","Sunset":"19:20"}}
15:45:57 MQT: stat/GarageDoor/STATUS10 = {"StatusSNS":{"Time":"2018-09-30T15:45:57","Switch2":"OFF","Switch3":"ON"}}
15:45:57 MQT: stat/GarageDoor/STATUS11 = {"StatusSTS":{"Time":"2018-09-30T15:45:57","Uptime":"0T01:00:16","Vcc":2.788,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"WiFi","RSSI":100,"APMac":"00:1D:AA:69:D5:88"}}}
```

Hello,

I'm building a Garage Door opener with two reed sensors to check the garage door's state.

The sensors are connected to switches that track the door's status:
Switch1 ON = door fully CLOSED
Switch2 ON = door fully OPENED

I would like the senor's to publish the OPENED & CLOSED state to the same topic so I created a rule to have Tasmota publish the state when a switch is triggered:
_rule2 on switch2#state=1 do publish stat/GarageDoor/DOOR CLOSED endon on switch3#state=1 do publish stat/GarageDoor/DOOR OPENED endon_

Unfortunately the rule is not triggered.

My module's configuration:
[image]

ascillato, 2018-09-30:
Do you have rule2 1 in order to enable rules? Do you have switchtopic2 and switchtopic3 both to 0? Which switchmode do you choose?

svh1985, 2018-09-30:
rule2 1:
`{"Rule2":"ON","Once":"OFF","Free":384,"Rules":"on switch2#state=1 do publish stat/GarageDoor/DOOR CLOSED endon on switch3#state=1 do publish stat/GarageDoor/DOOR OPENED endon"}`

switchtopic2:
`{"SwitchTopic":""}`

switchtopic3:
`{"SwitchTopic":""}`

switchmodes2:
`{"SwitchMode2":0}`

switchmodes3:
`{"SwitchMode3":0}`

ascillato2, 2018-09-30:
Hi,

Please, use command weblog 4 in your console in order to see more debug messages and try to push your switches to see what is coming up in the console. Thanks

TLCary, 2018-10-01:
For reference, this works for me:
// Set the relay on time to signal the opener
PulseTime 7

// Send ON and OFF as the switch is ON or OFF
switchmode1 1
switchmode2 1
switchmode3 1

// If you don’t set the topic to something other than the module topic it will run the relay when it talks to itself
switchTopic 2

//No need to save changes on power cycle
setoption0 0

//Don’t blindly run the door on powerup
poweronstate 0

//Turn on Rules
rule1 1
rule2 1
rule3 1

//Onsehot Detection off on most
rule1 4
rule2 4

rule1
 on Switch1#Boot=1 do backlog delay 99; event Opened endon
 on Switch2#Boot=1 do backlog delay 99; event Closed endon
 on Switch3#Boot do backlog delay 99; event PLight=%value% endon
 on EVENT#OPEN do power1 %var2% endon
 on EVENT#CLOSE do power1 %var1% endon
 on EVENT#STOP do backlog power1 %var3%; power1 %var4%; event PState=STOP endon
 on Switch1#State=1 do event Opened endon
 on Switch2#State=1 do event Closed endon
 on Switch1#State=0 do event Closing endon
 on Switch2#State=0 do event Opening endon

rule2
 on event#Opened do backlog var 1; var2 0; var3 0; var4 0; ruletimer1 0; event PState=OPEN endon
 on event#Closed do backlog var1 0; var2 1; var3 0; var4 0; ruletimer1 0; event PState=CLOSE endon
 on event#Opening do backlog var1 0; var2 0; var3 1; var4 0; ruletimer1 15; event PState=OPENING endon
 on event#Closing do backlog var1 0; var2 0; var3 0; var4 1; ruletimer1 15; event PState=CLOSING endon

rule3
 on counter#c1>1000 do event PObstr=0 endon
 on counter#c1<1000 do event PObstr=1 endon
 on event#PObstr do publish stat/GarageDoor/OBSTR %value% endon
 on event#PState do publish stat/GarageDoor/STATE %value% endon
 on rules#timer=1 do event PState=STOP endon

 on event#PLight do publish stat/GarageDoor/LIGHT %value% endon
CounterType 1
CounterDebounce 100
on counter#c1>1000 do publish stat/GarageDoor/OBSTR 0 endon
on counter#c1<1000 do publish stat/GarageDoor/OBSTR 1 endon

TLCary, 2018-10-01:
The door is controlled in parallel with the wall push button switch.
GPIO 04	Switch1	Optical (end stop) limit switch sensing full Open
GPIO 05 	Switch2	Optical (end stop) limit switch sensing full Closed
GPIO 14 	Switch3	Photoresistor facing the lightbulb

Optical switches don’t require a capacitor to filter noise and can be mounted with double sided tape since they don’t make physical contact. Also, they cost $ .76 each, delivered.
The light blinks when an obstruction is sensed, so the Light Sensitive Resistor is used for obstruction detection and to toggle the light its necessary to know if its on.

A Sonoff SV can be used to control the door, but an ESP board with two relays is necessary to control the door and the light. It would require four relays to run the light, switch, and lock.

The door is controlled by connecting a relay in parallel with the wall button and pulsing it on for .7 seconds.  
The light is controlled by connecting a second relay in parallel with the wall button and with a 200 Ohm resistor.  Lock could be controlled with two more relays: locked requires 82 Ohm resistor, unlocked requires 200 Ohms and an LED for voltage drop.

It takes less than 15 seconds for the door to cycle from Open to Closed.  If it takes longer than that between when the upper switch opens until the lower switch closes something has stopped the door and the status is changed to Stopped.  Typically this is a manual intervention with the wall switch.

When an obstruction is detected the light flashes twice a second.  Timers are used and the state changes are counted to identify an obstructed state.

ascillato, 2018-10-01:
Hi,

Are you sending by MQTT the command: `cmnd/GarageDoor/POWER2 = ON` ?

What are you trying to do? you don't have any RELAY2 defined. That is why you have COMMAND UNKNOWN

TLCary, 2018-10-01:
Can’t talk directly to the relay...  send 
Mqtt to cmnd/Sonoff/EVENT
OPEN

ascillato, 2018-10-03:
Hi,

@svh1985 

Have you managed to solve your issue?

svh1985, 2018-10-03:
I've managed to set it up using this rule:
`rule1 on power2#state=1 do publish stat/GarageDoor/DOOR CLOSED endon on power3#state=1 do publish stat/GarageDoor/DOOR OPENED endon`

closed 2018-10-03 by svh1985

ascillato, 2018-10-03:
Great Thanks. All this is a very good example. If you want, can you add all this to the wiki? Would be super useful for a lot of people. Thanks :+1:

andrethomas, 2018-10-03:
And if you don't want to add it to the wiki let me know and I will add it 👍

svh1985, 2018-10-03:
I'm still figuring out the Home Assistant part of it. If I have that part working (almost there..) I will definitely add it!

svh1985, 2018-10-15:
@benzino77 thanks for the suggestion. I'll definitely check it out, I'm having a little less time these weeks but I'll report back asap.

jeylites, 2019-12-13:
Hi I subtle upon this project by chance while looking for a pet feeder rule control of the limit switch

I plan to replace the electronic of my Pet Feeder a Sonoff Inching. The current setup has got a DC 5V motor, a limit switch for food portion and a push-button for manual control.

I have wired the Sonnof to the motor with a set a pulse time of 30sec to dispense one quart of food. The issue is the portion of food differs all the time despite having a 30 Sec set time. I would like to make use of the limit switch and Rule for better control of the food portions.

How do I go about doing this with Rules? What rule option should I use to do this?

[image]

ascillato, 2019-12-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

SirGoodenough, 2020-11-22:
I took this awesome Garage Door code 1 step further and made it work with RF433 door switches.  Check out my Blog and video about it.   Thanks @TLCary for the starting point and the elegant code.

https://whatarewefixing.today/541/control-your-garage-door-sonoff-sv-and-2-gs-wds07-rf433-reed-switches/

jeylites, 2022-03-23:
@svh1985 

Hello, I'm testing the garage door opener project. I got the relay to trigger but can't seem to get the state to work. Would you mind checking to see if the config is correct?

**Plugin**
[HB Plugin](https://github.com/arachnetech/homebridge-mqttthing#readme)

**HB Config**
```
"type": "garageDoorOpener",
            "name": "AUTO",
            "url": "192.168.1.200",
            "accessory": "mqttthing",
            "topics": {
                "getTargetDoorState": "stat/GarageDoor/STATE",
                "getCurrentDoorState": "stat/GarageDoor/STATE",
                "getLockTargetState": "tele/tasmota_65E47F/STATE",
                "setTargetDoorState": "cmnd/tasmota_65E47F/POWER"
            },
            "doorCurrentValues": [
                "Open",
                "Closed",
                "Opening",
                "Closing",
                "Stopped"
            ],
            "doorTargetValues": [
                "on",
                "off"
            ],
            "lockValues": [
                "Unsecured",
                "Secured",
                "Jammed",
                "Unknown"
            ]
```
**Wemos D1 - Settings & Logs**
[image]

```
03:44:21.859 MQT: stat/tasmota_65E47F/RESULT = {"Var1":"0"}
03:44:22.088 MQT: stat/tasmota_65E47F/RESULT = {"Var2":"0"}
03:44:22.314 MQT: stat/tasmota_65E47F/RESULT = {"Var3":"0"}
03:44:22.541 MQT: stat/tasmota_65E47F/RESULT = {"Var4":"1"}
03:44:22.772 MQT: stat/tasmota_65E47F/RESULT = {"T1":15,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
03:44:22.972 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:23.047 RUL: EVENT#PSTATE performs "publish stat/GarageDoor/STATE CLOSING"
03:44:23.053 MQT: stat/GarageDoor/STATE = CLOSING
03:44:25.392 RUL: SWITCH1#STATE=1 performs "backlog event Opened"
03:44:25.422 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:25.493 RUL: EVENT#OPENED performs "Backlog var 1; var2 0; var3 0; var4 0; ruletimer1 0; event PState=OPEN"
03:44:25.520 MQT: stat/tasmota_65E47F/RESULT = {"Var1":"0","Var2":"0","Var3":"0","Var4":"1","Var5":"","Var6":"","Var7":"","Var8":"","Var9":"","Var10":"","Var11":"","Var12":"","Var13":"","Var14":"","Var15":"","Var16":""}
03:44:25.729 MQT: stat/tasmota_65E47F/RESULT = {"Var2":"0"}
03:44:25.957 MQT: stat/tasmota_65E47F/RESULT = {"Var3":"0"}
03:44:26.171 MQT: stat/tasmota_65E47F/RESULT = {"Var4":"0"}
03:44:26.402 MQT: stat/tasmota_65E47F/RESULT = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
03:44:26.521 RUL: SWITCH1#STATE=0 performs "backlog event Closing"
03:44:26.545 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:26.619 RUL: EVENT#PSTATE performs "publish stat/GarageDoor/STATE OPEN"
03:44:26.624 MQT: stat/GarageDoor/STATE = OPEN
03:44:26.837 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:26.910 RUL: EVENT#CLOSING performs "Backlog var1 0; var2 0; var3 0; var4 1; ruletimer1 15; event PState=CLOSING"
03:44:26.931 MQT: stat/tasmota_65E47F/RESULT = {"Var1":"0"}
03:44:27.170 MQT: stat/tasmota_65E47F/RESULT = {"Var2":"0"}
03:44:27.395 MQT: stat/tasmota_65E47F/RESULT = {"Var3":"0"}
03:44:27.620 MQT: stat/tasmota_65E47F/RESULT = {"Var4":"1"}
03:44:27.848 MQT: stat/tasmota_65E47F/RESULT = {"T1":15,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
03:44:28.099 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:28.177 RUL: EVENT#PSTATE performs "publish stat/GarageDoor/STATE CLOSING"
03:44:28.184 MQT: stat/GarageDoor/STATE = CLOSING
03:44:28.215 RUL: SWITCH1#STATE=1 performs "backlog event Opened"
03:44:28.247 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:28.317 RUL: EVENT#OPENED performs "Backlog var 1; var2 0; var3 0; var4 0; ruletimer1 0; event PState=OPEN"
03:44:28.346 MQT: stat/tasmota_65E47F/RESULT = {"Var1":"0","Var2":"0","Var3":"0","Var4":"1","Var5":"","Var6":"","Var7":"","Var8":"","Var9":"","Var10":"","Var11":"","Var12":"","Var13":"","Var14":"","Var15":"","Var16":""}
03:44:28.554 MQT: stat/tasmota_65E47F/RESULT = {"Var2":"0"}
03:44:28.781 MQT: stat/tasmota_65E47F/RESULT = {"Var3":"0"}
03:44:29.009 MQT: stat/tasmota_65E47F/RESULT = {"Var4":"0"}
03:44:29.239 MQT: stat/tasmota_65E47F/RESULT = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
03:44:29.488 MQT: stat/tasmota_65E47F/RESULT = {"Event":"Done"}
03:44:29.564 RUL: EVENT#PSTATE performs "publish stat/GarageDoor/STATE OPEN"
03:44:29.572 MQT: stat/GarageDoor/STATE = OPEN
03:45:09.195 MQT: tele/tasmota_65E47F/STATE = {"Time":"2022-03-23T03:45:09","Uptime":"0T00:05:09","UptimeSec":309,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SMILE","BSSId":"98:77:E7:72:CB:65","Channel":6,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}
03:45:09.231 MQT: tele/tasmota_65E47F/SENSOR = {"Time":"2022-03-23T03:45:09","Switch1":"ON","Switch2":"ON","Switch3":"ON"}
03:48:41.123 MQT: stat/tasmota_65E47F/RESULT = {"POWER":"ON"}
03:48:41.126 MQT: stat/tasmota_65E47F/POWER = ON
03:48:41.879 MQT: stat/tasmota_65E47F/RESULT = {"POWER":"OFF"}
03:48:41.883 MQT: stat/tasmota_65E47F/POWER = OFF
```

svh1985, 2022-03-23:
@jeylites I switched to ESPHome a few years ago, so my knowledge about Tasmota become too rusty to help you out, sorry!

jeylites, 2022-03-23:
Would you mind sharing the setup you did for ESPhome. In fact I’m considering switching over to Home Assistance.

barbudor, 2022-03-23:
@jeylites from your logs, Tasmota is properly publishing the state on the stat/GarageDoor/STATE topic
So problem is more likely to be on your backend system
May be your system is using access rules on the MQTT broker which prevent the device

jeylites, 2022-03-23:
@barbudor 
What about the Homebridge config, is it setup correctly?

barbudor, 2022-03-23:
I have not even a single idea of what is Homebridge
My knowledge is limited to Tasmota, mosquitto, node-red, influxdb, grafana

barbudor, 2022-03-23:
I suggest that you open a new discussion with all details of your setup. That will bring more visibility than adding comments on an old and closed issue

jeylites, 2022-03-23:
@barbudor ok, will setup a new discussion shortly.